dusty-nv/jetson-inference #504: Raspberry Pi Camera not working

kiwibird22, 2020-02-14:
Hello @dusty-nv 

I have been following along with your amazing tutorial here https://www.youtube.com/watch?v=bcM5AQSAzUY.
Everything was going well until I installed and ran `v4l2-ctl --list-devices`
and it gives me the error: `Failed to open /dev/video0: No such file or directory`
My raspberry pi camera isn't genuine, but it worked fine with my raspberry pis.

I would really appreciate if you could help me out with this.

Looking forward to your reply,
kiwbird22

dusty-nv, 2020-02-14:
Hi kiwibird22, it needs to be an IMX219 sensor (like rpi camera module v2) to use mipi csi driver. Otherwise you could use USB camera.  Do you get a video stream from nvgstcapture program?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

kiwibird22, 2020-02-14:
What is nvgstcapture?
Do I just run the command?

kiwibird22, 2020-02-14:
Output I get:
`Encoder null, cannot set bitrate!
Encoder Profile = High
Supported resolutions in case of ARGUS Camera
  (2) : 640x480
  (3) : 1280x720
  (4) : 1920x1080
  (5) : 2104x1560
  (6) : 2592x1944
  (7) : 2616x1472
  (8) : 3840x2160
  (9) : 3896x2192
  (10): 4208x3120
  (11): 5632x3168
  (12): 5632x4224

Runtime ARGUS Camera Commands:

  Help : 'h'
  Quit : 'q'
  Set Capture Mode:
      mo:<val>
          (1): image
          (2): video
  Get Capture Mode:
      gmo
  Set sensor orientation:
      so:<val>
          (0): none
          (1): Rotate counter-clockwise 90 degrees
          (2): Rotate 180 degrees
          (3): Rotate clockwise 90 degrees
  Get sensor orientation:
      gso
  Set sensor mode:
      smo:<val> e.g., smo:1
  Get sensor mode:
      gsmo
  Set Whitebalance Mode:
      wb:<val>
          (0): off
          (1): auto
          (2): incandescent
          (3): fluorescent
          (4): warm-fluorescent
          (5): daylight
          (6): cloudy-daylight
          (7): twilight
          (8): shade
          (9): manual
  Get Whitebalance Mode:
      gwb
  Set Saturation (0 to 2):
      st:<val> e.g., st:1.25
  Get Saturation:
      gst
  Set Exposure Compensation (-2 to 2):
      ec:<val> e.g., ec:-2
  Get Exposure Compensation:
      gec
  Set Auto Whitebalance Lock:
      awbl:<val> e.g., awbl:0
  Get Auto Whitebalance Lock:
      awbl
  Set Auto Exposure Lock:
      ael:<val> e.g., ael:0
  Get Auto Exposure Lock:
      gael
  Set TNR Mode:
      tnrm:<val> e.g., tnrm:1
          (0): OFF
          (1): FAST
          (2): HIGH QUALITY
  Get TNR Mode:
      gtnrm
  Set TNR Strength (-1 to 1):
      tnrs:<val> e.g., tnrs:0.5
  Get TNR Strength:
      gtnrs
  Set EE Mode:
      eem:<val> e.g., eem:1
          (0): OFF
          (1): FAST
          (2): HIGH QUALITY
  Get EE Mode:
      geem
  Set EE Strength (-1 to 1):
      ees:<val> e.g., ees:0.5
  Get EE Strength:
      gees
  Set Auto Exposure Anti-Banding (0 to 3):
      aeab:<val> e.g., aeab:2
          (0): OFF
          (1): MODE AUTO
          (2): MODE 50HZ
          (3): MODE 60HZ
  Get Auto Exposure Anti-Banding:
      gaeab
  Set Gain Range:
      gr:<val><space><val> e.g., gr:1 16
  Get Gain Range:
      ggr
  Set Exposure Time Range:
      etr:<val><space><val> e.g., etr:34000 35000
  Get Exposure Time Range:
      getr
  Set ISP Digital Gain Range:
      dgr:<val><space><val> e.g., dgr:2 152
  Get ISP Digital Gain Range:
      gdgr
  Capture: enter 'j' OR
           followed by a timer (e.g., jx5000, capture after 5 seconds) OR
           followed by multishot count (e.g., j:6, capture 6 images)
           timer/multihot values are optional, capture defaults to single shot with timer=0s
  Start Recording : enter '1'
  Stop Recording  : enter '0'
  Video snapshot  : enter '2' (While recording video)
  Get Preview Resolution:
      gpcr
  Get Image Capture Resolution:
      gicr
  Get Video Capture Resolution:
      gvcr

Runtime encoder configuration options:

  Set Encoding Bit-rate(in bytes):
      br:<val> e.g., br:4000000
  Get Encoding Bit-rate(in bytes):
      gbr
  Set Encoding Profile(only for H.264):
      ep:<val> e.g., ep:1
          (0): Baseline
          (1): Main
          (2): High
  Get Encoding Profile(only for H.264):
      gep
  Force IDR Frame on video Encoder(only for H.264):
      Enter 'f' 

bitrate = 4000000
Encoder Profile = High
Encoder control-rate = 1
Encoder EnableTwopassCBR = 0`

I just tried again with my raspberry pi, and it works fine with the pi.

Thanks for helping

kiwibird22, 2020-02-14:
Also got this:
```
Opening in BLOCKING MODE
** Message: 11:46:55.625: <main:4670> iterating capture loop ....
NvMMLiteOpen : Block : BlockType = 4
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:532 No cameras available
```

dusty-nv, 2020-02-18:
@kiwibird22 if `nvgstcapture` is unable to detect the camera (it says `No cameras available`), then the jetson-inference program won't be able to, either.   It seems your Raspberry Pi Camera Module is not the v2 version that uses IMX219 sensor.  This is the sensor that is supported out-of-the-box on Jetson Nano by the camera drivers.

beilpig, 2020-02-28:
Hi Dusty
Having an issue with camera not showing when running ./detectnet-camera --camera=/dev/video0

The camera has tested as working no problem however just seems not to work with this code.

The terminal window shows camera initialises then continues to execute without actually showing the camera view...excerpt output from terminal shows this:

beil@beil-jetson:~/jetson-inference/build/aarch64/bin$ ./detectnet-camera --camera=/dev/video0
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera attempting to initialize with GST_SOURCE_NVARGUS, camera /dev/video0
[gstreamer] gstCamera pipeline string:
v4l2src device=/dev/video0 ! video/x-raw, width=(int)1280, height=(int)720, format=YUY2 ! videoconvert ! video/x-raw, format=RGB ! videoconvert !appsink name=mysink
[gstreamer] gstCamera successfully initialized with GST_SOURCE_V4L2, camera /dev/video0

detectnet-camera:  successfully initialized camera device
    width:  1280
   height:  720
    depth:  24 (bpp)

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]   TensorRT version 6.0.1
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]   Plugin Creator registration succeeded - GridAnchorRect_TRT
[TRT]   Plugin Creator registration succeeded - NMS_TRT
[TRT]   Plugin Creator registration succeeded - Reorg_TRT
[TRT]   Plugin Creator registration succeeded - Region_TRT
[TRT]   Plugin Creator registration succeeded - Clip_TRT
[TRT]   Plugin Creator registration succeeded - LReLU_TRT
[TRT]   Plugin Creator registration succeeded - PriorBox_TRT
[TRT]   Plugin Creator registration succeeded - Normalize_TRT
[TRT]   Plugin Creator registration succeeded - RPROI_TRT
[TRT]   Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]   Could not register plugin creator:  FlattenConcat_TRT in namespace: 
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - UFF  (extension '.uff')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.GPU.FP16.engine
[TRT]   cache file not found, profiling network model on device GPU
[TRT]   device GPU, loading /home/beil/jetson-inference/build/aarch64/bin/ networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
[TRT]   UFFParser: Parsing Input[Op: Input].
[TRT]   UFFParser: Input -> [3,300,300]
[TRT]   UFFParser: Applying order forwarding to: Input
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/weights[Op: Const].
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/weights -> [3,3,3,32]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/weights
[TRT]   UFFParser: Parsing FeatureExtractor/MobilenetV2/Conv/Conv2D[Op: Conv]. Inputs: Input, FeatureExtractor/MobilenetV2/Conv/weights
[TRT]   UFFParser: FeatureExtractor/MobilenetV2/Conv/Conv2D -> [32,150,150]
[TRT]   UFFParser: Applying order forwarding to: FeatureExtractor/MobilenetV2/Conv/Conv2D
...
Any advice would be greatly appreciated!

Cheers

beilpig, 2020-03-01:
Never mind... I reinstalled the jetpack then built inference package and now works...

mat-l, 2020-03-02:
I have a similar problem with a USB camera.

v4l2-ctl --list-devices shows this result:
```
USB Camera (046d:08af) (usb-70090000.xusb-2.1):
	/dev/video0
```

But anyways at some point gstreamer or anything else is crashing:
`[gstreamer] gstCamera onEOS
[gstreamer] gstreamer v4l2src0 ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink`

gooohjy, 2020-05-24:
Hi @dusty-nv, I have faced a similar problem. I have followed the youtube tutorial and the only changes in the code is changing video1 to video0 (since I am using a raspberry pi camera). 

This is the error, the program hangs here:
```
[gstreamer] gstreamer v4l2src0 ERROR Internal data stream error.
[gstreamer] gstreamer Debugging info: gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0:
streaming stopped, reason not-negotiated (-4)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
detectnet-camera:  camera open for streaming
```

For your information, here is my output for running `v4l2-ctl --list-devices`:

```
vi-output, imx219 7-0010 (platform:54080000.vi:0):
	/dev/video0
```
Here is my output for running `v4l2-ctl --device /dev/video0 --list-formats-ext`:
```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'RG10'
	Name        : 10-bit Bayer RGRG/GBGB
		Size: Discrete 3264x2464
			Interval: Discrete 0.048s (21.000 fps)
		Size: Discrete 3264x1848
			Interval: Discrete 0.036s (28.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
```

My camera works when I run `nvgstcapture` so I'm not too sure what's the issue here. Any help is appreciated!

dusty-nv, 2020-05-24:
Hi, you want to use the MIPI CSI interface for rPi camera and not v4l2.  So use "0" or --camera=0

ColtsMild, 2020-09-06:
This code worked for me with Nano Dev Kit and IMX219 camera...

import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.videoSource()
display = jetson.utils.videoOutput()

while camera.IsStreaming() or not display.IsStreaming():
	img = camera.Capture()
	detections = net.Detect(img)
	display.Render(img)
	display.SetStatus("Object Detections: {:d} ".format(len(detections)) + " | Network: {:.0f} FPS".format(net.GetNetworkFPS()))