dusty-nv/jetson-inference #505: jetson.inference -- imageNet.classify() encountered an error classifying the image

satinder147, 2020-02-15:
Instead of running the imageNet-camera.py as it is, I am trying to run detection using an image from OpenCV. I found out that NumPy array can be converted into pycapsule object using jetson.utils.cudaFromNumpy(). I am trying to run the detection on the result returned from jetson.utils.cudaFromNumpy(). 
But the above errors come. my code:
```
import jetson.inference
import jetson.utils
import cv2
import argparse
import sys
import numpy as np

# parse the command line
parser = argparse.ArgumentParser(description="Classify a live camera stream using an image recognition DNN.", 
						   formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.imageNet.Usage())

parser.add_argument("--network", type=str, default="googlenet", help="pre-trained model to load (see below for options)")
parser.add_argument("--camera", type=str, default="0", help="index of the MIPI CSI camera to use (e.g. CSI camera 0)\nor for VL42 cameras, the /dev/video device to use.\nby default, MIPI CSI camera 0 will be used.")
parser.add_argument("--width", type=int, default=1280, help="desired width of camera stream (default is 1280 pixels)")
parser.add_argument("--height", type=int, default=720, help="desired height of camera stream (default is 720 pixels)")

try:
	opt = parser.parse_known_args()[0]
except:
	print("")
	parser.print_help()
	sys.exit(0)

# load the recognition network
net = jetson.inference.imageNet(opt.network, sys.argv)

# create the camera and display
#font = jetson.utils.cudaFont()
#camera = jetson.utils.gstCamera(opt.width, opt.height, opt.camera)
#display = jetson.utils.glDisplay()
cap=cv2.VideoCapture(0)
# process frames until user exits
print(type(jetson.utils.cudaFromNumpy(np.ones((224,224,3)))))
while 1:
	# capture the image
	frame=cap.read()[1]
	print(frame.shape)
	frame=cv2.resize(frame,(1280,720))
	frames,width,height = jetson.utils.cudaFromNumpy(np.transpose(frame,(1,0,2))),1280,720

	#img, width, height = camera.CaptureRGBA()
	#print(width,height)
	#print(type(img))
	cv2.imshow("sd",frame)
	cv2.waitKey(1)
	# classify the image
	class_idx, confidence = net.Classify(frames, width, height)

	# find the object description
	class_desc = net.GetClassDesc(class_idx)
	
	# overlay the result on the image	
	#font.OverlayText(img, width, height, "{:05.2f}% {:s}".format(confidence * 100, class_desc), 5, 5, font.White, font.Gray40)
	
	# render the image
	#display.RenderOnce(img, width, height)

	# update the title bar
	#
	# print out performance info
	net.PrintProfilerTimes()
	break

```
**The error is **
```
[TRT]   engine.cpp (555) - Cuda Error in execute: 4
[TRT]   engine.cpp (555) - Cuda Error in execute: 4
[TRT]   imageNet::Process() -- failed to execute TensorRT network
[TRT]   imageNet::Process() failed
Traceback (most recent call last):
  File "imagenet-camera.py", line 70, in <module>
    class_idx, confidence = net.Classify(frames, width, height)
Exception: jetson.inference -- imageNet.Classify() encountered an error classifying the image
PyTensorNet_Dealloc()
[TRT]   runtime.cpp (30) - Cuda Error in free: 4
terminate called after throwing an instance of 'nvinfer1::CudaError'
  what():  std::exception
Aborted (core dumped)

```

dusty-nv, 2020-02-18:
Hi @satinder147 , the CV image passed to `cudaFromNumpy` needs to be a 4-channel RGBA image, as `imageNet.Classify()` also expects a 4-channel RGBA image.

See the code snippet from this post that uses `cv2.cvtColor()` to transform the image from 3-channel to 4-channel:   https://devtalk.nvidia.com/default/topic/1071209/jetson-nano/detectnet-video-/post/5427509/#5427509

surefyyq, 2021-07-09:
Hi @dusty-nv, same issue here
I followed Re-training on the Cat/Dog Dataset and have no problems.
According to `my-recognition.py`,
I load a resnet18 ImageNet successfully by `net = jetson.inference.imageNet('resnet-18', argv)` ,
but when it comes to inference: `class_idx, confidence = net.Classify(img)`,
it throws out the error Exception: `jetson.inference -- imageNet.Classify() encountered an error classifying the image`

the img is loaded by` img = jetson.utils.loadImage('test.jpg')`,
`print(img.shape, type(img)) `is like this: `(334, 500, 3) <class 'jetson.utils.cudaImage>`

So, if I want to load the image directly from its file path, what's the suggested way to do it?

surefyyq, 2021-07-14:
Hi @satinder147 , 
have you solved this issue and how, please?

closed 2023-04-27 by dusty-nv