dusty-nv/jetson-inference #507: Jetson crash after few seconds running the my-detection.py

WSPluta, 2020-02-18:
So I'm trying to run this using Pi camera

```
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.gstCamera(1280, 720, "0")  # using V4L2
display = jetson.utils.glDisplay()

while display.IsOpen():
	img, width, height = camera.CaptureRGBA()
	detections = net.Detect(img, width, height)
	display.RenderOnce(img, width, height)
	display.SetTitle("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))

```

It starts fro couple of seconds but then it crashes, n00b question but how can I add log file to see what was the thing that caused Jetson to shutdown?
I have tried 2 power sources but getting same problem on both

dusty-nv, 2020-02-18:
@WSPluta by crash, do you mean the application crashes, or that your Nano abruptly shuts down?

If your Nano turns off, that would point to an issue with power supply.  In that case, try putting your Nano in 5W mode before running the application:

`$ sudo nvpmodel -m 1`

If it no longer crashes, this would confirm an issue with your power supply.  Before running it in 10W mode again (`sudo nvpmodel -m 0`), try running one of these recommended power supplies listed here:

https://devtalk.nvidia.com/default/topic/1048640/jetson-nano/power-supply-considerations-for-jetson-nano-developer-kit/

jmeisele, 2020-02-21:
I used to have the same issue until I switched to the barrel plug option. Been running fine like a champ since then. Even trained multiple NNs on the nano to stress test, still no issues.

jmyth742, 2021-08-17:
hi, 
i am using a jetson xavier NX , i have installled all the artifacts to build dusty-nv.

All build fine and i can run imagenet and i get the classifcations of images etc...

However, when i run the detectnet with an image my xavier keeps crashing (reboot crash)

everytime here - 

[TRT]    --------------- Timing Runner: BoxPredictor_0/BoxEncodingPredictor/Conv2D || BoxPredictor_0/ClassPredictor/Conv2D (FusedConvActConvolution)

Just hangs and then the device crashes and reboots. 

I have tried re-downloading models, rebuilding etc. But nothing is working at all. 

 below are the pkgs etc installed with versions.
```
ii  graphsurgeon-tf                                    7.1.3-1+cuda10.2                                 arm64        GraphSurgeon for TensorRT package
ii  libnvinfer-bin                                     7.1.3-1+cuda10.2                                 arm64        TensorRT binaries
ii  libnvinfer-dev                                     7.1.3-1+cuda10.2                                 arm64        TensorRT development libraries and headers
ii  libnvinfer-doc                                     7.1.3-1+cuda10.2                                 all          TensorRT documentation
ii  libnvinfer-plugin-dev                              7.1.3-1+cuda10.2                                 arm64        TensorRT plugin libraries
ii  libnvinfer-plugin7                                 7.1.3-1+cuda10.2                                 arm64        TensorRT plugin libraries
ii  libnvinfer-samples                                 7.1.3-1+cuda10.2                                 all          TensorRT samples
ii  libnvinfer7                                        7.1.3-1+cuda10.2                                 arm64        TensorRT runtime libraries
ii  libnvonnxparsers-dev                               7.1.3-1+cuda10.2                                 arm64        TensorRT ONNX libraries
ii  libnvonnxparsers7                                  7.1.3-1+cuda10.2                                 arm64        TensorRT ONNX libraries
ii  libnvparsers-dev                                   7.1.3-1+cuda10.2                                 arm64        TensorRT parsers libraries
ii  libnvparsers7                                      7.1.3-1+cuda10.2                                 arm64        TensorRT parsers libraries
rc  nvidia-container-csv-tensorrt                      7.1.3.0-1+cuda10.2                               arm64        Jetpack TensorRT CSV file
ii  python-libnvinfer                                  7.1.3-1+cuda10.2                                 arm64        Python bindings for TensorRT
ii  python3-libnvinfer                                 7.1.3-1+cuda10.2                                 arm64        Python 3 bindings for TensorRT
ii  python3-libnvinfer-dev                             7.1.3-1+cuda10.2                                 arm64        Python 3 development package for TensorRT
ii  tensorrt                                           7.1.3.0-1+cuda10.2                               arm64        Meta package of TensorRT
ii  uff-converter-tf                                   7.1.3-1+cuda10.2                                 arm64        UFF converter for TensorRT package

```