dusty-nv/jetson-inference #509: jetson.utils.cudaToNumpy() fails with "failed to get array pointer from PyCapsule container

jtiscione, 2020-02-21:
```
import jetson.utils

camera = jetson.utils.gstCamera(1280, 720)
img, width, height = camera.CaptureRGBA()
numpyArray = jetson.utils.cudaToNumpy(img, width, height, 4)
```

Prints "Exception: jetson.utils -- cudaToNumpy() failed to get array pointer from PyCapsule container"

dusty-nv, 2020-02-21:
Try `camera.CaptureRGBA(zeroCopy=True)` instead.  This will enable CPU access so `cudaToNumpy()` should be able to use it

jtiscione, 2020-02-22:
OK I have another question. If I run this:
```
import jetson.utils
camera = jetson.utils.gstCamera(1280, 720)
img, width, height = camera.CaptureRGBA(zeroCopy=1)
jetson.utils.saveImageRGBA('/home/jetson/test.png', img, width, height)
```
If I then run `npArray=jetson.utils.cudaToNumpy(img, width, height)` and show it with matplotlib, I can see the image.

If I omit the zeroCopy=1 argument in camera.CaptureRGBA(), saveImageRGBA() prints "failed to get input image pointer from PyCapsule container". It looks like saveImageRGBA() is broken.

dusty-nv, 2020-02-22:
You need gstCamera.CaptureRGBA(zeroCopy=1) for saveImageRGBA because it needs the data on the CPU to save it.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

jtiscione, 2020-02-23:
Yes but it's saving a blank image if I do that.

dusty-nv, 2020-02-23:
Try inserting a call to jetson.utils.cudaDeviceSynchronize() before you call saveImageRGBA()

jtiscione, 2020-02-24:
Trying this:
```
import jetson.utils
camera = jetson.utils.gstCamera(1280, 720)
img, width, height = camera.CaptureRGBA(zeroCopy=1)
jetson.utils.cudaDeviceSynchronize()
jetson.utils.saveImageRGBA('test.png', img, width, height)
```
Ubuntu's default Image Viewer app still displays test.png as a blank/transparent image.

jtiscione, 2020-02-25:
This is the same issue that was reported in https://github.com/dusty-nv/jetson-inference/issues/383 where someone else who didn't realize they needed zeroCopy=1 in CaptureRGBA() filed an issue, and once that was made clear, another person ran into the subsequent problem with saveImageRGBA()saving blank images.

Since this is not about cudaToNumpy() or CaptureRGBA() anymore I'll close this issue and file a new one specifically about saveImageRGBA- since invoking cudaDeviceSynchronize() doesn't seem to fix the problem.

closed 2020-02-25 by jtiscione