dusty-nv/jetson-inference #511: Converting image with detections to proper image array?

Totemi1324, 2020-02-25:
**Hello there,**

I am confused by the image format of this networks. I am using SSD MobileNet v2 for object detection but don't want to use it with the compatible functions like `camera.CaptureRGBA() `or `display.RenderOnce()` and so on. In the rest of my code, I mainly use OpenCV functions, so I want to use it here, too. My code:

```
import cv2
import numpy as np
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold = 0.5)

def gstreamer_pipeline (capture_width=1280, capture_height=720, display_width=1280, display_height=720, framerate=21, flip_method=0):   
    return ('nvarguscamerasrc ! ' 
    'video/x-raw(memory:NVMM), '
    'width=(int)%d, height=(int)%d, '
    'format=(string)NV12, framerate=(fraction)%d/1 ! '
    'nvvidconv flip-method=%d ! '
    'video/x-raw, width=(int)%d, height=(int)%d, format=(string)BGRx ! '
    'videoconvert ! '
    'video/x-raw, format=(string)BGR ! appsink'  % (capture_width,capture_height,framerate,flip_method,display_width,display_height))

cap = cv2.VideoCapture(gstreamer_pipeline(flip_method=0), cv2.CAP_GSTREAMER)
while cap.isOpened():
    ret, img = cap.read()
    detections = net.Detect(img, 1280, 720)
    cv2.imshow('Object Detection', img)
    if cv2.waitKey(1) & 0xFF == ord('c'):
        break
```

So I am capturing images in 1280x720 format with Gstreamer and OpenCV which are normal readable arrays, but the net.Detect() function doesn't like this format. How can I convert my arrays to an image that can be used by DetectNet and back?

_Thanks for the answers in advance!_

jtiscione, 2020-02-25:
You can do it by converting your image to a numpy array and passing it to jetson.utils.cudaFromNumpy() to copy it into CUDA memory where net.Detect() can get to it.

closed 2020-02-25 by Totemi1324

bl6594, 2020-04-03:
@Totemi1324 ,did the approach proposed by Jason work? further, can your code be modified to take a video file instead of a live camera as input?

Totemi1324, 2020-04-03:
Yes, my code works correctly now. The answer by @jtiscione is only the half solution though, as you also must parse it back to Numpy and watch the correct data formats. Here the code for the image conversion:
```
img = cv2.cvtColor(img, cv2.COLOR_BGR2RGB)
img = cv2.cvtColor(img, cv2.COLOR_RGB2RGBA).astype(np.float32) //convert to RGBA with 32-bit float numbers
img = jetson.utils.cudaFromNumpy(img)
detections = net.Detect(img, 1280, 720) //detect objects
img = jetson.utils.cudaToNumpy(img, 1280, 720, 4)
img = cv2.cvtColor(img, cv2.COLOR_RGBA2RGB).astype(np.uint8) //back to OpenCV-format (BGR with 8-bit ints
img = cv2.cvtColor(img, cv2.COLOR_RGB2BGR)
```
You can also easily modify the code for video input by taking out the pipeline and just inserting the video file's name:
`cap = cv2.VideoCapture('Sample_Video.mp4')`
Of course, don't forget your file must be in the same directory as your code, otherwise just enter the relative path.
Hope it now works for you!

bl6594, 2020-04-03:
Thank you for sharing!  I found that .astype() slows down the process quite a bit.  Do you have any concern on the speed?
Thanks again.