dusty-nv/jetson-inference #513: I need filesrc function

maponarooo, 2020-02-26:
I am very pleased with the jetson-inference framework and am using it. 
Really Thank you. 
Recently I'm trying to object detection from a video file.
So I have tried find the function I could not find that. Like a GstgetCamera()...
I want to know how can I find it or have to modify source?

dusty-nv, 2020-02-26:
Hi @maponarooo, see this thread:   https://devtalk.nvidia.com/default/topic/1071209/jetson-nano/detectnet-video-/post/5427509/#5427509

maponarooo, 2020-02-27:
> Hi @maponarooo, see this thread: https://devtalk.nvidia.com/default/topic/1071209/jetson-nano/detectnet-video-/post/5427509/#5427509

Wow! It is what I looking for. Thanks a lot~~~

closed 2023-04-27 by dusty-nv