dusty-nv/jetson-inference #514: Converting cv::Mat back to float*

efedoganay, 2020-02-28:
Hi, I am getting a video stream from udp on Jetson Nano and I implemented it by VideoCapture function. To be able to detect objects, I need to convert cv::Mat to float*. Look at the code below for more detail. 

(modifying detectnet-camera.cpp)

`cap.read(frame);`
 `cv::cvtColor(frame, frame, cv::COLOR_BGR2RGBA);`
 ` float* imgRGBA = frame.ptr<float>(0);`
 ` detectNet::Detection* detections = NULL;`
  `const int numDetections = net->Detect(imgRGBA, camera->GetWidth(), camera->GetHeight(), 
  &detections, overlayFlags);`

This code seems not working. Any suggestions?

lilin531, 2020-04-06:
Have you solved this issue?

dusty-nv, 2021-10-04:
Sorry for the delay - you need to copy the memory to GPU first.  See this post:  https://github.com/dusty-nv/jetson-inference/issues/129#issuecomment-324130091

closed 2023-04-27 by dusty-nv