dusty-nv/jetson-inference #516: Creating object detection dataset with AttributeError '_csv.reader' object has no attribute 'next'

shinchenhu168, 2020-03-03:
Hello,
I got this problem
I'm trying to create an ObjectDetectionDataset with Digits 20.02-caffe-py3, but if I put something in custom classes I get an AttributeError ('_csv.reader' object has no attribute 'next').
If I leave this field blank it works

I wonder if it is the python version:
when i create an ObjectDetectionDataset with 19.05-caffe <--work fine
(before digits 19.10, the python version is 2.7)
Can anybody helps? thanks a lot.

ps. the problem seems the same as:

[https://devtalk.nvidia.com/default/topic/1072386/container-digits/creating-object-detection-dataset-throw-an-attribute-error/](https://devtalk.nvidia.com/default/topic/1072386/container-digits/creating-object-detection-dataset-throw-an-attribute-error/)

[https://devtalk.nvidia.com/default/topic/1070646/container-digits/locating-objects-with-detectnet-error/](https://devtalk.nvidia.com/default/topic/1070646/container-digits/locating-objects-with-detectnet-error/)

[https://github.com/NVIDIA/DIGITS/issues/2212](https://github.com/NVIDIA/DIGITS/issues/2212)

shinchenhu168, 2020-03-05:
Hi all, i guess the problem is from the difference between python3 and python2
seems i successfully bypassed this issue by using 2 digits container mounting the same "digits-jobs" folder:
1. digits version before 19.10 (the image is using python2) <-- create data sets in "digits-jobs" folder
 then stop above container and start second container:
2. digits version before 19.12 (the image is using python3)  <-- create model using the data sets created by container1

if this solution works for you please tell me. thanks

MaxKinny, 2020-04-25:
Leaving "Custom classes " blank can solve this problem.

drakorg, 2020-05-05:
Hi, I'm becoming increasingly frustrated with these docker images to be honest ... I've also come across this issue. Your suggested workaround worked for me. But it's still a pain to have to manage two different versions just for this. 

Just to know, is there any digits image that you recommend that works well enough to "at least" let you complete the 2 days to a demo? Every image that I try has a different problem. 

Is there at least one stable version?

Thanks.

drakorg, 2020-05-05:
Hi, after many failed attempts I finally found an image that works at least thru the two days to a demo tutorial: nvcr.io/nvidia/digits:19.10-caffe (the py3 versions have the csv parsing problem, and the tensorflow ones raise errors when trying to train any model, be it classification or detection.
This one, can work thru all 4 steps: preparing dataset for classification, training the classification network, preparing the detection dataset and training the detection network.
Hope it helps.

closed 2023-04-27 by dusty-nv