dusty-nv/jetson-inference #521: install-pytorch.sh does not work.

ZiCog, 2020-03-09:
I have a Jetson Nano with the OS installed from nv-jetson-nano-sd-card-image-r32.3.1.zip

As a newbie to Jetson and Python I started working through the Hello AI World pages and all was going well until I came to the "Transfer Learning with PyTorch". Running the "install-pytorch.sh" and selecting "PyTorch c1.1.0 for Pyhon 3.6" seemed to work OK. But the installed torchvision does not work.
```
$ python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> print(torch.__version__)
1.1.0
>>> print('CUDA available: ' + str(torch.cuda.is_available()))
CUDA available: True
>>> a = torch.cuda.FloatTensor(2).zero_()
>>> a = torch.cuda.FloatTensor(2).zero_()
>>> print('Tensor a = ' + str(a))
Tensor a = tensor([0., 0.], device='cuda:0')
>>> b = torch.randn(2).cuda()
>>> print('Tensor b = ' + str(b))
Tensor b = tensor([-0.0116, -0.2358], device='cuda:0')
>>> c = a + b
>>> print('Tensor c = ' + str(c))
Tensor c = tensor([-0.0116, -0.2358], device='cuda:0')
>>> import torchvision
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.3.0-py3.6-linux-aarch64.egg/torchvision/__init__.py", line 2, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.3.0-py3.6-linux-aarch64.egg/torchvision/datasets/__init__.py", line 9, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.3.0-py3.6-linux-aarch64.egg/torchvision/datasets/fakedata.py", line 3, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.3.0-py3.6-linux-aarch64.egg/torchvision/transforms/__init__.py", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.3.0-py3.6-linux-aarch64.egg/torchvision/transforms/transforms.py", line 17, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.3.0-py3.6-linux-aarch64.egg/torchvision/transforms/functional.py", line 5, in <module>
ImportError: cannot import name 'PILLOW_VERSION'
>>> 
```

I subsequently tried to install PyTorch v1.4.0 from the announcement post on the forum here: https://devtalk.nvidia.com/default/topic/1049071/jetson-nano/pytorch-for-jetson-nano-version-1-4-0-now-available/1

That also failed, sadly I did not record the error.

So how does one install PyTorch on  Nano with the current OS such that one can complete the inference exercises?

Can the document be updated to describe whatever installation process does work?

dusty-nv, 2020-03-09:
Hi @ZiCog, can you try running `pip3 install 'pillow<7'` and then see if you can `import torchvision` ?

Here is the PyTorch bug about this:  https://github.com/pytorch/vision/issues/1712

ZiCog, 2020-03-09:
Bingo! That did it.
```
$ pip3 install "pillow<7"
Defaulting to user installation because normal site-packages is not writeable
Collecting pillow<7
  Downloading Pillow-6.2.2.tar.gz (37.8 MB)
     |████████████████████████████████| 37.8 MB 6.4 kB/s 
Building wheels for collected packages: pillow
  Building wheel for pillow (setup.py) ... done
  Created wheel for pillow: filename=Pillow-6.2.2-cp36-cp36m-linux_aarch64.whl size=966497 sha256=d06570fe4c6adf508f1c4a3766f545ff62d77f7628e164733235e37faaf5d87e
  Stored in directory: /home/michael/.cache/pip/wheels/19/4f/2e/b77ea60ebf837f5b06ed497d4b6798c3e4da683def2ad65ca2
Successfully built pillow
Installing collected packages: pillow
Successfully installed pillow-6.2.2
michael@jetson-nano:~/jetson-inference/build$ python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torchvision
>>> print(torchvision.__version__)
0.3.0
>>> 
```
Thanks.

dusty-nv, 2020-03-09:
OK thanks, I just patched the `install-pytorch.sh` script in master for this - see commit [`83f2ae`](https://github.com/dusty-nv/jetson-inference/commit/83f2ae1c1071f7cedb1f8629107404d269de8c90)

I also updated the manual install instructions for torchvision [here](https://devtalk.nvidia.com/default/topic/1049071/jetson-nano/pytorch-for-jetson-nano-version-1-4-0-now-available/).

ZiCog, 2020-03-09:
Great.

As I said the instructions for installing PyTorch v1.4.0 on that forum page do not work on the Nano. But I guess that is another story.

Anyway the cat-dog training is now underway here.

Thanks again.

flurpo, 2020-06-02:
Dusty-nv,

Thanks for the update. I'm running Jetpack 4.3 on my TX2. I ran the wheel files to get pytorch installed on python 2.7 and then python 3.6.9. I seem to be able to import torch and torchvision fine under python 2.7 but I'm only able to import torch on python 3.6.9, no torchvision.

Jetpack 4.3
Ubuntu 18.04
Torch 1.4.0 (+libopenblas-base)

>>import torchvision
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No Module named 'torchvision'
>>

When trying to install torchvision on the string above my installation exits with;
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

Anything jump out here? I'm so close. Hate to have to rebuild this again. Best,

dusty-nv, 2020-06-02:
> When trying to install torchvision per the link in your message above my installation exits with;
> 
> error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

Hmm were there any other errors listed while building torchvision?   Typically you would see a fuller error log above a message like this.

If you are building torchvision for Python 2 and then Python 3, you may need a `sudo python setup.py clean` in between.  And then for Python 3, remember to use `sudo python3 setup.py install` (instead of python2)

flurpo, 2020-06-02:
Wow...thanks so much for quick response dusty-nv- I've been stepping through setup.py all morning. I suspect something in my branch.

I used your setup string;

$ sudo apt-get install libjpeg-dev zlib1g-dev
$ git clone --branch build/v0.5.0 https://github.com/pytorch/vision torchvision
$ cd torchvision
$ sudo python3 setup.py install

Unfortuneately, the log is several 1000 lines long so I appeviated it a bit to show the recycling error;

tx2:/jetson-inference/torchvision$ sudo python setup.py clean
Building wheel torchvision-0.5.0
running clean
tx2:/jetson-inference/torchvision$ sudo python3 setup.py install
Building wheel torchvision-0.5.0
running install
running bdist_egg
running egg_info
creating torchvision.egg-info
writing torchvision.egg-info/PKG-INFO
writing dependency_links to torchvision.egg-info/dependency_links.txt
writing requirements to torchvision.egg-info/requires.txt
writing top-level names to torchvision.egg-info/top_level.txt
writing manifest file 'torchvision.egg-info/SOURCES.txt'
reading manifest file 'torchvision.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no previously-included files matching '__pycache__' found under directory '*'
warning: no previously-included files matching '*.py[co]' found under directory '*'
writing manifest file 'torchvision.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-aarch64/egg
running install_lib
running build_py
creating build
creating build/lib.linux-aarch64-3.6
creating build/lib.linux-aarch64-3.6/torchvision
copying torchvision/extension.py -> build/lib.linux-aarch64-3.6/torchvision
copying torchvision/__init__.py -> build/lib.linux-aarch64-3.6/torchvision
copying torchvision/utils.py -> build/lib.linux-aarch64-3.6/torchvision
copying torchvision/version.py -> build/lib.linux-aarch64-3.6/torchvision
creating build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/alexnet.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/googlenet.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/densenet.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/shufflenetv2.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/__init__.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/utils.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/resnet.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/inception.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/mnasnet.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/_utils.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/mobilenet.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/squeezenet.py -> build/lib.linux-aarch64-3.6/torchvision/models
copying torchvision/models/vgg.py -> build/lib.linux-aarch64-3.6/torchvision/models
creating build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/boxes.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/deform_conv.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/_register_onnx_ops.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/ps_roi_align.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/roi_align.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/__init__.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/roi_pool.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/new_empty_tensor.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/_utils.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/feature_pyramid_network.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/misc.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/ps_roi_pool.py -> build/lib.linux-aarch64-3.6/torchvision/ops
copying torchvision/ops/poolers.py -> build/lib.linux-aarch64-3.6/torchvision/ops
creating build/lib.linux-aarch64-3.6/torchvision/datasets
copying torchvision/datasets/cifar.py -> build/lib.linux-aarch64-3.6/torchvision/datasets
[... 77 lines cut ...]
creating build/temp.linux-aarch64-3.6/jetson-inference/torchvision/torchvision/csrc/cuda
aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DWITH_CUDA -I/jetson-inference/torchvision/torchvision/csrc -I/home/markw/.local/lib/python3.6/site-packages/torch/include -I/home/markw/.local/lib/python3.6/site-packages/torch/include/torch/csrc/api/include -I/home/markw/.local/lib/python3.6/site-packages/torch/include/TH -I/home/markw/.local/lib/python3.6/site-packages/torch/include/THC -I/usr/local/cuda/include -I/usr/include/python3.6m -c /jetson-inference/torchvision/torchvision/csrc/vision.cpp -o build/temp.linux-aarch64-3.6/jetson-inference/torchvision/torchvision/csrc/vision.o -O0 -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=1 -std=c++11
In file included from /home/markw/.local/lib/python3.6/site-packages/torch/include/ATen/core/boxing/KernelFunction.h:5:0,
[... cut ...] packages/torch/include/torch/csrc/api/include -I/home/markw/.local/lib/python3.6/site-packages/torch/include/TH -I/home/markw/.local/lib/python3.6/site-packages/torch/include/THC -I/usr/local/cuda/include -I/usr/include/python3.6m -c /jetson-inference/torchvision/torchvision/csrc/cpu/PSROIPool_cpu.cpp -o build/temp.linux-aarch64-3.6/jetson-inference/torchvision/torchvision/csrc/cpu/PSROIPool_cpu.o -O0 -DTORCH_API_INCLUDE_EXTENSION_H -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=1 -std=c++11

dusty-nv, 2020-06-02:
Hmm I don't actually see a build error in there.  Are you sure the board isn't running out of memory during compilation?

You could keep an eye on `sudo tegratstats` to monitor the memory usage, or mount additional swap.

flurpo, 2020-06-02:
Thanks for the quick replies dusty-nv....I'll go use tegra to make sure I'm not running outta room.

I added the end of installation log below;

                                                                                                                                                                                                                                                                                                                                                                                                                                                         ^
/home/markw/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:322:1: note: declared here
   T * data() const {
 ^ ~~
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIPool_cuda.cu:249:795: warning: ‘T* at::Tensor::data() const [with T = int]’ is deprecated [-Wdeprecated-declarations]
   AT_DISPATCH_FLOATING_TYPES_AND_HALF(
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           ^
/home/markw/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:322:1: note: declared here
   T * data() const {
 ^ ~~
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIPool_cuda.cu:249:918: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated [-Wdeprecated-declarations]
   AT_DISPATCH_FLOATING_TYPES_AND_HALF(
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                      ^
/home/markw/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:322:1: note: declared here
   T * data() const {
 ^ ~~
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIPool_cuda.cu:249:957: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated [-Wdeprecated-declarations]
   AT_DISPATCH_FLOATING_TYPES_AND_HALF(
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                             ^
/home/markw/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:322:1: note: declared here
   T * data() const {
 ^ ~~
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIPool_cuda.cu: In lambda function:
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIPool_cuda.cu:249:1194: warning: ‘T* at::Tensor::data() const [with T = c10::Half]’ is deprecated [-Wdeprecated-declarations]
   AT_DISPATCH_FLOATING_TYPES_AND_HALF(
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               [... cut ...]
In file included from /jetson-inference/torchvision/torchvision/csrc/cpu/video_reader/FfmpegDecoder.h:6:0,
                 from /jetson-inference/torchvision/torchvision/csrc/cpu/video_reader/VideoReader.cpp:6:
/jetson-inference/torchvision/torchvision/csrc/cpu/video_reader/FfmpegHeaders.h:4:10: fatal error: libavcodec/avcodec.h: No such file or directory
 #include <libavcodec/avcodec.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

dusty-nv, 2020-06-02:
>/jetson-inference/torchvision/torchvision/csrc/cpu/video_reader/FfmpegHeaders.h:4:10: fatal error: libavcodec/avcodec.h: No such file or directory
#include <libavcodec/avcodec.h>

I see, you must have `ffmpeg` executable tool installed on your Jetson, but not the libraries.

It looks like if torchvision finds the `ffmpeg` executable, it assumes the libraries are also installed:  https://github.com/pytorch/vision/blob/cca0c77a9ac5aa782b0811d850f246d73b0b4a1b/setup.py#L130

Try running `sudo apt-get install libavcodec-dev` first.

flurpo, 2020-06-02:
Thanks so much dusty-nv....

I ran the install above and ran clean.....still getting the same error below. I'm looking into the tag you gave me above. Best,

                                                                                                        ^
/home/markw/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:322:1: note: declared here
   T * data() const {
 ^ ~~
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIAlign_cuda.cu:415:999: warning: ‘T* at::Tensor::data() const [with T = float]’ is deprecated [-Wdeprecated-declarations]
   AT_DISPATCH_FLOATING_TYPES_AND_HALF(
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                       ^
/home/markw/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:322:1: note: declared here
   T * data() const {
 ^ ~~
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIAlign_cuda.cu: In lambda function:
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIAlign_cuda.cu:415:1241: warning: ‘T* at::Tensor::data() const [with T = c10::Half]’ is deprecated [-Wdeprecated-declarations]
   AT_DISPATCH_FLOATING_TYPES_AND_HALF(
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                         ^
/home/markw/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:322:1: note: declared here
   T * data() const {
 ^ ~~
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIAlign_cuda.cu:415:1274: warning: ‘T* at::Tensor::data() const [with T = int]’ is deprecated [-Wdeprecated-declarations]
   AT_DISPATCH_FLOATING_TYPES_AND_HALF(
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          ^
/home/markw/.local/lib/python3.6/site-packages/torch/include/ATen/core/TensorBody.h:322:1: note: declared here
   T * data() const {
 ^ ~~
/jetson-inference/torchvision/torchvision/csrc/cuda/PSROIAlign_cuda.cu:415:1417: warning: ‘T* at::Tensor::data() const [with T = c10::Half]’ is deprecated [-Wdeprecated-declarations]
   AT_DISPATCH_FLOATING_TYPES_AND_HALF(
[... cut ...]
In file included from /jetson-inference/torchvision/torchvision/csrc/cpu/video_reader/FfmpegDecoder.h:6:0,
                 from /jetson-inference/torchvision/torchvision/csrc/cpu/video_reader/VideoReader.cpp:6:
/jetson-inference/torchvision/torchvision/csrc/cpu/video_reader/FfmpegHeaders.h:5:10: fatal error: libavformat/avformat.h: No such file or directory
 #include <libavformat/avformat.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1

flurpo, 2020-06-02:
1000 thanks dusty-nv....not sure why but I'm successfully importing torchvision with python3...I believe problem solved!!!!!!!!!!!!!!!

flurpo, 2020-06-02:
ah.....looks like torchvision is successfully importing but torchvision sub calls are not working. I'll continue with the FFMPEG libraries

flurpo, 2020-06-02:
dusty-nv, thanks so much.....its working fine.....I'm able to get to all the libraries, working well

I simply removed ffmpeg from my system, ran the clean and the torchvision installation worked perfectly as described in your string. 

I hope this will help someone else. Maybe bypass the ffmpeg loop in that installation call?

Best wishes, MW

closed 2023-04-27 by dusty-nv