dusty-nv/jetson-inference #524: Can not run RPi camera at full resolution

badVibes--, 2020-03-12:
Hello, i tried running the detectnet-camera example with the camera at 3264x2464 but it doesn't want to work. 

I get this error:
(Argus) Error InvalidState:  (propagating from src/eglstream/FrameConsumerImpl.cpp, function streamEventThread(), line 135)
(Argus) Error InvalidState:  (propagating from src/eglstream/FrameConsumerImpl.cpp, function streamEventThreadStatic(), line 177)
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:654 Frame Rate specified is greater than supported

Its strange because the camera woks fine at this res with the example from here (it captures with opencv):
https://github.com/JetsonHacksNano/CSI-Camera

I want full resolution so I get the image without cropping (really wish it supported some smaller uncropped modes)

Thank you for any help with this.

dusty-nv, 2020-03-12:
I havent seen rPI camera do video at 4K, I would recommend using a lower resolution.  This shouldn't impact cropping as the lower resolutions have same aspect ratio and FoV.  The input to the networks gets downsampled to a resolution like 224x224 anyways, so 4K is not really a help for DNN.

https://github.com/JetsonHacksNano/CSI-Camera

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

dusty-nv, 2020-03-12:
Also from the GStreamer error above, it seems like the requested framerate is too high for 4K mode, maybe that is the problem.

https://github.com/JetsonHacksNano/CSI-Camera

badVibes--, 2020-03-12:
Thank you for your quick response. 

> I havent seen rPI camera do video at 4K

As i mentioned it works at  3264x2464  when i run the example from the link above.  I would like to set a lower framerate but have not yet figured out how I can do that. Do you have any ideas?

From the Jetson hacks link:
$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'RG10'
	Name        : 10-bit Bayer RGRG/GBGB
		Size: Discrete 3280x2464
			Interval: Discrete 0.048s (21.000 fps)
		Size: Discrete 3280x1848
			Interval: Discrete 0.036s (28.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)

all the other supported resolutions are not 4:3 ratio so they crop the bottom and top parts of the image. The camera supposedly supports one more 4:3 resolution (1640x1232 binned 4:3, up 40fps) but I guess the driver doesn't. https://www.raspberrypi.org/forums/viewtopic.php?t=157384

badVibes--, 2020-03-13:
Well i found how to change in the frame rate but it didn't really help. The error went away but I guess other parts just can't handle the bigger res. It tries to do it but the image never gets displayed and then it just quits. 
I guess it could do it if it displayed the scaled down image instead of the full res one but figuring that out is over my head. In any case, thanks for your help:)

badVibes--, 2020-03-13:
Well i guess my problems would be solved if i could somehow scale down the image right after capture and before passing it to the detect routine. Can you point me at something i could use to do that efficiently? It would really help my project if the robot im making could see under its feet/wheels. I bought a wide lens for that purpose but the cropping really narrows down the field of view :(. The robot is an art project on a really limited budget so i can't afford buy a more suitable camera. It's a drunk robot using three networks. One for detecting people so it can pee on them, one for avoiding obstacles and the last to make up lyrics for its drunken singing. On the first iteration I made it with jetbot code but with the update that got broken and i would like to upgrade it to the faster code from this repo. Some images of the robot:http://www.cirkulacija2.org/?p=6204

closed 2023-04-27 by dusty-nv