dusty-nv/jetson-inference #528: External Power Supply for Jetson Nano

achen173, 2020-03-18:
I have been using the barrel jack for power supply (5V 4A) and it works perfectly. Now I am trying to transition to a portable power source. Specifically I bought this Power Supply (https://www.amazon.com/gp/product/B07H87HKKM/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) which has an Output of 5V-2.1A. The video feed stop working, and I figured it was because of the power supply. Is there any good portable power supply you recommend?

Alex-Beh, 2020-12-04:
@achen173 Do you have the solution for the portable power?

achen173, 2020-12-05:
@Alex-Beh Anything with 5V and 4 Amps will work. The ampere is the problem since most voltage at that level wont supply that much current on a portable device. It turns out that you can set the max current but that will affect the performance. Long story short I bought "Krisdonia Portable Laptop Charger (TSA-Approved) 92.5Wh/25000mAh" and it works. However, I recommend you look into batteries more along the lines of those specification

closed 2023-04-27 by dusty-nv