dusty-nv/jetson-inference #532: USB Camera - Logitech C920 = Jetsoin Nano

leandrovrabelo, 2020-03-23:
I'm trying to use CV2 with detecNet and i'm having some problems when I try to use my USB Camera, it's a Logitech C920.

When I use this code in the command line it works normally:

`gst-launch-1.0 -v v4l2src device=/dev/video1 ! video/x-raw,framerate=30/1,width=640,height=480 ! xvimagesink`

and when I create this pipeline below just using CV2, it works pretty well:

`import cv2
from datetime import datetime

def gstreamer_pipeline(capture_width=640, capture_height=360, display_width=640, display_height=360, framerate=30):

    return f'v4l2src device=/dev/video1 ! video/x-raw, width=(int){capture_width}, height=(int){capture_height}, format=(string)YUY2, framerate=(fraction){framerate}/1 ! videoconvert ! video/x-raw, width=(int){display_width}, height=(int){display_height}, format=(string)RGB ! videoconvert ! appsink'

cam = cv2.VideoCapture(gstreamer_pipeline(), cv2.CAP_GSTREAMER)

while True:

    ret, frame = cam.read()

    cv2.imshow('frame',frame)

    # close the camera
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

cam.release()
cv2.destroyAllWindows()`

But when I try to use the same pipeline in an adaptation to the detectnet-camera.py it shows this message:

`[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
PyTensorNet_Dealloc()
`
Can anyone hellp me? By the way, i'm using a CSI Camera, but it's not activated when I try to activate my USB Camera.

AndreyIsak, 2020-03-27:
Hi leandrovrabelo you can check your video port ?
device=/dev/video1 ? can you change it to device=/dev/video0 ? and try to run it

leandrovrabelo, 2020-03-29:
I have a CSI Camera allocated on /dev/video0 and I still couldn't solve this.

When I use this pipeline without DetectNet, the camera opens normally but this message appears on the terminal:
`[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1`

Does it give some clue?

leandrovrabelo, 2020-04-01:
Instead of creating a hole pipeline I put '/dev/video1' in the cv2.VideoCapture() and it worked, then I set the others video aspects and it worked properly.

closed 2020-04-01 by leandrovrabelo

bl6594, 2020-04-03:
leandrovrabelo,
would you please share your code on how you made cv2 to work with detectnet?  
Thanks!

leandrovrabelo, 2020-04-03:
Sure, see below:

```
import cv2
import jetson.utils
import jetson.inference

model = jetson.inference.detectNet()
width = 640
height = 320

cam = cv2.VideoCapture('/dev/video1')
cam.set(cv2.CAP_PROP_FRAME_WIDTH, width )
cam.set(cv2.CAP_PROP_FRAME_HEIGHT, height )

# Opening the camera
while cam.isOpened():

    ret, frame = cam.read()
    frame_rgba = cv2.cvtColor(frame, cv2.COLOR_BGR2RGBA).astype(np.float32)

    img = jetson.utils.cudaFromNumpy(frame_rgba)

    detections = model.Detect(img, width, height, "box,labels,conf")
    conv1 = jetson.utils.cudaToNumpy(img, width, height, 4)        
    conv2 = cv2.cvtColor(conv1, cv2.COLOR_RGBA2RGB).astype(np.uint8)
    conv3 = cv2.cvtColor(conv2, cv2.COLOR_RGB2BGR)
```

If you want to use the image without bounding boxes you use `frame` variable, but if you want the image with it you will use `conv3`.