dusty-nv/jetson-inference #538: CMake binding error on jetson nano

mahdi1978, 2020-04-05:
Hello, i try to build the Hello AI on virtualenv, and CMake produce this error, can you help me ?

`-- trying to build Python bindings for Python versions:  2.7;3.6;3.7
-- detecting Python 2.7...
-- found Python version:  3.6 (3.6.9)
-- found Python include:  /usr/include/python3.6m
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.6m.so
-- CMake module path:  /home/mahdi/jetson-inference/utils/cuda;/home/mahdi/jetson-inference/utils/python/bindings
-- NumPy ver. 1.18.2 found (include: /home/mahdi/.virtualenvs/dlcv/lib/python3.6/site-packages/numpy/core/include)
-- found NumPy version:  1.18.2
-- found NumPy include:  /home/mahdi/.virtualenvs/dlcv/lib/python3.6/site-packages/numpy/core/include
-- detecting Python 3.6...
-- found Python version:  3.6 (3.6.9)
-- found Python include:  /usr/include/python3.6m
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.6m.so
-- CMake module path:  /home/mahdi/jetson-inference/utils/cuda;/home/mahdi/jetson-inference/utils/python/bindings
-- NumPy ver. 1.18.2 found (include: /home/mahdi/.virtualenvs/dlcv/lib/python3.6/site-packages/numpy/core/include)
-- found NumPy version:  1.18.2
-- found NumPy include:  /home/mahdi/.virtualenvs/dlcv/lib/python3.6/site-packages/numpy/core/include
CMake Error at utils/cuda/FindCUDA.cmake:1802 (add_library):
  add_library cannot create target "jetson-utils-python-36" because another
  target with the same name already exists.  The existing target is a shared
  library created in source directory
  "/home/mahdi/jetson-inference/utils/python/bindings".  See documentation
  for policy CMP0002 for more details.
Call Stack (most recent call first):
  utils/python/bindings/CMakeLists.txt:57 (cuda_add_library)

CMake Error at utils/python/bindings/CMakeLists.txt:59 (target_link_libraries):
  Attempt to add link library "jetson-utils" to target
  "jetson-utils-python-36" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

CMake Error at utils/python/bindings/CMakeLists.txt:62 (target_link_libraries):
  Attempt to add link library "npymath" to target "jetson-utils-python-36"
  which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

-- detecting Python 3.7...
-- Python 3.7 wasn't found
-- Copying /home/mahdi/jetson-inference/utils/python/examples/camera-viewer.py
-- Copying /home/mahdi/jetson-inference/utils/python/examples/cuda-from-numpy.py
-- Copying /home/mahdi/jetson-inference/utils/python/examples/cuda-to-numpy.py
-- Copying /home/mahdi/jetson-inference/utils/python/examples/gl-display-test.py
-- trying to build Python bindings for Python versions:  2.7;3.6;3.7
-- detecting Python 2.7...
-- found Python version:  3.6 (3.6.9)
-- found Python include:  /usr/include/python3.6m
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.6m.so
-- detecting Python 3.6...
-- found Python version:  3.6 (3.6.9)
-- found Python include:  /usr/include/python3.6m
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.6m.so
CMake Error at utils/cuda/FindCUDA.cmake:1802 (add_library):
  add_library cannot create target "jetson-inference-python-36" because
  another target with the same name already exists.  The existing target is a
  shared library created in source directory
  "/home/mahdi/jetson-inference/python/bindings".  See documentation for
  policy CMP0002 for more details.
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:41 (cuda_add_library)

CMake Error at python/bindings/CMakeLists.txt:43 (target_link_libraries):
  Attempt to add link library "jetson-inference" to target
  "jetson-inference-python-36" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

-- detecting Python 3.7...
-- Python 3.7 wasn't found
-- Copying /home/mahdi/jetson-inference/python/examples/detectnet-camera.py
-- Copying /home/mahdi/jetson-inference/python/examples/detectnet-console.py
-- Copying /home/mahdi/jetson-inference/python/examples/imagenet-camera.py
-- Copying /home/mahdi/jetson-inference/python/examples/imagenet-console.py
-- Copying /home/mahdi/jetson-inference/python/examples/my-detection.py
-- Copying /home/mahdi/jetson-inference/python/examples/my-recognition.py
-- Copying /home/mahdi/jetson-inference/python/examples/segnet-batch.py
-- Copying /home/mahdi/jetson-inference/python/examples/segnet-camera.py
-- Copying /home/mahdi/jetson-inference/python/examples/segnet-console.py
-- Linking jetson-inference with OpenCV 4.3.0
-- Configuring incomplete, errors occurred!
See also "/home/mahdi/jetson-inference/build/CMakeFiles/CMakeOutput.log".
See also "/home/mahdi/jetson-inference/build/CMakeFiles/CMakeError.log".`

closed 2020-05-03 by mahdi1978

muhammadabdullah34907, 2020-05-07:
Hi @mahdi1978 

Did you find the solution to above problem ? Please share with me if yes. I am having same problem after upgrading to ubuntu 20.04 with python3.7 and CUDA 10.2

mahdi1978, 2020-05-08:
Hello, Sorry for late reply, unfortuntely i forgot the step how to fix that, but i manage to get it work, if memory serves i reinstall the python 2.7 and install some libaries and modules needed (should check your error log), because most of the code still using python 2.7 library, so basically i’m not using python 3.6, But i’m also confuse, because i manage to run it smoothly using python 3.6 before by following this tutorial. BTW, i’m on fresh installation now so i can’t help you reproduce the error, here what i remember, the problem persist after i remove the python 2.7 completely and i’m using cmake 3.17.0 at that time. As for my new fresh installation i haven’t re-install jetson-inference. Hope this help. Best regard,Mahdi. PS: I’m using Jetpack 4.3 (based on ubuntu 18.04 arm64) on Jetson Nano Dev Kit, if you are using CUDA 10.2 (only supported on newer Jetpack)i assume you are using Jetpack 4.4 Developer Preview ?, just curious, ignore me if i’m wrong. Dikirim dari Email untuk Windows 10 Dari: muhammadabdullah34907Dikirim: Kamis, 07 Mei 2020 20.30Kepada: dusty-nv/jetson-inferenceCc: mahdi1978; MentionSubjek: Re: [dusty-nv/jetson-inference] CMake binding error on jetson nano (#538) Hi @mahdi1978Did you find the solution to above problem ? Please share with me if yes. I am having same problem after upgrading to ubuntu 20.04 with python3.7 and CUDA 10.2