dusty-nv/jetson-inference #544: How do I use object detection on a video ?

AndreyIsak, 2020-04-14:
I have a video of couple of objects and I would like to know how do I use object detection on a video.
Now I am using object detection model ResNet-18 with c920 logitech camera.

leandrovrabelo, 2020-04-14:
Are you using a pre recorded video or do you want to use your web cam? The code below is for using the C920 web cam:

```
import cv2
import jetson.utils
import jetson.inference

model = jetson.inference.detectNet()
width = 640
height = 320

cam = cv2.VideoCapture('/dev/video1')
cam.set(cv2.CAP_PROP_FRAME_WIDTH, width )
cam.set(cv2.CAP_PROP_FRAME_HEIGHT, height )

# Opening the camera
while cam.isOpened():

    ret, frame = cam.read()
    frame = cv2.flip(frame, 1)
    frame_rgba = cv2.cvtColor(frame, cv2.COLOR_BGR2RGBA).astype(np.float32)

    img = jetson.utils.cudaFromNumpy(frame_rgba)

    detections = model.Detect(img, width, height, "box,labels,conf")
    conv1 = jetson.utils.cudaToNumpy(img, width, height, 4)        
    conv2 = cv2.cvtColor(conv1, cv2.COLOR_RGBA2RGB).astype(np.uint8)
    conv3 = cv2.cvtColor(conv2, cv2.COLOR_RGB2BGR)

    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

cam.release()
cv2.destroyAllWindows()
```

If you want to use the image without bounding boxes you use frame variable, but if you want the image with it you will use conv3.

if you want to load a video that you already have just put the link in `cv2.VideoCapture('LINK HERE')`

AndreyIsak, 2020-04-14:
I am using pre recorded video from the camera

AndreyIsak, 2020-04-15:
I just tried this code leandrovrabelo and this is not working because you are using np.uint8 
this kinda of variable is not in the program so how it should run ? 
the error is np is not defined

AndreyIsak, 2020-04-15:
I found out that there is 1 line missing for some reason should be import numpy as np :)
by the way can i see live object detection on a video ? 
when i will run a program the program will open a window that i can see live object detection from the video ?

leandrovrabelo, 2020-04-15:
The code above works with the web cam.

use the details below instead of inserting the video link:

cam = cv2.VideoCapture('/dev/video1')

to confirm which device you should use open a terminal and paste the following code:

v4l2-ctl --list-devices

It will show the available cameras.

AndreyIsak, 2020-04-16:
I used your code as it is and I get this error : np is not defined

ynjiun, 2020-04-21:
> I used your code as it is and I get this error : np is not defined
perhaps you need to:
`import numpy as np`

KarimJedda, 2020-09-10:
Thanks @leandrovrabelo finally something that works :)

dusty-nv, 2020-09-10:
If you update or re-clone your repo from the latest master and re-run `make` and `sudo make install`, support for video playback/recording was also [added](https://github.com/dusty-nv/jetson-inference/blob/master/CHANGELOG.md#july-15-2020) earlier this summer.

You should now be able to do something like:

``` bash
$ detectnet my_video_input.mp4 my_video_output.mp4
```

closed 2023-04-27 by dusty-nv