dusty-nv/jetson-inference #545: custom model object detection

elementdl, 2020-04-15:
Hi,

In the tutorial Hello AI World it's possible to use pre-trained models.
Let's say I want to train SSD-Mobilenet-v2 with specifics classes, how would it be possible to use the project jetson-inference to use my new model?

Sincerely,

dusty-nv, 2020-04-15:
Yes, you would need to re-train SSD-Mobilenet with TensorFlow as shown in some of these tutorials:

https://forums.developer.nvidia.com/t/digits-or-somthing-else/110790/7?u=dusty_nv

And then convert it to TensorRT UFF with this tool:   https://github.com/AastaNV/TRT_object_detection

I am also working on PyTorch/ONNX workflow for re-training SSD-Mobilenet which will simplify matters and can be trained onboard the Jetson.

elementdl, 2020-04-16:
Thanks a lot @dusty-nv!! I will close this issue, just one question, I want to use a CSI camera (raspberry pi V2), how can I change "my_detection.py" in order to use this camera? If I try the program the jetson crash.

closed 2020-04-17 by elementdl