dusty-nv/jetson-inference #547: What is recommended system configuration for Deep Learning?

spurani, 2020-04-19:
Basically I am having Acer Nitro 50 Desktop with system configuration
Processor: Intel® Core™ i5-8400 CPU @ 2.80GHz × 6
Graphics: GeForce GTX 1050/PCIe/SSE2 (2 GB)
Memory(RAM): 8GB DDR4 Memory

I am working with **caffe  | python 3.5 | GCC 4.8 | cudnn 7 | Cuda 9.0** to train a faster_rcnn_inception_v2 model on my custom dataset with 10.4 GB(65000 images) of training data and 533.4 MB(3333) of validation data for object detection for 600k epochs(num_steps it will be training at 640 x 640 resolution). I am training and validation model on 8 classes. So when I am training the model the accuracy(map) is not increasing and loss is not decreasing after a while. After the completion of the training successfully I ran the model on various images and noticed couple of things.
1)    Less accuracy
 2)   In some images objects does not get detected at all no bounding boxes. while in some multiclass object detection is spot on perfectly all objects gets detected but with less accuracy around 60 - 75.
3)   From second point above If I train a separate model with less images and less number of classes(3 or 4) it works well but with decent amount of accuracy around 75 - 95

Question 1) What would be ideal system to work comfortably with deep learning? and get desired amount of results

So based on my system configuration should I keep on training or invest into new system which will be lot faster and accurate in deep learning. It would be really helpful for me to continue my learning if anyone could suggest or recommend something

dusty-nv, 2020-04-20:
Hi @spurani, it is probably not related to your HW itself, but rather the network you are training or the dataset.  If you have been unsuccessful with Faster-RCNN, have you tried training with YOLO or SSD?   I have been using this for training SSD-Mobilenet with PyTorch:  https://github.com/dusty-nv/pytorch-ssd

spurani, 2020-04-23:
Hi @dusty-nv thanks a bunch for your quick response I was trying to make SSD and other faster rcnn to work on my system So starting with ssd I trained my custom dataset with image 640 x 640 batch_size 1 and num_steps 200k but the map was not increasing for a while and loss was not decreasing below 5 due to which I believe it didn't show even a single bounding box in image. while Faster_rcnn_resnet101_kitti max's out my memory even though image resolution is less and batch size is 1 as mentioned in my first post faster_rcnn_inception_v2_coco works but with less accuracy and inconsistency. May I please know in what direction I should work towards to continue my learning about deep learning? it is really interesting

spurani, 2020-04-30:
Hi @dusty-nv I have couple of questions
1) I am training my model on MS COCO dataset using pretrained faster_rcnn_inception_v2_coco in tensorflow. As the images in MS COCO dataset does not have same size do I need to resize before training change the image_resizer function in faster_rcnn_inception_v2_coco.config file(below)?

[image]
[image]

As you can see the above images my model is not training effectively with total loss and learning rate being inconsistent. Here is what I am using to train the model faster_rcnn_inception_v2_coco.config
````
# Faster R-CNN with Inception v2, configuration for MSCOCO Dataset.
# Users should configure the fine_tune_checkpoint field in the train config as
# well as the label_map_path and input_path fields in the train_input_reader and
# eval_input_reader. Search for "PATH_TO_BE_CONFIGURED" to find the fields that
# should be configured.
model {
  faster_rcnn {
    num_classes: 8
#Default configuration for image_resizer no changes made in this function
    image_resizer {
      keep_aspect_ratio_resizer {
        min_dimension: 600
        max_dimension: 1024
      }
    }
    feature_extractor {
      type: 'faster_rcnn_inception_v2'
      first_stage_features_stride: 16
    }
    first_stage_anchor_generator {
      grid_anchor_generator {
        scales: [0.25, 0.5, 1.0, 2.0]
        aspect_ratios: [0.5, 1.0, 2.0]
        height_stride: 16
        width_stride: 16
      }
    }
    first_stage_box_predictor_conv_hyperparams {
      op: CONV
      regularizer {
        l2_regularizer {
          weight: 0.0
        }
      }
      initializer {
        truncated_normal_initializer {
          stddev: 0.01
        }
      }
    }
    first_stage_nms_score_threshold: 0.0
    first_stage_nms_iou_threshold: 0.7
    first_stage_max_proposals: 300
    first_stage_localization_loss_weight: 2.0
    first_stage_objectness_loss_weight: 1.0
    initial_crop_size: 14
    maxpool_kernel_size: 2
    maxpool_stride: 2
    second_stage_box_predictor {
      mask_rcnn_box_predictor {
        use_dropout: false
        dropout_keep_probability: 1.0
        fc_hyperparams {
          op: FC
          regularizer {
            l2_regularizer {
              weight: 0.0
            }
          }
          initializer {
            variance_scaling_initializer {
              factor: 1.0
              uniform: true
              mode: FAN_AVG
            }
          }
        }
      }
    }
    second_stage_post_processing {
      batch_non_max_suppression {
        score_threshold: 0.0
        iou_threshold: 0.6
        max_detections_per_class: 100
        max_total_detections: 300
      }
      score_converter: SOFTMAX
    }
    second_stage_localization_loss_weight: 2.0
    second_stage_classification_loss_weight: 1.0
  }
}

train_config: {
  batch_size: 1
  optimizer {
    momentum_optimizer: {
      learning_rate: {
        manual_step_learning_rate {
          initial_learning_rate: 0.0002
          schedule {
            step: 900000
            learning_rate: .00002
          }
          schedule {
            step: 1200000
            learning_rate: .000002
          }
        }
      }
      momentum_optimizer_value: 0.9
    }
    use_moving_average: false
  }
  gradient_clipping_by_norm: 10.0
  fine_tune_checkpoint: "models/model.ckpt"
  from_detection_checkpoint: true
  # Note: The below line limits the training process to 200K steps, which we
  # empirically found to be sufficient enough to train the COCO dataset. This
  # effectively bypasses the learning rate schedule (the learning rate will
  # never decay). Remove the below line to train indefinitely.
  num_steps: 200000
  data_augmentation_options {
    random_horizontal_flip {
    }
  }
}

train_input_reader: {
  tf_record_input_reader {
    input_path: "data/dataset_tools_train.tfrecord"
  }
  label_map_path: "data/label.pbtxt"
}

eval_config: {
  num_examples: 3300
  # Note: The below line limits the evaluation process to 10 evaluations.
  # Remove the below line to evaluate indefinitely.
  max_evals: 10
}

eval_input_reader: {
  tf_record_input_reader {
    input_path: "data/dataset_tools_val.tfrecord"
  }
  label_map_path: "data/dataset_tools_val.tfrecord"
  shuffle: false
  num_readers: 1
} 
```

closed 2023-04-27 by dusty-nv