dusty-nv/jetson-inference #549: [TRT] ... Caffe Parser: Invalid axis in softmax layer

ynjiun, 2020-04-21:
Hi 
I am trying a custom caffe .prototxt which includes the following layers:

```
layer {
  name: "mbox_conf_reshape"
  type: "Reshape"
  bottom: "mbox_conf"
  top: "mbox_conf_reshape"
  reshape_param {
    shape {
      dim: 0
      dim: -1
      dim: 3
    }
  }
}
layer {
  name: "mbox_conf_softmax"
  type: "Softmax"
  bottom: "mbox_conf_reshape"
  top: "mbox_conf_softmax"
  softmax_param {
    axis: 2
  }
}
layer {
  name: "mbox_conf_flatten"
  type: "Flatten"
  bottom: "mbox_conf_softmax"
  top: "mbox_conf_flatten"
  flatten_param {
    axis: 1
  }
}
```

I was able to test the network in Digits. But when port to TensorRT in Jetson Nano (JetPack 4.3), it keeps complaining:

```
[TRT]   mbox_loc: all concat input tensors must have the same dimensions except on the concatenation axis (0), but dimensions mismatched at input 2 at index 1. Input 0 shape: [12,39,16], Input 2 shape: [6,20,24]
[TRT]   mbox_conf: all concat input tensors must have the same dimensions except on the concatenation axis (0), but dimensions mismatched at input 2 at index 1. Input 0 shape: [12,39,12], Input 2 shape: [6,20,18]
Caffe Parser: Invalid axis in softmax layer - TensorRT expects NCHW input. Negative axis is not supported in TensorRT, please use positive axis indexing
error parsing layer type Softmax index 230
```

I wonder what shall I modify either the reshape layer or the softmax layer to fix this Caffe Parser invalid axis in softmax layer erro. Please advise.

Thanks a lot for your help.

-Paul

ynjiun, 2020-04-21:
I am able to workaround the "Flatten" layers by using "Reshape" layers to flatten them:

`reshape_param { shape { dim: 0 dim: -1 } } #for TensorRT flatten workaround`

closed 2020-04-21 by ynjiun

G2020sudo, 2020-07-24:
Can you share the file with your changes?