dusty-nv/jetson-inference #550: A network trained and tested on Digits successful but fail to load in Jetson Nano thru TensorRT

ynjiun, 2020-04-21:
Hi

I am able to train and test mypvdetector (see attached mypvdetector.prototxt in zip file below) in Digits. When I port the trained model from Digits to Jetson Nano through TensorRT, it shows the following error message:
```
[TRT]   device GPU, loading /home/Public/mypvdetector/deploy.prototxt /home//Public/mypvdetector/snapshot_iter_655.caffemodel
Plugin layer output count is not equal to caffe output count
[TRT]   device GPU, failed to parse caffe network
[TRT]   device GPU, failed to load /home/Public/mypvdetector/snapshot_iter_655.caffemodel
detectNet -- failed to initialize.
jetson.inference -- detectNet failed to load built-in network 'ssd-mobilenet-v2'
PyTensorNet_Dealloc()
Traceback (most recent call last):
  File "./vidar-detectnet-video.py", line 38, in <module>
    net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)
Exception: jetson.inference -- detectNet failed to load network
```
Is there a way to fix this error? What am I missing in the workflow? Thanks a lot for your help.

[mypvdetector.zip](https://github.com/dusty-nv/jetson-inference/files/4513086/mypvdetector.zip)

WhaSukGO, 2020-05-07:
Any update? I'm having the same issue!

hyattbaker, 2020-05-20:
Hmm this doesn't bode well, i was just about to start with Digits and Nano.

ynjiun, 2020-05-21:
So far I can only duplicate what already converted in this site. Anything else no luck. It seems all the tools only work for those already done on this site. Every custom trained network is a new adventure: for example, I am able to trained and run Object Detection API network such as ssdlite_mobilenet_v3_kitti with frozen.pb and try to port over to nano using tensorRT and encounter never ending errors (fix one another popup...) and now almost giving up after 3 weeks trial and error..... ; (

ynjiun, 2020-05-21:
@WhaSukGO : the reason that TRT complaining "[TRT]   device GPU, loading /home/Public/mypvdetector/deploy.prototxt /home//Public/mypvdetector/snapshot_iter_655.caffemodel
Plugin layer output count is not equal to caffe output count" is because on Digits Caffe the "detection_out" implement ONE output but when port to tensorRT, the tensorRT plugin implement "detection_out" layer with TWO outputs. I did fake it by adding a junk output in the port over detection_out graph but right after knockout this error, another error popup..... that's why I didn't bother to post my experiences... ; (

closed 2023-04-27 by dusty-nv