dusty-nv/jetson-inference #551: How to load custom trained model on my-detection.py

rasheeddo, 2020-04-22:
Hi, I have tried retraining a detectNet on DIGITS and it was success. I got the snapshot_iter_xxx.caffemodel, deploy.prototxt and other stuff. It works well when I tried
`
./detectnet-camera --model=/home/nvidia/testScript-jetson-inference/coco-person/snapshot_iter_903600.caffemodel --prototxt=/home/nvidia/testScript-jetson-inference/coco-person/deploy.prototxt --camera=/dev/video0 --output_bbox=bboxes --class_labels=/home/nvidia/testScript-jetson-inference/coco-person/classes.txt
`
it can detect as I expected. But then I want to use this on a python script according to your guide on [this](https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-example-2.md) 
I tried to point detectNet to my custom model path and prototxt file like this
```
prototxt_path = "/home/nvidia/testScript-jetson-inference/coco-person/deploy.prototxt"
model_path = "/home/nvidia/testScript-jetson-inference/coco-person/snapshot_iter_903600.caffemodel"

#net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
net = jetson.inference.detectNet(model=model_path, prototxt=prototxt_path, threshold=0.5)
```
but I got this error
```
File "my-detection.py", line 31, in <module>
    net = jetson.inference.detectNet(model=model_path, prototxt=prototxt_path, threshold=0.5)
Exception: jetson.inference -- detectNet.__init()__ failed to parse args tuple
```
 I am not sure how to properly pass the custom model on detectNet in the code.
 I checked on this [link](https://rawgit.com/dusty-nv/jetson-inference/python/docs/html/python/jetson.inference.html#detectNet), seems like I can pass model and prototxt to detectNet class, isn't it? 

Please guide me a bit, thank you.

rasheeddo, 2020-04-23:
Tried [this](https://github.com/dusty-nv/jetson-inference/issues/411#issuecomment-533473692), it works.

closed 2020-04-23 by rasheeddo

Hansansui-cv, 2021-06-20:
So, how do you sovle the problem? I meet the same problem. I need your help. Thank you.

dusty-nv, 2021-06-21:
@Hansansui-cv you can load custom detection model from python like this:

``` python
net = jetson.inference.detectNet(argv=['--model=models/your_model.onnx', '--labels=models/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes'])
```