dusty-nv/jetson-inference #554: Unable to convert resnet18 ONNX model to TensorRT Engine

mbufi, 2020-04-23:
Hello @dusty-nv 
I trained a sample resnet18/resnet50 model with the repo and successfully converted it to onnx format.

Then I wanted to use it in my code: 
```
net = jetson.inference.imageNet('googlenet',
['--model=resnet18.onnx',
'--input_blob=input_0',
'--output_blob=output_0',
'--labels=labels.txt'])
```

and was met with the following: 

> nvidia@nvidia:~/ai/algorithms/jetson_inference$ python3 imagenet-opencv.py 
> jetson.inference.__init__.py
> jetson.inference -- initializing Python 3.6 bindings...
> jetson.inference -- registering module types...
> jetson.inference -- done registering module types
> jetson.inference -- done Python 3.6 binding initialization
> jetson.utils.__init__.py
> jetson.utils -- initializing Python 3.6 bindings...
> jetson.utils -- registering module functions...
> jetson.utils -- done registering module functions
> jetson.utils -- registering module types...
> jetson.utils -- done registering module types
> jetson.utils -- done Python 3.6 binding initialization
> SYSTEM ARG ['imagenet-opencv.py']
> jetson.inference -- PyTensorNet_New()
> jetson.inference -- PyImageNet_Init()
> jetson.inference -- imageNet loading network using argv command line params
> jetson.inference -- imageNet.__init__() argv[0] = '--model=resnet18.onnx'
> jetson.inference -- imageNet.__init__() argv[1] = '--input_blob=input_0'
> jetson.inference -- imageNet.__init__() argv[2] = '--output_blob=output_0'
> jetson.inference -- imageNet.__init__() argv[3] = '--labels=labels.txt'
> 
> imageNet -- loading classification network model from:
>          -- prototxt     (null)
>          -- model        resnet18.onnx
>          -- class_labels labels.txt
>          -- input_blob   'input_0'
>          -- output_blob  'output_0'
>          -- batch_size   1
> 
> [TRT]   TensorRT version 5.1.6
> [TRT]   loading NVIDIA plugins...
> [TRT]   Plugin Creator registration succeeded - GridAnchor_TRT
> [TRT]   Plugin Creator registration succeeded - NMS_TRT
> [TRT]   Plugin Creator registration succeeded - Reorg_TRT
> [TRT]   Plugin Creator registration succeeded - Region_TRT
> [TRT]   Plugin Creator registration succeeded - PriorBox_TRT
> [TRT]   Plugin Creator registration succeeded - Normalize_TRT
> [TRT]   Plugin Creator registration succeeded - RPROI_TRT
> [TRT]   Plugin Creator registration succeeded - BatchedNMS_TRT
> [TRT]   Could not register plugin creator:  FlattenConcat_TRT in namespace: 
> [TRT]   Plugin Creator registration succeeded - CropAndResize
> [TRT]   Plugin Creator registration succeeded - Proposal
> [TRT]   Plugin Creator registration succeeded - BatchTilePlugin_TRT
> [TRT]   completed loading NVIDIA plugins.
> [TRT]   detected model format - ONNX  (extension '.onnx')
> [TRT]   desired precision specified for GPU: FASTEST
> [TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
> [TRT]   native precisions detected for GPU:  FP32, FP16, INT8
> [TRT]   selecting fastest native precision for GPU:  FP16
> [TRT]   attempting to open engine cache file resnet18.onnx.1.1.GPU.FP16.engine
> [TRT]   cache file not found, profiling network model on device GPU
> [TRT]   device GPU, loading /usr/bin/ resnet18.onnx
> ----------------------------------------------------------------
> Input filename:   resnet18.onnx
> ONNX IR version:  0.0.4
[... 144 lines cut ...]
> op_type: "Gather"
> attribute {
>   name: "axis"
>   i: 0
>   type: INT
> }
> 
> --- End node ---
> ERROR: onnx2trt_utils.hpp:277 In function convert_axis:
> [8] Assertion failed: axis >= 0 && axis < nbDims
> [TRT]   failed to parse ONNX model 'resnet18.onnx'
> [TRT]   device GPU, failed to load resnet18.onnx
> [TRT]   failed to load resnet18.onnx
> [TRT]   imageNet -- failed to initialize.
> jetson.inference -- imageNet failed to load built-in network 'googlenet'
> PyTensorNet_Dealloc()
> Traceback (most recent call last):
>   File "imagenet-opencv.py", line 66, in <module>
>     '--labels=labels.txt'])
> Exception: jetson.inference -- imageNet failed to load network
> double free or corruption (!prev)
> Aborted (core dumped)
> 

Do you know what could be the problem?

dusty-nv, 2020-04-23:
Hi, there is a patch for this in my torchvision fork, in the v0.3.0 branch.  You can either install my branch or cherry-pick the resnet fix for it.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

mbufi, 2020-04-23:
Thanks @dusty-nv ! Before I go off to attempt, is there a quick command for doing this? Or something in the tutorials that I missed?

closed 2020-04-24 by mbufi