dusty-nv/jetson-inference #556: Intel Realsens Cam - cudaFromNumpy issue

gregtyminski, 2020-04-24:
Dear Dustin Franklin,

I've connected Intel Realsense D435 cam to Jetson Nano and I'm trying to call 
```python
net.Detect(frame_gpu, width, height)
```
... on a frame from camera.

Entire pipeline works, however nothing is detected (no matter if there is any face in front of camera) due to issue with coping this color frame to CUDA memory with usage of
```python
frame_gpu = jetson.utils.cudaFromNumpy(frame)
```

When I save the frame with
```python
jetson.utils.saveImageRGBA(test_img_name, frame_gpu, width, height)
```
... I get really strange image.

[//]: # (Image Reference)
[saved-image]: https://github.com/gregtyminski/jetson-inference-with-intel-realsense/blob/master/saved_image.png "saved-image"
![alt text][saved-image]

You can find entire code here (with all results)
https://github.com/gregtyminski/jetson-inference-with-intel-realsense/blob/master/try-camera-facenet-tensorrt.ipynb

Can you tell me, what's wrong? What I'm doing wrong?

gregtyminski, 2020-04-25:
@dusty-nv: any idea, what it's wrong?

devshank3, 2020-05-07:
Hi @gregtyminski  you can review my snippet for Realsense D435i integration, it works fine with jetson nano detectnet and also gives out mean depth of the detected object.  

https://gist.github.com/devshank3/0b65c5c94b10111b39f612248d81f95a

yuchen2580, 2021-04-12:
@gregtyminski 
Hi 
I'm facing the same issue, I think is not about the camera, but the opencv(CPU) -> GPU conversion is wrong, Have you fix it?

closed 2023-04-27 by dusty-nv