dusty-nv/jetson-inference #557: CudaFromNumpy Bottleneck Performance Issues

mbufi, 2020-04-27:
Hello @dusty-nv !

I have been running performance tests with a sample resnet18 network on images. 
Currently there is one bottleneck to the whole system:
**Total amount of images = 8006
Image dim = 224x224
Platform = python**

CudaFromNumpy (over 8006 imgs)= **26.22 secs** 
`cuda_img = jetson.utils.cudaToNumpy(img)`

Resnet18 Total Inference Time (over 8006 imgs)= **16.22 seconds** 

Overall Total = > 42 sec
We spend more time moving an array to GPU than doing inference!
This is turns converts our **500FPS system --> 190 FPS.**

As you can see there is a huge bottle neck when it comes to doing inference == The biggest one being copying the numpy array to the GPU to be used by cuda. 

Now I know that usually its smarter to copy larger sized arrays in order to see the speed increases... but its a shame that the biggest bottleneck to the system is moving the arrays around. 

Is there a way to speed this up?

Thank you!

dusty-nv, 2020-04-27:
@mbufi, using numpy for the data is not the optimal path - where is your data originating from?

mbufi, 2020-04-27:
@dusty-nv 
I am loading the image with `cv2.imread() ` just to replicate the process of using a baslar camera that converts the image into a numpy array. 

Should I not be using numpy? What do you suggest instead? I have to do extra preprocessing to the images such as equalizatio and specific cropping, so numpy worked the best in this case...

dusty-nv, 2020-04-27:
It would be preferred to load the image into shared CPU/GPU memory (like `jetson.utils.loadImageRGBA()` does) and perform extra pre-processing on the GPU, otherwise you are going to be bottlenecked with CPU.

mbufi, 2020-04-27:
You load in the image to CPU/GPU like so:
`img, width, height = jetson.utils.loadImageRGBA(opt.filename)`

This requires just passing in the filename. If I were to pass in a numpy array representing the image, would that also work? If I remember correctly, that function didnt allow me to do that.

dusty-nv, 2020-04-28:
It would need to do a similar conversion of the data as `cudaFromNumpy()` does.   If you need the top performance, I recommend that you either use the C++ API to jetson-inference, or dig into the C implementation of the `cudaFromNumpy()` extension and optimize it for your use-case.

ozett, 2020-05-16:
i have the same approach, getting images from cv2 into the tensorRT inference.
maybe someon could help with  a sample-fragment how to convert?
(without saving to disk)

thx alot,
toz

closed 2023-04-27 by dusty-nv