dusty-nv/jetson-inference #558: Reference paper for pednet (ped-100)

nkhdiscovery, 2020-04-28:
Hi @dusty-nv 

I am trying to directly use pednet caffemodel in python (building tensorrt engine from scratch, without using your c code but just by using tensorrt python API). I am building my engine, and I get output of layers named "coverage" and "bboxes" but I could not figure out how to decode their output. 

Is there any references on the pednet model (ped-100)? Which one is the original paper?
I was reading this:
[PedNet: A Spatio-Temporal Deep Convolutional Neural Network for Pedestrian Segmentation](https://ntnuopen.ntnu.no/ntnu-xmlui/bitstream/handle/11250/2586966/1613698.pdf?sequence=2)
but I did not found any clues that your pednet is the same PedNet, as it was using more than one frame and the final output is a mask, not bounding boxes and coverage. Would you please give me more information on what exactly this pednet is and how its output should be interpreted? 
(The wired thing was that the length of bboxes is fixed for different images. So is the result of coverage)

dusty-nv, 2020-04-28:
Hi @nkhdiscovery , the PedNet model in jetson-inference uses the DetectNet architecture - https://devblogs.nvidia.com/detectnet-deep-neural-network-object-detection-digits/

Going forward, it is recommended to use an architecture more recent for object detection (e.g. SSD, YOLOv3, ect).  These legacy DetectNet models are provided mostly for backwards-compatibility and are slower than SSD-Mobilenet would be.

nkhdiscovery, 2020-04-28:
Thank you for your clarification, @dusty-nv 
I am familiar with state of the art models, I was just curious about this and how it works, specifically as I was going to test the python API with it, and as I successfully did prediction but I could not figure out the output's values, as coverage values are larger than 1 (or even negative) some times and interpretation of values for bboxes are not documented anywhere. 
I will again take a look on detectnet implementation to figure it out.

closed 2023-04-27 by dusty-nv