dusty-nv/jetson-inference #559: Installing PyTorch -> ImportError: libcudart.so.10.0: cannot open shared object file: No such file or directory

penizekr, 2020-04-28:
Hello,
I have always error "ImportError: libcudart.so.10.0: cannot open shared object file: No such file or directory" during installation of Pytorch on Jetson Nano. I am going step by step according to instruction "Building the Project from Source". Similar also for Python3. How to solve it?

Thanks.

`^[[M#|I
[jetson-inference]  Package selection status:  0
[jetson-inference]  Packages selected for download:  1
[jetson-inference]  Downloading PyTorch v1.1.0 (Python 2.7)...
dpkg-query: no packages found matching python-pip
[jetson-inference]  Checking for 'python-pip' deb package...not installed
[jetson-inference]  Missing 'python-pip' deb package...installing 'python-pip' package.
Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  apt-clone archdetect-deb bogl-bterm busybox-static cryptsetup-bin
  dpkg-repack gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 grub-common
  kde-window-manager kinit kio kpackagetool5 kwayland-data kwin-common
  kwin-data kwin-x11 libdebian-installer4 libkdecorations2-5v5
  libkdecorations2private5v5 libkf5activities5 libkf5attica5
  libkf5completion-data libkf5completion5 libkf5declarative-data
  libkf5declarative5 libkf5doctools5 libkf5globalaccel-data libkf5globalaccel5
  libkf5globalaccelprivate5 libkf5idletime5 libkf5jobwidgets-data
  libkf5jobwidgets5 libkf5kcmutils-data libkf5kcmutils5 libkf5kiocore5
  libkf5kiontlm5 libkf5kiowidgets5 libkf5newstuff-data libkf5newstuff5
  libkf5newstuffcore5 libkf5package-data libkf5package5 libkf5plasma5
  libkf5quickaddons5 libkf5solid5 libkf5solid5-data libkf5sonnet5-data
  libkf5sonnetcore5 libkf5sonnetui5 libkf5textwidgets-data libkf5textwidgets5
  libkf5waylandclient5 libkf5waylandserver5 libkf5xmlgui-bin libkf5xmlgui-data
  libkf5xmlgui5 libkscreenlocker5 libkwin4-effect-builtins1 libkwineffects11
  libkwinglutils11 libkwinxrenderutils11 libqgsttools-p1 libqt5designer5
  libqt5help5 libqt5multimedia5 libqt5multimedia5-plugins
  libqt5multimediaquick-p5 libqt5multimediawidgets5 libqt5opengl5
  libqt5positioning5 libqt5printsupport5 libqt5qml5 libqt5quick5
  libqt5quickwidgets5 libqt5sensors5 libqt5sql5 libqt5test5 libqt5webchannel5
  libqt5webkit5 libxcb-composite0 libxcb-cursor0 libxcb-damage0 os-prober
  python3-dbus.mainloop.pyqt5 python3-icu python3-pam python3-pyqt5
  python3-pyqt5.qtsvg python3-pyqt5.qtwebkit python3-sip
  qml-module-org-kde-kquickcontrolsaddons qml-module-qtmultimedia
  qml-module-qtquick2 rdate tasksel tasksel-data
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libpython-all-dev python-all python-all-dev python-asn1crypto
  python-cffi-backend python-crypto python-cryptography python-enum34
  python-idna python-ipaddress python-keyring python-keyrings.alt
  python-pip-whl python-secretstorage python-setuptools python-wheel
  python-xdg
Suggested packages:
  python-crypto-doc python-cryptography-doc python-cryptography-vectors
  python-enum34-doc gir1.2-gnomekeyring-1.0 python-fs python-gdata
  python-keyczar python-secretstorage-doc python-setuptools-doc
The following NEW packages will be installed:
  libpython-all-dev python-all python-all-dev python-asn1crypto
  python-cffi-backend python-crypto python-cryptography python-enum34
  python-idna python-ipaddress python-keyring python-keyrings.alt python-pip
  python-pip-whl python-secretstorage python-setuptools python-wheel
  python-xdg
0 upgraded, 18 newly installed, 0 to remove and 131 not upgraded.
Need to get 2 894 kB of archives.
After this operation, 8 742 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 libpython-all-dev arm64 2.7.15~rc1-1 [1 092 B]
Get:2 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-all arm64 2.7.15~rc1-1 [1 076 B]
Get:3 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-all-dev arm64 2.7.15~rc1-1 [1 100 B]
Get:4 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-asn1crypto all 0.24.0-1 [72,7 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-cffi-backend arm64 1.11.5-1 [59,2 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-crypto arm64 2.6.1-8ubuntu2 [240 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-enum34 all 1.1.6-2 [34,8 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-idna all 2.6-1 [32,4 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports bionic/main arm64 python-ipaddress all 1.0.17-1 [18,2 kB]
[... 369 lines cut ...]
-- Copying /home/rp/jetson-inference/utils/python/examples/gl-display-test.py
-- trying to build Python bindings for Python versions:  2.7;3.6;3.7
-- detecting Python 2.7...
-- found Python version:  2.7 (2.7.17)
-- found Python include:  /usr/include/python2.7
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython2.7.so
-- detecting Python 3.6...
-- found Python version:  3.6 (3.6.9)
-- found Python include:  /usr/include/python3.6m
-- found Python library:  /usr/lib/aarch64-linux-gnu/libpython3.6m.so
-- detecting Python 3.7...
-- Python 3.7 wasn't found
-- Copying /home/rp/jetson-inference/python/examples/detectnet-camera.py
-- Copying /home/rp/jetson-inference/python/examples/detectnet-console.py
-- Copying /home/rp/jetson-inference/python/examples/imagenet-camera.py
-- Copying /home/rp/jetson-inference/python/examples/imagenet-console.py
-- Copying /home/rp/jetson-inference/python/examples/my-detection.py
-- Copying /home/rp/jetson-inference/python/examples/my-recognition.py
-- Copying /home/rp/jetson-inference/python/examples/segnet-batch.py
-- Copying /home/rp/jetson-inference/python/examples/segnet-camera.py
-- Copying /home/rp/jetson-inference/python/examples/segnet-console.py
-- Linking jetson-inference with OpenCV 4.1.1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/rp/jetson-inference/build`

dusty-nv, 2020-04-28:
Hi @penizekr, are you on JetPack 4.4?   Sorry, the script needs updated for that.  Can you uninstall torch/torchvision pip/pip3 packages and instead install from here please?   https://forums.developer.nvidia.com/t/pytorch-for-jetson-nano-version-1-5-0-now-available/72048/261

penizekr, 2020-04-28:
Hi, how to find which version of JetPack I have?

penizekr, 2020-04-28:
Which version of PyTorch should I install? Thanks.

dusty-nv, 2020-04-28:
You can determine your JetPack version with this terminal command:

```
dpkg-query --showformat='${Version}' --show nvidia-l4t-core
```

If that doesn't work, run `cat /etc/nv_tegra_release`

These commands will print out the L4T version (let me know what that is for you), which corresponds to the JetPack version.  Try using PyTorch 1.2 for your appropriate version of JetPack.

penizekr, 2020-04-28:
First command returned:
32.4.2-20200408182156

Second command:
# R32 (release), REVISION: 4.2, GCID: 20074772, BOARD: t210ref, EABI: aarch64, DATE: Thu Apr  9 01:22:12 UTC 2020

dusty-nv, 2020-04-28:
OK, use these wheels for JetPack 4.4:

Python 2.7 [`torch-1.2.0-cp27-cp27mu-linux_aarch64.whl`](https://nvidia.box.com/shared/static/8673jprub45abv3tmx4sscdllpbinoh9.whl)
Python 3.6 [` torch-1.2.0-cp36-cp36m-linux_aarch64.whl`](https://nvidia.box.com/shared/static/lufbgr3xu2uha40cs9ryq1zn4kxsnogl.whl)

The install instructions for PyTorch and torchvision are on this page:  https://forums.developer.nvidia.com/t/pytorch-for-jetson-nano-version-1-5-0-now-available/72048/1

You will want to install torchvision v0.4.0 since it is with PyTorch 1.2.

penizekr, 2020-04-28:
OK, thanks. I will try it.

penizekr, 2020-04-28:
During installation of torchvision I got an error after comman  "sudo python setup.py install" :

Traceback (most recent call last):
  File "setup.py", line 13, in <module>
    import torch
  File "/home/rp/.local/lib/python2.7/site-packages/torch/__init__.py", line 81, in <module>
    from torch._C import *
ImportError: libmpi_cxx.so.20: cannot open shared object file: No such file or directory

penizekr, 2020-04-28:
Same error in python after "import torch" 

>>> import torch
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/rp/.local/lib/python2.7/site-packages/torch/__init__.py", line 81, in <module>
    from torch._C import *
ImportError: libmpi_cxx.so.20: cannot open shared object file: No such file or directory
>>>

penizekr, 2020-04-28:
These problems I have with  Jetson nano image from 2020/04/21 (JP4.4). Would you recommend to use version from 2019/12/17 (JP4.3) to avoid the problems?

dusty-nv, 2020-04-28:
> Would you recommend to use version from 2019/12/17 (JP4.3) to avoid the problems?

For JP 4.3, you would need to use the JetPack 4.3 version of the PyTorch wheels.   But these do work on Nano with JP 4.4, perhaps you are just missing MPI (which is typically installed by default).  Can you try installing `libopenmpi-dev`:

`sudo apt-get install libopenmpi-dev`

penizekr, 2020-04-28:
I can confirm step forward. After installation of libopenmpi-dev I can run "import torch" and verify version.
Now is running "sudo python setup.py install" for torchvision. With many warnings.

rp@rp-desktop:~$ python
Python 2.7.17 (default, Nov  7 2019, 10:07:09) 
[GCC 7.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> print(torch.__version__)
1.2.0

dusty-nv, 2020-04-28:
Thanks, I have added `libopenmpi-dev` to the instructions on the forum just in case other users encounter that too.

penizekr, 2020-04-28:
torchvision verification shows:

Python 2.7.17 (default, Nov  7 2019, 10:07:09) 
[GCC 7.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import torchvision
>>> print(torchvision.__version__)
0.4.1a0+a263704

penizekr, 2020-04-28:
It works now. Thanks for quick help!
BR
Radek

closed 2020-04-28 by penizekr