dusty-nv/jetson-inference #563: How do I train SSD-MobileNet-v2 with my custom data?

66Leo66, 2020-05-02:
Sorry for asking here, but I really can't find anything useful about training SSD-MobileNet-v2 to detect my own stuffs. Can anyone please provide something about it? Great thanks!

dusty-nv, 2020-05-02:
Hi Leo, for TensorFlow see here - https://forums.developer.nvidia.com/t/digits-or-somthing-else/110790/7?u=dusty_nv

For PyTorch, see github.com/dusty-nv/pytorch-ssd

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

66Leo66, 2020-05-06:
Will SSD-Mibilenet-v2 work with Nvidia Digits?