dusty-nv/jetson-inference #565: Getting top-n predictions instead of top-1

py-sk, 2020-05-04:
For imagenet, is there a way to get the top-n predictions like what is printed out on console instead of the top-1 prediction that is returned from the classify function?

AiZUBI, 2020-09-10:
I need this too. @py-sk, did you figure out how to extract top-n predictions?

vjsrinivas, 2021-12-28:
If anyone else is using the Classify function, I currently can't find an easy way of getting the Top-K predictions.
I was able to get the whole output layer value though, and from there, you can run something like `np.argsort` or `torch.topk` to get what you want.

To get the output layer value was convoluted for me, but a starting point is to look at the following functions to modify or expand:
* In `./c/imageNet.cpp`, `Classify` with top-1 happens in the function signature `int imageNet::Classify( float* confidence )` and is called by the following overloaded function `int imageNet::Classify( void* image, uint32_t width, uint32_t height, imageFormat format, float* confidence )`. You should create your own function that returns all the values and run it in the overloaded function.
* The `mOutput` that is looped through to find the Top-1 is the output layer value and is just a vector<layerInfo> [[docs](https://github.com/dusty-nv/jetson-inference/blob/821bea9ec7e1e4fc61c7d7e30832ed53222510dc/c/tensorNet.h#L740)]. You can get the raw data with `mOutput[0].CPU`.
* C++ to Python bindings happen in `./python/bindings` and, in our instance, in the `PyImageNet.cpp` file, where the Classify function returns a `PyObject` primitive. The hardest part for me was figuring out how to get a `float*` to something that can be interpreted by Python.
* Some ideas are using the PyArray class from numpy or just creating a Python list object and filling it with the float pointer's values. I tried the second one since there's no extra dependencies. It worked but the conversion slows the FPS from a 100 to 80., so there's room for improvement.

jiefengsun, 2023-04-04:
@vjsrinivas Is this problem solved. It seems that this function is added but it does not work. 

""""""
   # classify the image and get the topK predictions
    # if you only want the top class, you can simply run:
    #   class_id, confidence = net.Classify(img)
    predictions = net.Classify(img, topK=args.topK)

    # draw predicted class labels
    for n, (classID, confidence) in enumerate(predictions):
        classLabel = net.GetClassLabel(classID)
        confidence *= 100.0

        print(f"imagenet:  {confidence:05.2f}% class #{classID} ({classLabel})")

        font.OverlayText(img, text=f"{confidence:05.2f}% {classLabel}", 
                         x=5, y=5 + n * (font.GetSize() + 5),
                         color=font.White, background=font.Gray40)
"""""""

dusty-nv, 2023-04-04:
@jiefengsun this was added yes, what is the problem you are having with it not working?

jiefengsun, 2023-04-04:
@dusty-nv , thanks for reply.

Here is my code, it just print out the first nomatter topK

# classify the image and get the topK predictions
# if you only want the top class, you can simply run:
#   class_id, confidence = net.Classify(img)
predictions = net.Classify(img, topK= 3)
font = cudaFont()
# draw predicted class labels
image_out = img
for n, (classID, confidence) in enumerate(predictions):
    classLabel = net.GetClassLabel(classID)
    confidence *= 100.0
    print(n)

    # print out the result
    print("'{:s}'  with {:f}% confidence".format(classLabel, confidence))

    # put the results on the image
    font.OverlayText(image_out, text=f"{confidence:05.2f}% {classLabel}", 
                    x=5, y=5 + n * (font.GetSize() + 5),
                    color=font.White, background=font.Gray40)

# save the image 
current_time = datetime.datetime.now().strftime('%d_%H-%M-%S')
saveImage( f'images_done/{current_time}.jpg', img, 224)

jiefengsun, 2023-04-04:
[image]

Here is a screen shot of whole code.

jiefengsun, 2023-04-04:
It is also very strange that the confidence is more than 100%.

dusty-nv, 2023-04-04:
> Here is my code, it just print out the first nomatter topK

Any classes who's confidence is less than the threshold will be dropped, regardless of topK

> It is also very strange that the confidence is more than 100%.

Yes that is strange, does that happen with any of the built-in pretrained models like `googlenet` or `resnet18`?

jiefengsun, 2023-04-05:
@dusty-nv No, it does not happen to the built-in pre-trained models. 
I think that the problem might be that I used customized code to train and export to the onnx. The trained model works great. The problem might be the customized code to export it to onnx.

closed 2023-04-27 by dusty-nv