dusty-nv/jetson-inference #566: Problem install

MaulyGabriel, 2020-05-07:
[jetson-inference]  BUILD_INTERACTVE=
[jetson-inference]  non-interactive mode, skipping PyTorch install...
 
[jetson-inference]  installation complete, exiting with status code 0
[jetson-inference]  to run this tool again, use the following commands:
 
    $ cd <jetson-inference>/build
    $ ./install-pytorch.sh

hyattbaker, 2020-05-08:
@MaulyGabriel , get any resolution? Even though I thought for sure during install I installed pytorch I couldn't import it. So I just tried to install again with the interactive pop up and got the same error as you.  

[image]

dusty-nv, 2020-05-08:
When you guys run cmake, can you try running it like:

``` bash
cmake -DBUILD_INTERACTIVE=YES ../
```

hyattbaker, 2020-05-08:
HMM I just ran cmake -DBUILD_INTERACTIVE=YES ../ and it appeared to work without issue but I still have the same problem.
`
xv1@xv1-desktop:~/jetson-inference/build$ cmake -DBUILD_INTERACTIVE=YES ../
 
Note:  this project uses git submodules in the source tree.
       if you haven't already, run the following command from
       the project's root directory:
 
           git submodule update --init

-- using patched FindCUDA.cmake
-- CUDA version: 10.2
-- CUDA 10 detected, enabling SM_72
-- OpenCV version:  4.1.1
-- OpenCV version >= 3.0.0, enabling OpenCV
-- system arch:  aarch64
-- output path:  /home/xv1/jetson-inference/build/aarch64
-- Copying /home/xv1/jetson-inference/c/detectNet.h
-- Copying /home/xv1/jetson-inference/c/homographyNet.h
-- Copying /home/xv1/jetson-inference/c/imageNet.h
-- Copying /home/xv1/jetson-inference/c/segNet.h
-- Copying /home/xv1/jetson-inference/c/superResNet.h
-- Copying /home/xv1/jetson-inference/c/tensorNet.h
-- Copying /home/xv1/jetson-inference/c/imageNet.cuh
-- Copying /home/xv1/jetson-inference/calibration/randInt8Calibrator.h
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- found Qt5Widgets version:  5.9.5
-- found Qt5Widgets defines:  -DQT_WIDGETS_LIB;-DQT_GUI_LIB;-DQT_CORE_LIB
-- found Qt5Widgets library:  Qt5::Widgets
-- found Qt5Widgets include:  /usr/include/aarch64-linux-gnu/qt5/;/usr/include/aarch64-linux-gnu/qt5/QtWidgets;/usr/include/aarch64-linux-gnu/qt5/QtGui;/usr/include/aarch64-linux-gnu/qt5/QtCore;/usr/lib/aarch64-linux-gnu/qt5//mkspecs/linux-g++
-- camera-capture:  building as submodule, /home/xv1/jetson-inference/tools
-- jetson-utils:  building as submodule, /home/xv1/jetson-inference
-- Copying /home/xv1/jetson-inference/utils/XML.h
-- Copying /home/xv1/jetson-inference/utils/commandLine.h
-- Copying /home/xv1/jetson-inference/utils/filesystem.h
-- Copying /home/xv1/jetson-inference/utils/mat33.h
-- Copying /home/xv1/jetson-inference/utils/pi.h
-- Copying /home/xv1/jetson-inference/utils/rand.h
-- Copying /home/xv1/jetson-inference/utils/timespec.h
-- Copying /home/xv1/jetson-inference/utils/camera/gstCamera.h
-- Copying /home/xv1/jetson-inference/utils/camera/v4l2Camera.h
-- Copying /home/xv1/jetson-inference/utils/codec/gstDecoder.h
-- Copying /home/xv1/jetson-inference/utils/codec/gstEncoder.h
-- Copying /home/xv1/jetson-inference/utils/codec/gstUtility.h
-- Copying /home/xv1/jetson-inference/utils/cuda/cudaFont.h
-- Copying /home/xv1/jetson-inference/utils/cuda/cudaMappedMemory.h
-- Copying /home/xv1/jetson-inference/utils/cuda/cudaNormalize.h
-- Copying /home/xv1/jetson-inference/utils/cuda/cudaOverlay.h
-- Copying /home/xv1/jetson-inference/utils/cuda/cudaRGB.h
-- Copying /home/xv1/jetson-inference/utils/cuda/cudaResize.h
-- Copying /home/xv1/jetson-inference/utils/cuda/cudaUtility.h
-- Copying /home/xv1/jetson-inference/utils/cuda/cudaWarp.h
-- Copying /home/xv1/jetson-inference/utils/cuda/cudaYUV.h
-- Copying /home/xv1/jetson-inference/utils/display/glDisplay.h
-- Copying /home/xv1/jetson-inference/utils/display/glTexture.h
-- Copying /home/xv1/jetson-inference/utils/display/glUtility.h
-- Copying /home/xv1/jetson-inference/utils/image/imageIO.h
-- Copying /home/xv1/jetson-inference/utils/image/loadImage.h
-- Copying /home/xv1/jetson-inference/utils/input/devInput.h
-- Copying /home/xv1/jetson-inference/utils/input/devJoystick.h
-- Copying /home/xv1/jetson-inference/utils/input/devKeyboard.h
-- Copying /home/xv1/jetson-inference/utils/network/Endian.h
-- Copying /home/xv1/jetson-inference/utils/network/IPv4.h
-- Copying /home/xv1/jetson-inference/utils/network/NetworkAdapter.h
-- Copying /home/xv1/jetson-inference/utils/network/Socket.h
-- Copying /home/xv1/jetson-inference/utils/threads/Event.h
-- Copying /home/xv1/jetson-inference/utils/threads/Mutex.h
-- Copying /home/xv1/jetson-inference/utils/threads/Process.h
-- Copying /home/xv1/jetson-inference/utils/threads/Thread.h
-- trying to build Python bindings for Python versions:  2.7;3.6;3.7
[... 31 lines cut ...]
-- detecting Python 3.7...
-- Python 3.7 wasn't found
-- Copying /home/xv1/jetson-inference/python/examples/detectnet-camera.py
-- Copying /home/xv1/jetson-inference/python/examples/detectnet-console.py
-- Copying /home/xv1/jetson-inference/python/examples/imagenet-camera.py
-- Copying /home/xv1/jetson-inference/python/examples/imagenet-console.py
-- Copying /home/xv1/jetson-inference/python/examples/my-detection.py
-- Copying /home/xv1/jetson-inference/python/examples/my-recognition.py
-- Copying /home/xv1/jetson-inference/python/examples/segnet-batch.py
-- Copying /home/xv1/jetson-inference/python/examples/segnet-camera.py
-- Copying /home/xv1/jetson-inference/python/examples/segnet-console.py
-- Linking jetson-inference with OpenCV 4.1.1
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xv1/jetson-inference/build
xv1@xv1-desktop:~/jetson-inference/build$ ./install-pytorch.sh
[jetson-inference]  BUILD_INTERACTVE=
[jetson-inference]  non-interactive mode, skipping PyTorch install...
 
[jetson-inference]  installation complete, exiting with status code 0
[jetson-inference]  to run this tool again, use the following commands:
 
    $ cd <jetson-inference>/build
    $ ./install-pytorch.sh
`

hyattbaker, 2020-05-08:
FWIW, the interactive mode came up on the initial install and I thought for sure i added support for both versions of pytorch

dusty-nv, 2020-05-08:
Are you by chance on JetPack 4.4 Developer Preview?  If so, you should manually install the PyTorch wheels for JetPack 4.4 from here:  https://forums.developer.nvidia.com/t/pytorch-for-jetson-nano-version-1-5-0-now-available/72048

Sorry, I need to update the interactive PyTorch installer script so that it selects the correct version based on your JetPack version, since JetPack 4.4 requires a different build of wheel now.

hyattbaker, 2020-05-08:
Well snap I did install DP. I'm now wondering what other issues I might have.  The tutorial up to the pytorch stuff worked just fine.

hyattbaker, 2020-05-08:
Thanks BTW really appreciate the help

dusty-nv, 2020-05-08:
The inferencing part of the tutorial has been updated for TensorRT 7.1 and JetPack 4.4 DP, just not the interactive PyTorch installer yet.   Should be fine if you manually uninstall it with `pip uninstall torch && pip uninstall torchvision` or similar, and then install the JP 4.4 wheels for PyTorch and torchvision (torchvision from source).

flurpo, 2020-05-10:
Did you get a successful Torch installation? I've run into the same. I'm believe I'm running JP 4.4. Maybe we need to go back to 4.3?

 tx2:/jetsonUtilities$ ./jetsonInfo.py
 NVIDIA Jetson TX2
 L4T 32.4.2 [ JetPack UNKNOWN ]
 Ubuntu 18.04.4 LTS
 Kernel Version: 4.9.140-tegra
 CUDA 10.2.89

I tried installting the whl file as dusty-nv recommended but ran into some unusual issues.

tx2:/$ pip install torch-1.4.0-cp27-cp27mu-linux_aarch64.whl
Defaulting to user installation because normal site-packages is not writeable
ERROR: torch-1.4.0-cp27-cp27mu-linux_aarch64.whl is not a supported wheel on this platform.

? Not supported on JP 4.4?

x2:/$ pip3 install numpy torch-1.4.0-cp36-cp36m-linux_aarch64.whl
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (1.13.3)
Processing /torch-1.4.0-cp36-cp36m-linux_aarch64.whl
Installing collected packages: torch
  WARNING: The scripts convert-caffe2-to-onnx and convert-onnx-to-caffe2 are installed in '/home/w/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed torch-1.4.0

Looks like the whl file on 3.6 got through the installation, unlike 2.7, but with a weird warning. I echo PATH$ and that path is there....

sure enough. no luck with torch

tx2:/torchvision$ python3
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/w/.local/lib/python3.6/site-packages/torch/__init__.py", line 81, in <module>
    from torch._C import *
ImportError: libcudart.so.10.0: cannot open shared object file: No such file or directory
>>> 

Any idea's? Would appreciate any help! Best.

hyattbaker, 2020-05-10:
@flurpo, my linux skills are a bit weak, so I moved back to 4.3.

flurpo, 2020-05-10:
@hyattbaker - thanks for quick response! How did you re-flash back to 4.3? Just go back through installation fodder? Best.

hyattbaker, 2020-05-10:
I was using Ubuntu and the SDK manager to flash XAVIER.  In small text on the 1st or 2nd screen of the manager there is an uninstall option. I uninstalled the 4.4 then reflashed with 4.3, which I had to poke around on the manager to find as it wasn't apparent at first.  Maybe you can just re-flash it, but I decided to go with uninstall first.

flurpo, 2020-05-11:
@hyattbaker -Thanks for the help...I went back to 4.3 using your description above and torch installed perfectly for py2.7. Small bug coming back with numpy and py3.6 at test but at least I'm up and moving again. Again, thanks.......

jetsontz, 2021-01-13:
im getting this issue

[image]

closed 2023-04-27 by dusty-nv