dusty-nv/jetson-inference #567: Cmake Failed to Compile CUDA issues

muhammadabdullah34907, 2020-05-07:
Hi @dusty-nv 
Anyone please ?

Compiling on PC.
CUDA 10.2
Tensorflow-gpu 1.14

-- using patched FindCUDA.cmake
-- CUDA version: 10.2
-- CUDA 10 detected, enabling SM_72
-- Found OpenCV: /usr (found version "4.2.0") found components:  core calib3d 
-- OpenCV version:  4.2.0
-- OpenCV version >= 3.0.0, enabling OpenCV
-- system arch:  x86_64
-- output path:  /home/admin-connection1/Documents/jetson-inference/build/x86_64
-- Copying /home/admin-connection1/Documents/jetson-inference/c/detectNet.h
-- Copying /home/admin-connection1/Documents/jetson-inference/c/homographyNet.h
-- Copying /home/admin-connection1/Documents/jetson-inference/c/imageNet.cuh
-- Copying /home/admin-connection1/Documents/jetson-inference/c/imageNet.h
-- Copying /home/admin-connection1/Documents/jetson-inference/c/segNet.h
-- Copying /home/admin-connection1/Documents/jetson-inference/c/superResNet.h
-- Copying /home/admin-connection1/Documents/jetson-inference/c/tensorNet.h
-- Copying /home/admin-connection1/Documents/jetson-inference/calibration/randInt8Calibrator.h
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- found Qt5Widgets version:  5.9.7
-- found Qt5Widgets defines:  -DQT_WIDGETS_LIB;-DQT_GUI_LIB;-DQT_CORE_LIB
-- found Qt5Widgets library:  Qt5::Widgets
-- found Qt5Widgets include:  /home/admin-connection1/anaconda3/include/qt/;/home/admin-connection1/anaconda3/include/qt/QtWidgets;/home/admin-connection1/anaconda3/include/qt/QtGui;/home/admin-connection1/anaconda3/include/qt/QtCore;/home/admin-connection1/anaconda3/.//mkspecs/linux-g++
-- camera-capture:  building as submodule, /home/admin-connection1/Documents/jetson-inference/tools
-- jetson-utils:  building as submodule, /home/admin-connection1/Documents/jetson-inference
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/XML.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/camera/gstCamera.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/camera/v4l2Camera.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/codec/gstDecoder.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/codec/gstEncoder.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/codec/gstUtility.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/commandLine.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/cuda/cudaFont.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/cuda/cudaMappedMemory.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/cuda/cudaNormalize.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/cuda/cudaOverlay.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/cuda/cudaRGB.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/cuda/cudaResize.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/cuda/cudaUtility.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/cuda/cudaWarp.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/cuda/cudaYUV.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/display/glDisplay.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/display/glTexture.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/display/glUtility.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/filesystem.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/image/imageIO.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/image/loadImage.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/input/devInput.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/input/devJoystick.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/input/devKeyboard.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/mat33.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/network/Endian.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/network/IPv4.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/network/NetworkAdapter.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/network/Socket.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/pi.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/rand.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/threads/Event.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/threads/Mutex.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/threads/Process.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/threads/Thread.h
-- Copying /home/admin-connection1/Documents/jetson-inference/utils/timespec.h
-- trying to build Python bindings for Python versions:  2.7;3.6;3.7
-- detecting Python 2.7...
-- found Python version:  3.7 (3.7.6)
[... 99 lines cut ...]
  shared library created in source directory
  "/home/admin-connection1/Documents/jetson-inference/python/bindings".  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:41 (cuda_add_library)

CMake Error at python/bindings/CMakeLists.txt:43 (target_link_libraries):
  Attempt to add link library "jetson-inference" to target
  "jetson-inference-python-37" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

-- Copying /home/admin-connection1/Documents/jetson-inference/python/examples/detectnet-camera.py
-- Copying /home/admin-connection1/Documents/jetson-inference/python/examples/detectnet-console.py
-- Copying /home/admin-connection1/Documents/jetson-inference/python/examples/imagenet-camera.py
-- Copying /home/admin-connection1/Documents/jetson-inference/python/examples/imagenet-console.py
-- Copying /home/admin-connection1/Documents/jetson-inference/python/examples/my-detection.py
-- Copying /home/admin-connection1/Documents/jetson-inference/python/examples/my-recognition.py
-- Copying /home/admin-connection1/Documents/jetson-inference/python/examples/segnet-batch.py
-- Copying /home/admin-connection1/Documents/jetson-inference/python/examples/segnet-camera.py
-- Copying /home/admin-connection1/Documents/jetson-inference/python/examples/segnet-console.py
-- Linking jetson-inference with OpenCV 4.2.0
-- Configuring incomplete, errors occurred!
See also "/home/admin-connection1/Documents/jetson-inference/build/CMakeFiles/CMakeOutput.log".
See also "/home/admin-connection1/Documents/jetson-inference/build/CMakeFiles/CMakeError.log"

anfxanfx, 2020-05-16:
I had a similar problem trying to install on my TX2 with Jetpack 4.3.  My error was:

-- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 

It took me about three hours of Googling to find this old post from Dusty from back in 2017 that pointed towards the root issue of the problem: https://github.com/dusty-nv/jetson-inference/issues/113#issuecomment-332316250.

As explained in the post, I added CUDA Toolkit, cuDNN, and TensorRT to the target and successfully reran the jetson-inference cmake process.

felipevw, 2020-05-19:
As far as I know, Tensorflow does not support CUDA 10.2. This might be one thing, but I am not sure if you are building cmake in a pc or in the jetson. 
Please provide mode details of your system and if you could also, use the insert code option when providing the cmake output messages.

dusty-nv, 2020-05-19:
It appears that @muhammadabdullah34907's issue is related to python symlinks, as opposed to TensorFlow or JetPack install.   

From this log output, it is detecting both Python 2.7/3.6 as Python 3.7 instead:

```
-- detecting Python 2.7...
-- found Python version: 3.7 (3.7.6)
-- found Python include: /home/admin-connection1/anaconda3/include/python3.7m
-- found Python library: /home/admin-connection1/anaconda3/lib/libpython3.7m.so
-- detecting Python 3.6...
-- found Python version: 3.7 (3.7.6)
```

Then this error occurs because the versions are the same:

```
add_library cannot create target "jetson-inference-python-37" because
another target with the same name already exists. The existing target is a
shared library created in source directory
```

Other than unlinking `python` from `python3.7` in your shell environment, I'm not sure how to fix it.   How I detect the python versions is with cmake `find_package(PythonInterp)`, so the issue lies under the hood.

muhammadabdullah34907, 2020-05-20:
Hi Guys 

Thanks. As pointed by @dusty-nv  this issue occurred because I had only python3.7. Then I installed python2.7 and python3.6, issue resolved. It automatically detects python2.7 and python3.6 during cmake compilation. 

thanks

closed 2020-05-20 by muhammadabdullah34907

charithmu, 2023-09-07:
Additional info:

I used this PPA to install Python 3.7, 3.6 and 2.7.
Script correctly finds them and completes with success. 
Tested on Ubuntu:22.04