dusty-nv/jetson-inference #569: how to run tensorflow frozen_inference_graph.pb in jetson.inference.detectNet

ynjiun, 2020-05-11:
Hi,

I have trained my ssdlite_mobilenet_v3 in tensorflow and export as frozen_inference_graph.pb. I am able to run it under my ti1080 GPU. Now I would like to transfer to Jetson Nano. How can I run it using jetson.inference.detectNet? Please advise. Thank you !

Attached please find my model files in a sample_model_inference.zip file

[sample_model_inference.zip](https://github.com/dusty-nv/jetson-inference/files/4607243/sample_model_inference.zip)

ynjiun, 2020-05-13:
I follow the end_to_end_tensorflow_mnist example, the first step is to convert my frozen_inference_graph.py to uff file, but fail with following error messages:

`convert-to-uff frozen_inference_graph.pb `

WARNING:tensorflow:From /home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/conversion_helpers.py:18: The name tf.GraphDef is deprecated. Please use tf.compat.v1.GraphDef instead.

Loading frozen_inference_graph.pb
WARNING:tensorflow:From /home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/conversion_helpers.py:231: The name tf.gfile.GFile is deprecated. Please use tf.io.gfile.GFile instead.

NOTE: UFF has been tested with TensorFlow 1.12.0. Other versions are not guaranteed to work
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
WARNING:tensorflow:From /home/paul/tf1.15/lib/python2.7/site-packages/graphsurgeon/_utils.py:2: The name tf.NodeDef is deprecated. Please use tf.compat.v1.NodeDef instead.

UFF Version 0.6.3
=== Automatically deduced input nodes ===
[name: "image_tensor"
op: "Placeholder"
attr {
  key: "dtype"
  value {
    type: DT_UINT8
  }
}
attr {
  key: "shape"
  value {
    shape {
      dim {
        size: -1
      }
      dim {
        size: -1
      }
      dim {
        size: -1
      }
      dim {
        size: 3
      }
    }
  }
}
]
=========================================

=== Automatically deduced output nodes ===
[name: "detection_boxes"
op: "Identity"
input: "Postprocessor/BatchMultiClassNonMaxSuppression/map/TensorArrayStack/TensorArrayGatherV3"
attr {
  key: "T"
  value {
    type: DT_FLOAT
  }
}
, name: "detection_scores"
op: "Identity"
input: "Postprocessor/BatchMultiClassNonMaxSuppression/map/TensorArrayStack_1/TensorArrayGatherV3"
attr {
  key: "T"
  value {
    type: DT_FLOAT
  }
}
, name: "detection_multiclass_scores"
op: "Identity"
input: "Postprocessor/BatchMultiClassNonMaxSuppression/map/TensorArrayStack_5/TensorArrayGatherV3"
attr {
  key: "T"
  value {
[... 209 lines cut ...]
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/bin/convert_to_uff.py", line 89, in main
    debug_mode=args.debug
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/conversion_helpers.py", line 233, in from_tensorflow_frozen_model
    return from_tensorflow(graphdef, output_nodes, preprocessor, **kwargs)
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/conversion_helpers.py", line 181, in from_tensorflow
    debug_mode=debug_mode)
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/converter.py", line 94, in convert_tf2uff_graph
    uff_graph, input_replacements, debug_mode=debug_mode)
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/converter.py", line 79, in convert_tf2uff_node
    op, name, tf_node, inputs, uff_graph, tf_nodes=tf_nodes, debug_mode=debug_mode)
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/converter.py", line 41, in convert_layer
    fields = cls.parse_tf_attrs(tf_node.attr)
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/converter.py", line 222, in parse_tf_attrs
    return {key: cls.parse_tf_attr_value(val) for key, val in attrs.items() if val is not None and val.WhichOneof('value') is not None}
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/converter.py", line 222, in <dictcomp>
    return {key: cls.parse_tf_attr_value(val) for key, val in attrs.items() if val is not None and val.WhichOneof('value') is not None}
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/converter.py", line 218, in parse_tf_attr_value
    return cls.convert_tf2uff_field(code, val)
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/converter.py", line 190, in convert_tf2uff_field
    return TensorFlowToUFFConverter.convert_tf2numpy_dtype(val)
  File "/home/paul/tf1.15/lib/python2.7/site-packages/uff/converters/tensorflow/converter.py", line 103, in convert_tf2numpy_dtype
    return tf.as_dtype(dtype).as_numpy_dtype
  File "/home/paul/tf1.15/lib/python2.7/site-packages/tensorflow_core/python/framework/dtypes.py", line 126, in as_numpy_dtype
    return _TF_TO_NP[self._type_enum]
KeyError: 20

leandrovrabelo, 2020-05-13:
I had the same issue, after searching a lot I could convert my pb file to uff with the instructions that I put in my Github.
[https://github.com/leandrovrabelo/object_detection/tree/master/uff_convert](url)

Now i'm trying to make it work with DetecNet.

I hope it helps.

ynjiun, 2020-05-13:
Appreciated your help. I tried in my virtualenv of
tensorflow-gpu==1.15.0
python2.7
and having following error messages:

(tf1.15) paul@tensor:~/tf1.15/models/research/object_detection/samples/sample_model_inference$ ./convert_to_uff.py frozen_inference_graph.pb -O NMS -p config.pyTraceback (most recent call last):
  File "./convert_to_uff.py", line 17, in <module>
    import argparse
  File "/usr/lib/python3.6/argparse.py", line 89, in <module>
    import re as _re
  File "/usr/lib/python3.6/re.py", line 142, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 53, in apport_excepthook
    if not enabled():
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 24, in enabled
    import re
  File "/usr/lib/python3.6/re.py", line 142, in <module>
    class RegexFlag(enum.IntFlag):
AttributeError: module 'enum' has no attribute 'IntFlag'

Could you let me know what environment you have? what should I do to duplicate your success? Plus, if you could elaborate a little bit on your config.py? what's the purpose of config.py? Thanks a lot again for your help!

leandrovrabelo, 2020-05-13:
I used python 3.6, maybe that's your problem.

ynjiun, 2020-05-15:
Yes, I am able to run your program to output frozen_inference_graph.uff successfully. Thanks a lot.

any luck on convert the .uff to .engine file for jetson.inference.detectNet to run yet ; )

DrChungAlbert, 2020-05-19:
Hello ynjiun and leandrovrabelo,
I am using google object detection api to retrain the SSD-Mobilenet-v2-coco and using https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/exporting_models.md to get the frozen_inference_graph.pb , and using python convert_to_uff.py frozen_inference_graph.pb -O NMS -p config.py , it output frozen_inference_graph.uff successfully too.
But when I am putting the retrained uff into the jetson xavier's /jetson-inference/data/networks/ssd-mobilenet-v2/   for doing  ./detectnet-console.py --network=ssd-mobilenet-v2 images/peds_0.jpg output.jpg ,the detectNet can't convert the uff by TensorRT engine in sucessfully. Is there something that I missed to process it ?

RoboBoat1, 2020-09-30:
Did @leandrovrabelo ever figure out how to run a .uff into detectnet?

leandrovrabelo, 2020-09-30:
> Did @leandrovrabelo ever figure out how to run a .uff into detectnet?
> 
> 

What version of jetpack are you using?

RoboBoat1, 2020-09-30:
Believe it is 3.1, using a TX2 in a research lab so I never flashed it originally

leandrovrabelo, 2020-09-30:
Hi @RoboBoat1, the whole process is in the attached notebook, it converts to uff, then creates an engine, remember to convert it in the device that you will use: [Convert UFF and Create Engine-To GitHub Issue.zip](https://github.com/dusty-nv/jetson-inference/files/5307901/Convert.UFF.and.Create.Engine-To.GitHub.Issue.zip)

Also remember to change the "numClasses" to your custom quantity.

Try to save your custom model on /usr/local/bin/networks/NEW_FOLDER and run your custom model, I believe it won't work, if it doesn't, try to do this [work around.](https://github.com/dusty-nv/jetson-inference/issues/450#issuecomment-623283704)

Please, tell me if it works.

250zhanghu, 2020-12-13:
> I used python 3.6, maybe that's your problem.

Hello,leandrovrabelo, could you help me with my issues?
I followed your method to convert pb to uff,  but failed.
$ python convert_to_uff.py frozen_model.pb -O NMS -p config.py
2020-12-13 22:44:20.882766: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.10.2
WARNING:tensorflow:Deprecation warnings have been disabled. Set TF_ENABLE_DEPRECATION_WARNINGS=1 to re-enable them.
Loading frozen_model.pb
WARNING:tensorflow:From /usr/lib/python3.6/dist-packages/uff/converters/tensorflow/conversion_helpers.py:274: The name tf.gfile.GFile is deprecated. Please use tf.io.gfile.GFile instead.

NOTE: UFF has been tested with TensorFlow 1.15.0.
WARNING: The version of TensorFlow installed on this system is not guaranteed to work with UFF.
Traceback (most recent call last):
  File "convert_to_uff.py", line 96, in <module>
    main()
  File "convert_to_uff.py", line 92, in main
    debug_mode=args.debug
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/conversion_helpers.py", line 276, in from_tensorflow_frozen_model
    return from_tensorflow(graphdef, output_nodes, preprocessor, **kwargs)
  File "/usr/lib/python3.6/dist-packages/uff/converters/tensorflow/conversion_helpers.py", line 152, in from_tensorflow
    pre = importlib.import_module(os.path.splitext(os.path.basename(preprocessor))[0])
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/nvidia/Desktop/pulsar/models/pb/config.py", line 20, in <module>
    all_assert_nodes = graph.find_nodes_by_op("Assert")
NameError: name 'graph' is not defined

leandrovrabelo, 2020-12-14:
Try to convert the file using this notebook
[Convert UFF and Create Engine-To GitHub Issue.ipynb.zip](https://github.com/dusty-nv/jetson-inference/files/5686158/Convert.UFF.and.Create.Engine-To.GitHub.Issue.ipynb.zip)

closed 2023-04-27 by dusty-nv