dusty-nv/jetson-inference #570: import torch error with python3

flurpo, 2020-05-11:
I ran a test on my torch installation after installing jetson-inference.

>>import torch works perfectly with python but with python3;

tx2:/jetson-inference/build/aarch64/bin$ python3
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 79, in <module>
    from torch._C import *
RuntimeError: module compiled against API version 0xc but this version of numpy is 0xb
>>> 

Has anyone run into this before? Best.

hyattbaker, 2020-05-13:
@flurpo ended up moving back to 4.4 DP and installing the wheel for pytorch and the torchvisin per @dusty-nv's advice in the previous issue we were on.  Currently, retraining with pytorch and ResNet-18 per the example using python 3.6. I will say moving back and forth between Jetpack 4.3 and 4.4 seemed a little buggy for me, with SDK manager failing to flash or just hanging. I ended up having to delete all the downloads from SDK Manager and restarting the manager a couple of times.

flurpo, 2020-05-13:
@hyattbaker -thanks for the quick comment...I've moved back to 4.3. Still  have the issue above but running ok with Py2.7 for the moment. I'll just have to table the problem until I can solve some of my other issues. We are certainly learning something going back and forth between Jetpacks!

closed 2023-04-27 by dusty-nv