dusty-nv/jetson-inference #572: camera-detectnet.py ONNX inference issues

rsamvelyan, 2020-05-15:
Hello
I managed to transfer learning with Pytorch for object detection on my Jetson Nano. I converted the resulting .pth.tar model to ONNX. And then I loaded the model into camera-detectnet.py provided with Jetson Inference library by using --model argument.
Although I was receiving over 80FPS, there was no bounding boxes whatsoever. 
Is there a specific example on how to load an ONNX model and run object detection inference on it with Jetson Inference library?
Thanks

felipevw, 2020-05-19:
Jetson inference library uses under the hood tensorRT. TensorRT offers a ONNX model importer and many other features to perform inference from a neural network. You can try to code a model inferencing engine but be aware that depending on the model layers, you will need a higher version of TensorRT (one not available for the Jetson Nano yet).

Hope it helps

rsamvelyan, 2020-05-19:
Thanks Felipevw. 
Ideally I would like to stick with Jetson Inference and not deal with TensorRT. But so far I haven’t been able to import a custom ONNX model in camera-detectnet.py. There’s always an issue of some sort, such as wrong object id. 
This is why I’m looking for a specific example of using an ONNX model with camera-detectnet.py to see if I’m doing something wrong.

dusty-nv, 2020-05-19:
Hi @rsamvelyan, I had added preliminary ONNX support to detectNet, in the `depth` branch of jetson-inference:

https://github.com/dusty-nv/jetson-inference/tree/depth
https://github.com/dusty-nv/jetson-inference/commit/972cce82e268cdb9cfd9511da2a65f550031e723

And here is the PyTorch code I use for training object detection model:  https://github.com/dusty-nv/pytorch-ssd

I am working on creating the tutorial around it for adding to Hello AI World.

rsamvelyan, 2020-05-19:
**So in regards to: https://github.com/dusty-nv/pytorch-ssd**:
First of all great job, thanks a lot for all your support!!! 
**run_ssd_live_demo.py** worked very well with **MobileNetV1** model.

What it tried then is to export **MobileNetV1** to **ONNX** using **convert_to_onnx.py**
It sort of worked but ended with an error message saying **Segmentation Error**
I tried that resulting ONNX model with **detectnet-camera**. I was getting a very high FPS but no bounding boxes showing. Here is my call:

**detectnet-camera --model=models/mb1-ssd.onnx --class_labels=models/voc-model-labels.txt --width=640 --height=480 --camera=/dev/video0**

And here is a portion from the terminal: 
[TRT]   ../rtSafe/cuda/caskConvolutionRunner.cpp (491) - Cuda Error in execute: 1 (invalid argument)
[TRT]   FAILED_EXECUTION: std::exception
[TRT]   detectNet::Detect() -- failed to execute TensorRT context
[cuda]  cudaEventElapsedTime(&cuda_time, mEventsGPU[evt], mEventsGPU[evt+1])
[cuda]     invalid resource handle (error 400) (hex 0x190)
[cuda]     /home/ryan/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:499

[TRT]   ------------------------------------------------
[TRT]   Timing Report models/mb2-ssd-lite.onnx
[TRT]   ------------------------------------------------
[TRT]   Pre-Process   CPU   0.06724ms  CUDA   0.00672ms
[TRT]   Network       CPU   0.00000ms  CUDA   0.00000ms
[TRT]   Total         CPU   0.06724ms  CUDA   0.00672ms
[TRT]   ------------------------------------------------

Did I do something wrong? 

Do you actually recommend using Jetson Inference library at all or is it better to simply stick with **run_ssd_live_demo.py** and write the rest of my stuff based on that?

dusty-nv, 2020-05-19:
Are you able to get detections from a test image using detectnet-console?   Also, are you sure you are on the `depth` development branch of jetson-inference (after compiling, be sure to `sudo make install` it)

For testing the model, you can use `run_ssd_live_demo.py`, but it doesn't use TensorRT, so the performance won't be as good.

rsamvelyan, 2020-05-19:
I will verify all that tonight for sure. 

So basically if I understand correctly, if I want to do transfer learning on object detection models for my robotics applications, the best is to do it with Pytorch, and then run inference with the live webcam using Jetson Inference, correct?

dusty-nv, 2020-05-19:
That's correct, yes.  jetson-inference uses TensorRT, which will give you the best performance for inferencing.

rsamvelyan, 2020-05-20:
**detectnet-console with ONNX Model taken from dusty-nv/pytorch-ssd:**

**detectnet-console busy_street.jpg busy_street_output.jpg  --model=models/mb1-ssd.onnx --class_labels=models/voc-model-labels.txt** 

detectNet -- loading detection network model from:
          -- prototxt     NULL
          -- model        models/mb1-ssd.onnx
          -- input_blob   'data'
          -- output_cvg   'coverage'
          -- output_bbox  'bboxes'
          -- mean_pixel   0.000000
          -- mean_binary  NULL
          -- class_labels models/voc-model-labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]   TensorRT version 7.1.0
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin creator registration succeeded - ::GridAnchor_TRT
[TRT]   Plugin creator registration succeeded - ::NMS_TRT
[TRT]   Plugin creator registration succeeded - ::Reorg_TRT
[TRT]   Plugin creator registration succeeded - ::Region_TRT
[TRT]   Plugin creator registration succeeded - ::Clip_TRT
[TRT]   Plugin creator registration succeeded - ::LReLU_TRT
[TRT]   Plugin creator registration succeeded - ::PriorBox_TRT
[TRT]   Plugin creator registration succeeded - ::Normalize_TRT
[TRT]   Plugin creator registration succeeded - ::RPROI_TRT
[TRT]   Plugin creator registration succeeded - ::BatchedNMS_TRT
[TRT]   Could not register plugin creator:  ::FlattenConcat_TRT
[TRT]   Plugin creator registration succeeded - ::CropAndResize
[TRT]   Plugin creator registration succeeded - ::DetectionLayer_TRT
[TRT]   Plugin creator registration succeeded - ::Proposal
[TRT]   Plugin creator registration succeeded - ::ProposalLayer_TRT
[TRT]   Plugin creator registration succeeded - ::PyramidROIAlign_TRT
[TRT]   Plugin creator registration succeeded - ::ResizeNearest_TRT
[TRT]   Plugin creator registration succeeded - ::Split
[TRT]   Plugin creator registration succeeded - ::SpecialSlice_TRT
[TRT]   Plugin creator registration succeeded - ::InstanceNormalization_TRT
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - ONNX  (extension '.onnx')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file models/mb1-ssd.onnx.1.1.GPU.FP16.engine
[TRT]   loading network profile from engine cache... models/mb1-ssd.onnx.1.1.GPU.FP16.engine
[TRT]   device GPU, models/mb1-ssd.onnx loaded
[TRT]   Deserialize required 5566586 microseconds.
[TRT]   device GPU, CUDA engine context initialized with 3 bindings
[TRT]   binding -- index   0
               -- name    'input_0'
               -- type    FP32
               -- in/out  INPUT
               -- # dims  4
               -- dim #0  1 (SPATIAL)
               -- dim #1  3 (SPATIAL)
               -- dim #2  300 (SPATIAL)
               -- dim #3  300 (SPATIAL)
[TRT]   binding -- index   1
               -- name    'scores'
               -- type    FP32
               -- in/out  OUTPUT
               -- # dims  3
               -- dim #0  1 (SPATIAL)
               -- dim #1  3000 (SPATIAL)
               -- dim #2  21 (SPATIAL)
[TRT]   binding -- index   2
               -- name    'boxes'
               -- type    FP32
               -- in/out  OUTPUT
[... 26 lines cut ...]
[TRT]   detectNet::Detect() -- failed to render overlay
1 objects detected
detected obj 0  class #0 (BACKGROUND)  confidence=1.000000
bounding box 0  (1587.109375, 533.681030)  (800.269714, 534.015381)  w=-786.839661  h=0.334351

[TRT]   ------------------------------------------------
[TRT]   Timing Report models/mb1-ssd.onnx
[TRT]   ------------------------------------------------
[TRT]   Pre-Process   CPU   0.08599ms  CUDA   0.02302ms
[TRT]   Network       CPU  71.05159ms  CUDA  70.81036ms
[TRT]   Post-Process  CPU   0.05901ms  CUDA   0.05838ms
[cuda]  cudaEventElapsedTime(&cuda_time, mEventsGPU[evt], mEventsGPU[evt+1])
[cuda]     invalid resource handle (error 400) (hex 0x190)
[cuda]     /home/ryan/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:499
[TRT]   Visualize     CPU   0.00000ms  CUDA   0.00000ms
[TRT]   Total         CPU  71.19659ms  CUDA  70.89177ms
[TRT]   ------------------------------------------------

[TRT]   note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

detectnet-console:  writing 1600x1067 image to 'busy_street_output.jpg'
detectnet-console:  successfully wrote 1600x1067 image to 'busy_street_output.jpg'
detectnet-console:  shutting down...
detectnet-console:  shutdown complete

rsamvelyan, 2020-05-20:
**This time I used the uff model taken from the pre-trained models that come with Jetson Inference installation**

**Again, no bounding box on the output photo**

**So all it does is loop non-stop and display:**
[TRT]   detectNet::Detect() -- detected object has invalid classID (879)
[TRT]   detectNet::Detect() -- detected object has invalid classID (76824)
[TRT]   detectNet::Detect() -- detected object has invalid classID (76824)
...

**detectnet-console busy_street.jpg busy_street_output.jpg  --model=models/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff --class_labels=models/SSD-Mobilenet-v2/ssd_coco_labels.txt**

detectNet -- loading detection network model from:
          -- prototxt     NULL
          -- model        models/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'data'
          -- output_cvg   'coverage'
          -- output_bbox  'bboxes'
          -- mean_pixel   0.000000
          -- mean_binary  NULL
          -- class_labels models/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]   TensorRT version 7.1.0
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin creator registration succeeded - ::GridAnchor_TRT
[TRT]   Plugin creator registration succeeded - ::NMS_TRT
[TRT]   Plugin creator registration succeeded - ::Reorg_TRT
[TRT]   Plugin creator registration succeeded - ::Region_TRT
[TRT]   Plugin creator registration succeeded - ::Clip_TRT
[TRT]   Plugin creator registration succeeded - ::LReLU_TRT
[TRT]   Plugin creator registration succeeded - ::PriorBox_TRT
[TRT]   Plugin creator registration succeeded - ::Normalize_TRT
[TRT]   Plugin creator registration succeeded - ::RPROI_TRT
[TRT]   Plugin creator registration succeeded - ::BatchedNMS_TRT
[TRT]   Could not register plugin creator:  ::FlattenConcat_TRT
[TRT]   Plugin creator registration succeeded - ::CropAndResize
[TRT]   Plugin creator registration succeeded - ::DetectionLayer_TRT
[TRT]   Plugin creator registration succeeded - ::Proposal
[TRT]   Plugin creator registration succeeded - ::ProposalLayer_TRT
[TRT]   Plugin creator registration succeeded - ::PyramidROIAlign_TRT
[TRT]   Plugin creator registration succeeded - ::ResizeNearest_TRT
[TRT]   Plugin creator registration succeeded - ::Split
[TRT]   Plugin creator registration succeeded - ::SpecialSlice_TRT
[TRT]   Plugin creator registration succeeded - ::InstanceNormalization_TRT
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - UFF  (extension '.uff')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file models/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.GPU.FP16.engine
[TRT]   loading network profile from engine cache... models/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.GPU.FP16.engine
[TRT]   device GPU, models/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff loaded
[TRT]   Deserialize required 3207073 microseconds.
[TRT]   device GPU, CUDA engine context initialized with 3 bindings
[TRT]   binding -- index   0
               -- name    'Input'
               -- type    FP32
               -- in/out  INPUT
               -- # dims  3
               -- dim #0  3 (SPATIAL)
               -- dim #1  300 (SPATIAL)
               -- dim #2  300 (SPATIAL)
[TRT]   binding -- index   1
               -- name    'NMS'
               -- type    FP32
               -- in/out  OUTPUT
               -- # dims  3
               -- dim #0  1 (SPATIAL)
               -- dim #1  100 (SPATIAL)
               -- dim #2  7 (SPATIAL)
[TRT]   binding -- index   2
               -- name    'NMS_1'
               -- type    FP32
               -- in/out  OUTPUT
               -- # dims  3
               -- dim #0  1 (SPATIAL)
               -- dim #1  1 (SPATIAL)
               -- dim #2  1 (SPATIAL)
[TRT]   INVALID_ARGUMENT: Cannot find binding of given name: data
[TRT]   binding to input 0 data  binding index:  -1
[TRT]   Parameter check failed at: engine.cpp::getBindingDimensions::1977, condition: bindIndex >= 0 && bindIndex < getNbBindings()
[TRT]   binding to input 0 data  dims (b=1 c=1 h=1 w=1) size=4
[TRT]   INVALID_ARGUMENT: Cannot find binding of given name: coverage
[TRT]   binding to output 0 coverage  binding index:  -1
[TRT]   Parameter check failed at: engine.cpp::getBindingDimensions::1977, condition: bindIndex >= 0 && bindIndex < getNbBindings()
[TRT]   binding to output 0 coverage  dims (b=1 c=1 h=1 w=1) size=4
[TRT]   INVALID_ARGUMENT: Cannot find binding of given name: bboxes
[TRT]   binding to output 1 bboxes  binding index:  -1
[TRT]   Parameter check failed at: engine.cpp::getBindingDimensions::1977, condition: bindIndex >= 0 && bindIndex < getNbBindings()
[TRT]   binding to output 1 bboxes  dims (b=1 c=1 h=1 w=1) size=4
device GPU, models/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff initialized.
W = 1  H = 1  C = 1
detectNet -- maximum bounding boxes:  1
detectNet -- loaded 91 class info entries
detectNet -- number of object classes:  91
[image] loaded 'busy_street.jpg'  (1600 x 1067, 3 channels)
[TRT]   detectNet::Detect() -- detected object has invalid classID (879)
[TRT]   detectNet::Detect() -- detected object has invalid classID (76824)
[TRT]   detectNet::Detect() -- detected object has invalid classID (76824)
[TRT]   detectNet::Detect() -- detected object has invalid classID (403)
[TRT]   detectNet::Detect() -- detected object has invalid classID (652)
[TRT]   detectNet::Detect() -- detected object has invalid classID (1513)
[TRT]   detectNet::Detect() -- detected object has invalid classID (1513)

rsamvelyan, 2020-05-20:
Now I'm going to check if I'm using the depth development branch of jetson-inference

rsamvelyan, 2020-05-20:
I quickly compared the c++ source of https://github.com/dusty-nv/jetson-inference/tree/depth
972cce8 in my jetson inference installation, and I see differences..
I'm going to re-install it, but this time from the depth

rsamvelyan, 2020-05-20:
**I made the changes to detectNet.cpp file as per **972cce8** but it didn't compile properly.
modelTypeFromPath(model) was not previously declared. 
Am I missing something in this new branch?**

/home/ryan/jetson-inference/c/detectNet.cpp: In member function ‘bool detectNet::init(const char*, const char*, const char*, const char*, float, const char*, const char*, const char*, uint32_t, precisionType, deviceType, bool)’:
**/home/ryan/jetson-inference/c/detectNet.cpp:114:6: error: ‘modelTypeFromPath’ was not declared in this scope
  if( modelTypeFromPath(model) == MODEL_ONNX )**
      ^~~~~~~~~~~~~~~~~
/home/ryan/jetson-inference/c/detectNet.cpp:114:6: note: suggested alternative: ‘modelTypeFromStr’
  if( modelTypeFromPath(model) == MODEL_ONNX )
      ^~~~~~~~~~~~~~~~~
      modelTypeFromStr
/home/ryan/jetson-inference/c/detectNet.cpp:115:3: error: ‘mWorkspaceSize’ was not declared in this scope
   mWorkspaceSize = 2048 << 20;
   ^~~~~~~~~~~~~~
CMakeFiles/jetson-inference.dir/build.make:97: recipe for target 'CMakeFiles/jetson-inference.dir/c/detectNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/detectNet.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

rsamvelyan, 2020-05-20:
Dusty, I compiled and installed Jetson Inference from Depth branch.

Here's a quick question: 
If I want to use my custom mobilenet-v1 model with Jetson Inference, in the call for **detectnet-camera.py** do I still need to specify the --network parameter or do I omit it?

In other words which one of these two is correct?

1- python3 my-detectnet-camera.py **--network=ssd-mobilenet-v1** --model=models/mb1-ssd.onnx --class_labels=models/voc-model-labels.txt 

OR

2- python3 my-detectnet-camera.py --model=models/mb1-ssd.onnx --class_labels=models/voc-model-labels.txt 

Thanks!!

closed 2020-05-21 by rsamvelyan