dusty-nv/jetson-inference #573: How to get a detection label?

gg15c, 2020-05-17:
Dear All,
I just want to get a label information from the detectNet. For example, if there is a person label, I want to catch it for reference. Here is my code.

while display.IsOpen():
	img, width, height = camera.CaptureRGBA()
	detections = net.Detect(img, width, height)
	display.RenderOnce(img, width, height)
	display.SetTitle("Object Detection | Network {:.0f}FPS".format(net.GetNetworkFPS()))
 
I also see someone put a function called "net.GetClassDesc(detections.classID)", but it does not work for me. 

The system gave a message as following.

AttributeError: 'list' object has no attribute 'classID'

My question is how to get a label?

Thanks, gg15c

felipevw, 2020-05-19:
Well I looked at the code and I suppose that the line that outputs the label is:

`sprintf(str, "TensorRT %i.%i.%i | %s | %s | Network %.0f FPS", NV_TENSORRT_MAJOR, NV_TENSORRT_MINOR, NV_TENSORRT_PATCH, net->GetNetworkName(), precisionTypeToStr(net->GetPrecision()), net->GetNetworkFPS());`

Shouldn't it be `net->GetNetworkName()` ?
If this object has no attribute, I suggest to take a glance at the object definition. It may help you to understand what is going on and there might be comments explaining it.

Hope it helps

gg15c, 2020-05-19:
Dear felipevw,

Actually, I found the problem and solved it. 

I need to print it on the screen when needing a label for further reference.

=> print(net.GetClassDesc(detections.classID)) 

This command is correct because the GetClassDesc will map a result for you.

For example, if you put 1 in the argument, the system will give you a person label.

Hope it helps

closed 2023-04-27 by dusty-nv