dusty-nv/jetson-inference #579: camera-console.py ONNX issues

qjk5736, 2020-05-21:
Dusty. I've been working through your tutorials and ran into a problem under "Collecting Your Own Datasets" that I am hoping you can clear up for me. I generated a resnet18.onnx files as described and then tried to use it with camera_console.py as follows.

./imagenet-console.py --model=$MYMODEL/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=$DATASET/labels.txt ./rhino_0.jpg ./rhino_0_out.jpg

I get the following results (at the end ofprocess):

device GPU, /home/user/jetson-inference/python/training/classification/testCreatingDatasets/resnet18.onnx initialized.
[TRT]   /home/user/jetson-inference/python/training/classification/testCreatingDatasets/resnet18.onnx loaded
imageNet -- loaded 6 class info entries
imageNet -- didn't load expected number of class descriptions  (6 of 4)
imageNet -- failed to load synset class descriptions  (6 / 6 of 4)
[TRT]   imageNet -- failed to initialize.
jetson.inference -- imageNet failed to load built-in network 'resnet-18'
PyTensorNet_Dealloc()
Traceback (most recent call last):
  File "./imagenet-console_jik.py", line 50, in <module>
    net = jetson.inference.imageNet(opt.network, sys.argv)
Exception: jetson.inference -- imageNet failed to load network
jetson.utils -- freeing CUDA mapped memory

If I put in --network=resnet-18 then the classification appears to run though but ignores my model. Can you provide some insight here? Thanks.

dusty-nv, 2020-05-21:
Hi @qjk5736, from this log text, it looks like your `labels.txt` file lists 6 classes, but your model only supports 4 classes:

```
imageNet -- loaded 6 class info entries
imageNet -- didn't load expected number of class descriptions (6 of 4)
imageNet -- failed to load synset class descriptions (6 / 6 of 4)
```

Can you remove the classes from your `label.txt` that are not in your model?

qjk5736, 2020-05-21:
The labels.txt I point to in the command line only has four labels as shown below. Do you mean I need to change the labels.txt somewhere else?
background
bear
elephant
rhino

maggie450, 2020-11-26:
I had a similar issue (5 of 4).  It was fixed by removing the final new-line character in labels.txt.

closed 2023-04-27 by dusty-nv