dusty-nv/jetson-inference #580: Depth branch - issues with webcam

rsamvelyan, 2020-05-21:
I'm having an isssue everytime I ran detectnet-camera. 
The webcam window opens up, and after a couple seconds the image freezes and I get this message:

1 objects detected
detected obj 0  class #1 (person)  confidence=0.924142
bounding box 0  (83.388382, 98.156975)  (449.361633, 479.830536)  w=365.973267  h=381.673553
X Error of failed request:  BadCursor (invalid Cursor parameter)
  Major opcode of failed request:  2 (X_ChangeWindowAttributes)
  Resource id in failed request:  0xbe179ed1
  Serial number of failed request:  163
  Current serial number in output stream:  165

rsamvelyan, 2020-05-21:
I resolved it. 
The issue was that the pre-trained models which came with the jetson-inference installation in usr/local/bin/networks directory didn't have write permissions. So the GPU engine(ssd_mobilenet_v1_coco.uff.1.1.GPU.FP16.engine) was not being created since detectnet-camera was run without root permissions. 

I changed the permissions in these folders, and after the GPU engines got saved the problem didn't re-occur. 
It's still weird that it was happening in first place.

closed 2020-05-21 by rsamvelyan

dusty-nv, 2020-05-21:
> X Error of failed request: BadCursor (invalid Cursor parameter)
> Major opcode of failed request: 2 (X_ChangeWindowAttributes)
> Resource id in failed request: 0xbe179ed1
> Serial number of failed request: 163
> Current serial number in output stream: 165

This was a different error I believe, I just fixed it in the `depth` dev branch via commit [`e27a83`](https://github.com/dusty-nv/jetson-inference/commit/e27a8319b27cd1160b9098eb1bc0e712654dcaa2).