dusty-nv/jetson-inference #582: convert_to_onnx.py

rsamvelyan, 2020-05-22:
I tried to convert the **mobilenet-v1-ssd-mp-0_675.pth** (from  https://storage.googleapis.com/models-hao/mobilenet-v1-ssd-mp-0_675.pth)  to ONNX to run inference with jetson-inference with a live camera. 

Here is my call to the python script: 
`python3 convert_to_onnx.py mb1-ssd ./models/mobilenet-v1-ssd-mp-0_675.pth ./models/voc-model-labels.txt`

It ended up with segmentation error and Python crashed. 

It still created the ONNX file. 

Am I doing something wrong here?

Here is the screenshot:
https://drive.google.com/open?id=1mV5negN6YC1UbcnlAZuHSojoTzphhN7y

dusty-nv, 2020-05-22:
> It ended up with segmentation fault and Python crashed.
> It still created the ONNX file.

You can ignore the crash for now - I am debugging why sometimes PyTorch is crashing when it exits.  It doesn't have to do with running PyTorch, just when unloading the PyTorch module after everything has already ran.

rsamvelyan, 2020-05-22:
OK great, that's what I figured. It must be like killing OpenCV webcam capture. Very often it crashes.
The ONNX file worked perfectly, so that's what counts.

closed 2020-05-22 by rsamvelyan