dusty-nv/jetson-inference #586: Load custom model to the TensorRT

msts35, 2020-05-26:
Hello dusty, 

I want to use the fast inference feature on my project , I have trained a model that will be introduced below and I converted it to the onnx , however I could not load the model into TensorRT it will give me error.

Also firstly I tried to convert frozen graph to uff , however this process lead a lot of error with unsupported layers in ssd mobilenet and ssd inception models.

I think the problem is the Jetpack , I have not latest version that supports TensorRT 6, which have support to the UINT8, but I am not sure the update will be the solution, I do not have a sd card adapter so I could not flash my microsd , I will but it tomorrow.

I am too new to create a CNN from strach, so I use fine-tuning techniques to produce good results, do you have any link to a model that can work with tensorRT ? 

Can you give me Framework advice to train and use object detection models with a custom dataset and use it with this repository or anything else? 

Framework = Tensorflow object detection API
tried Models = Ssd inception v2 from model zoo, SSD mobilenet v1 
Error = Unsupported ONNX data type : UINT8 (2)
TensorRT version = 5.1
Tensorflow version  = 1.13

Thank you for the repository, we love Nvidia teams!

leandrovrabelo, 2020-05-26:
Maybe you should try these steps https://github.com/AastaNV/TRT_object_detection

The inference runs pretty well on their code, I'm still trying to make it work with DetecNet.

msts35, 2020-05-27:
@leandrovrabelo thank you for the information, I think they use models from tensorflow model zoo right? did you try this?

leandrovrabelo, 2020-05-27:
Yes, they use it, i'm running a ssd mobilenet v2, so far these are their supported models:

ssd_inception_v2_coco_2017_11_17
ssd_mobilenet_v1_coco
ssd_mobilenet_v2_coco

msts35, 2020-05-27:
Mr. Leandro does the repo can be used with a custom object detection model which trained on tensorflow API , and can be used to inference on a video stream ?

leandrovrabelo, 2020-05-27:
Yes.

msts35, 2020-05-27:
I see you also cannot implement tensorrt with onnx or uff conversations,

I think now to switch to the DetectNet framework to train my models, should
I do ?

Here is a good example

https://github.com/NVIDIA/DIGITS/tree/master/examples/object-detection

Leandro Vieira Rabelo da Silva <notifications@github.com>, 27 May 2020 Çar,
15:09 tarihinde şunu yazdı:

dusty-nv, 2020-05-27:
> I think now to switch to the DetectNet framework to train my models, should I do ?

DetectNet model architecture isn't recommended to use going forward - perhaps you meant DarkNet (YOLO), in which case yes that is an option.  YOLO isn't explicitly supported by this jetson-inference repo but there are TensorRT samples (included under `/usr/src/tensorrt/samples`) that show it being used with TensorRT.

You could also try this PyTorch SSD-Mobilenet implementation for re-training object detection:  https://github.com/dusty-nv/pytorch-ssd

I am making that PyTorch repo into a new tutorial part of Hello AI World for re-training object detection models with SSD-Mobilenet.

msts35, 2020-05-27:
Hi Dustin, thanks for the reply,

Excuse me to give wrong info, I wanted to say DIGITS when I say DetectNet
framework to teach detection network, however I will think about the
pytorch.

What do you think about DIGITS? is it worth trying?

Dustin Franklin <notifications@github.com>, 27 May 2020 Çar, 22:15
tarihinde şunu yazdı:

msts35, 2020-05-27:
I think, What the community need  is the complete tutorial from labelling
to inference with , every tutorial uses ready to use datasets or models,
there is no full tutorial about that.

Mesut Şimşek <mesutttsimsek@gmail.com>, 28 May 2020 Per, 01:11 tarihinde
şunu yazdı:

closed 2020-05-28 by msts35

flurpo, 2020-06-09:
Dusty-nv, look forward to your pytorch repo. I am trying to do exactly what you described above. Is the detectNet architecture considered obsolete? Best.