dusty-nv/jetson-inference #587: Cat_dog Segmentation fault (core dumped) at the end

Philliec459, 2020-05-27:
Hello. I love your GitHub site. Great tools and you give an excellent explanation of how to ready the Jetson NX for all of this. I am on the cat_dog training now. I have run the cat_dog training and created the onnx model, but the end of each process i get a  Segmentation fault (core dumped). I trained for 35 epochs and all went well and at the end during what I think is the process of writing the checkpoint and Model_best tar files is where I got my first core dump. However, the tar files were saved. It was all GPU on the training and then CPU was used most at the end. 

The second time I had the core dump was when I created the onnx mode and was writing the model to disk. I used your command:

python onnx_export.py --model-dir=cat_dog

It went through the model creation and wrote out the onnx model, but then the program hung up eventually with the dreaded error Segmentation fault (core dumped). 

I am using a Jetson NX with 256 GB micro sd card. I have an ssd on the way. Could it be that writing to the micro sd card is the issue?

dusty-nv, 2020-05-27:
Hi @Philliec459 , you can ignore the seg faults, it is an issue I am looking into that only happens when PyTorch unloads (i.e. after the program has completed running)

Philliec459, 2020-05-27:
Thank you dusty-nv. That is good to hear. I have all the new models and checkpoints and could use the trained data, I just did not understand the error. No problem.  Thanks to you we now know how to do image classification. 

Do you have a tutorial on Object Detection (your way) that demonstrates the preparation of the dataset and takes advantage of the Jetson NX GPU capabilities? I have been using Detectron2 with my mac (and Colab for training) for a few months now to do Object Detection and Instance Segmentation.  I would like to use your techniques to do the same training my marine database using the NX.

dusty-nv, 2020-05-27:
> Do you have a tutorial on Object Detection (your way) that demonstrates the preparation of the dataset and takes advantage of the Jetson NX GPU capabilities?

I have been preparing a new tutorial to add to Hello AI World on re-training object detection with PyTorch - here is the PyTorch code awhile:  https://github.com/dusty-nv/pytorch-ssd

For inferencing of the re-trained model, the support is currently in the `depth` dev branch of this repo, see here for more info:  https://github.com/dusty-nv/jetson-inference/issues/572#issuecomment-630957050

Philliec459, 2020-05-27:
Thank you. I will look into this now.

Philliec459, 2020-06-06:
Hi Dusty-nv, Per your suggestion I have been working on your pytorch-ssd repository and it is working great. I have used your gun training example and tried to back-engineer into how you setup your training data. I am using 'buoys' and 'boats' for my training. This of course has led to some questions. I would have asked the questions on the pytorch-ssd repository, but see no way to do this. 

It appears that when you start to train the train_ssd.py only loads sub-train-annotations-bbox.csv and the sub-test-annotations-bbox.csv. The other csv files in the data/open_images in the guns example do not appear to be loaded. Is this correct?

I have used labelimg.py to box in my marine objects, and this creates an xml fiel for each image file. I have a program downloaded from GitHub (?) called xml_to_csv.py that creates the sub-train-annotations-bbox.csv or sub-test-annotations-bbox.csv files that are needed for training. There are a few difference in what this creates vs. what you had. My program uses pixels for xmin and xmax, ymin and ymax, but you appear to be using decimal values and not pixels. Correct?

Finally, in your sub-train/test.... csv files, I am wondering what columns are required. ImageID of course and the min and max values for x/y, and ClassName. Are you using LabelName to compare to the class-descriptions-boxable.csv?  

I have about 315 training images and 50 test images, and the results look very good, but I want to make sure that I am not missing anything. I am working on a Jetson NX.  The training is quite fast and video fps appears to be good.

manrayboy, 2020-07-19:
@dusty-nv  Any response to this question. I as long as it seems a few others have been having these same issues with training a custom object detector using imagelabel

dusty-nv, 2020-07-19:
Hi manarayboy, here is the actual tutorial that goes with this:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md

I am not sure about imagelabel.py, was that in pytorch-ssd?  If so I removed it.  It was a forked project.

For custom dataset, I use Pascal VOC format, which the LabelImg project can make annotations for.

manrayboy, 2020-07-19:
Thanks @dusty-nv

Tomasm64, 2022-02-07:
I'm getting the segmentation fault before it even starts the epoch

dusty-nv, 2022-02-08:
Hi @Tomasm64, can you provide the log from the terminal?   Is the board running out of memory?

Tomasm64, 2022-02-08:
Hi @dusty-nv I tried to replicate the issue, but I did not get a segmentation fault this time. I posted the issue on the nvidia forums.

dusty-nv, 2022-02-08:
@Tomasm64 can you link me to the forum issue?  Thanks.

Tomasm64, 2022-02-08:
@dusty-nv  https://forums.developer.nvidia.com/t/jetson-inference-cannot-train-model-with-custom-data-set/202871

dusty-nv, 2022-02-08:
Thanks @Tomasm64, I've replied to your topic there on the forums.