dusty-nv/jetson-inference #588: Build FAIL with Python3.7

elPrac, 2020-05-28:
Hi everyone!

I wish someone could give me an idea of how to solve the following problem...

I'm trying to build jetson-inference with a specific Python3.7.5 version 
the way i installed/compiled python was from source code and installed with:
```
make install
```
So **/usr/local/bin/python3** now points to 3.7.5 version, then i followed jetson-inference build steps
skipping pytorch installation, then when i ran make i get the following:

```bash
[ 81%] Built target jetson-utils-python-27
[ 81%] Built target jetson-utils-python-36
[ 81%] Linking CXX shared library ../../../aarch64/lib/python/3.7/jetson_utils_python.so
[ 81%] Built target jetson-inference-python-27
[ 81%] Built target jetson-inference-python-36
[ 81%] Built target imagenet-console
[ 81%] Linking CXX shared library ../../aarch64/lib/python/3.7/jetson_inference_python.so
[ 81%] Built target imagenet-camera
[ 81%] Built target detectnet-console
[ 81%] Built target detectnet-camera
[ 81%] Built target segnet-console
[ 81%] Built target superres-console
[ 81%] Built target segnet-camera
[ 81%] Linking CXX executable ../../aarch64/bin/trt-console
[ 81%] Built target trt-bench
[ 81%] Built target trt-console
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.7m.a(object.o): In function `_PyObject_GetMethod':
/home/prac/Python-3.7.5/Objects/object.c:1135:(.text+0x238c): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.7m.a(object.o): In function `_PyObject_LookupAttr':
/home/prac/Python-3.7.5/Objects/object.c:936:(.text+0x2a00): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/prac/Python-3.7.5/Objects/object.c:936:(.text+0x2b48): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/prac/Python-3.7.5/Objects/object.c:936:(.text+0x2c80): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/prac/Python-3.7.5/Objects/object.c:936:(.text+0x2dc0): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(object.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyObject_GenericGetAttr' which may bind externally can not be used when making a shared object; recompile with -fPIC
/home/prac/Python-3.7.5/Objects/object.c:936:(.text+0x2f10): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(typeobject.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyType_GenericAlloc' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.7m.a(typeobject.o): In function `type_new':
/home/prac/Python-3.7.5/Objects/typeobject.c:2805:(.text+0x109f4): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(typeobject.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `PyType_GenericAlloc' which may bind externally can not be used when making a shared object; recompile with -fPIC
...
```

I'm not sure why i'm getting the **dangerous relocation: unsupported relocation** message.
About **/usr/local/lib/libpython3.7m.a** i'm not sure if this was built when i installed python3.7.5 but looks like there is a problem with it, message say **recompile with -fPIC** but i'm not sure if i need to recompile libpython3.7m.a object with this flag or should i need to add this to a jetson-inference build rule...

Anyway, i will really appreaciate any clue on what could be the problem...
Thanks!

elPrac, 2020-05-28:
It was because the way i built python3
should use:
```
./configure --enable-shared --enable-optimizations
```

closed 2020-05-28 by elPrac

Yao1up, 2020-09-26:
> It was because the way i built python3
> should use:
> 
> ```
> ./configure --enable-shared --enable-optimizations
> ```

hey, I have the same problem, where did you use this command, thank you

Yao1up, 2020-09-26:
> It was because the way i built python3
> should use:
> 
> ```
> ./configure --enable-shared --enable-optimizations
> ```

 Linking CXX shared library ../../../x86_64/lib/python/3.6/jetson_utils_python.so
/usr/bin/ld: /usr/local/lib/libpython3.6m.a(boolobject.o): relocation R_X86_64_32 against `.data' can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.6m.a: error adding symbols: Bad value

could you help me

AbdallahAlhaddad, 2021-01-29:
**He has built python3.7 from source with `--enable-shared` flag**
you can do this by typing the following commands: 
```bash
$ cd ~/Desktop

$ wget https://www.python.org/ftp/python/3.7.9/Python-3.7.9.tgz

$ tar -xf Python-3.7.9.tgz

$ cd Python-3.7.9

$ ./configure --enable-shared --enable-optimizations

$ make

$ sudo make install 

$ sudo ldconfig /usr/local/lib
```

and then try to to build `jetson-inference` again by following nvidia guide.