dusty-nv/jetson-inference #589: Please make it clear that a special patched version of Torchvision is required

gituuu, 2020-05-28:
For DIGITS, please make it clear that a special patched version of Torchvision is required. It looks like it is the normal 0.3.0 version but it is not and it would be really helpful if you made it clear on this page:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo.md

dusty-nv, 2020-05-28:
Hi @gituuu - thanks for your feedback, I will do that.  Note the forked version is automatically installed by the script during `cmake` configuration, and if you are on the latest JetPack 4.4 the forked torchvision is no longer required and the upstream version of torchvision is installed.

gituuu, 2020-05-29:
Great. I am waiting for 4.4 to be "stable" before I move to it. I think it is only "developer preview" at the moment.

closed 2023-04-27 by dusty-nv