dusty-nv/jetson-inference #592: MobileNet-v1 transfer learning LOSS = NAN

rsamvelyan, 2020-05-28:
I was training the mobilenet-v1 on my Jetson Nano on a custom data-set in Open Image format.
What surprised me is that during the training I was not seeing any LOSS. It was all NAN.
After the training I ran inference on the model but it wouldn't detect any object at all whatsoever. 
Nevertheless though is that I did everything exactly the same way as in handgun training tutorial. Only the data-set was different. 

Any idea what could've gone wrong?

```
python3 train_ssd.py \
> --dataset_type open_images \
> --datasets ~/data/other_datasets/smarties \
> --net mb1-ssd \
> --pretrained_ssd models/mb1-ssd/mobilenet-v1-ssd-mp-0_675.pth \
> --scheduler cosine \
> --lr 0.01 \
> --t_max 100 \
> --validation_epochs 5 \
> --num_epochs 100 \
> --base_net_lr 0.001  \
> --batch_size 8

```

```
2020-05-28 17:47:19,761 - root - INFO - Epoch: 0, Validation Loss: nan, Validation Regression Loss nan, Validation Classification Loss: nan
2020-05-28 17:47:19,966 - root - INFO - Saved model models/mb1-ssd-Epoch-0-Loss-nan.pth
2020-05-28 17:50:08,128 - root - INFO - Epoch: 5, Validation Loss: nan, Validation Regression Loss nan, Validation Classification Loss: nan
2020-05-28 17:50:08,371 - root - INFO - Saved model models/mb1-ssd-Epoch-5-Loss-nan.pth
2020-05-28 17:52:50,017 - root - INFO - Epoch: 10, Validation Loss: nan, Validation Regression Loss nan, Validation Classification Loss: nan
2020-05-28 17:52:50,161 - root - INFO - Saved model models/mb1-ssd-Epoch-10-Loss-nan.pth
2020-05-28 17:55:35,288 - root - INFO - Epoch: 15, Validation Loss: nan, Validation Regression Loss nan, Validation Classification Loss: nan
2020-05-28 17:55:35,436 - root - INFO - Saved model models/mb1-ssd-Epoch-15-Loss-nan.pth
2020-05-28 17:58:21,034 - root - INFO - Epoch: 20, Validation Loss: nan, Validation Regression Loss nan, Validation Classification Loss: nan
2020-05-28 17:58:21,187 - root - INFO - Saved model models/mb1-ssd-Epoch-20-Loss-nan.pth
2020-05-28 18:01:04,537 - root - INFO - Epoch: 25, Validation Loss: nan, Validation Regression Loss nan, Validation Classification Loss: nan
2020-05-28 18:01:04,679 - root - INFO - Saved model models/mb1-ssd-Epoch-25-Loss-nan.pth

```

rsamvelyan, 2020-05-29:
I messed up the bounding boxes. After fixing the annotations, everything worked.

closed 2020-05-29 by rsamvelyan

pradhunmya01, 2020-11-17:
Hi @dusty-nv @rsamvelyan  I am also getting the same issue but here it comes while saving the model it saves like:- 
mb1-ssd-Epoch-0-Loss-inf.pth
mb1-ssd-Epoch-1-Loss-inf.pth
mb1-ssd-Epoch-2-Loss-inf.pth
[image]

I have checked my annotations it seems fine, any other issues in that?