dusty-nv/jetson-inference #597: Custom detection model with Jetson Inference

denizcelikgh, 2020-05-31:
Hi,

In the last week, i tried to use Hello World AI with detectNet for real time object detection and i did it. I used SSD-MobileNet-V2 like on this page. Thanks for this beautiful guide.
[https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-example-2.md](url)

Its working well in my project code with about 10-12 fps on my Jetson Nano. 
And now, I need to use custom 15 classes detection model (dataset). I have read many pages about that. I followed the issues. But i can't find the way that i should go. 

Where should i start? How can i do this with Jetson Inference? Or, should i try to do this with an another way like TensorFlow Lite. I guess i'm lost. A lot of thanks.

denizcelikgh, 2020-06-02:
Hi Dusty, 

With your recommendations, i'm trying to use this repo and instructions for train a custom object detection model
[https://github.com/dusty-nv/pytorch-ssd](url)

But i can't find a way for installation of repo libraries like this:
`from vision.ssd.vgg_ssd import create_vgg_ssd, create_vgg_ssd_predictor
from vision.ssd.mobilenetv1_ssd import create_mobilenetv1_ssd, create_mobilenetv1_ssd_predictor
from vision.ssd.mobilenetv1_ssd_lite import create_mobilenetv1_ssd_lite create_mobilenetv1_ssd_lite_predictor
from vision.ssd.squeezenet_ssd_lite import create_squeezenet_ssd_lite, create_squeezenet_ssd_lite_predictor
from vision.ssd.mobilenet_v2_ssd_lite import create_mobilenetv2_ssd_lite, create_mobilenetv2_ssd_lite_predictor
from vision.utils.misc import Timer`

And so, i can't run the demo files. What should i do?

SusanaPineda, 2020-06-25:
Hi!
I am trying to do the same. Have you managed to do it?

SusanaPineda, 2020-06-25:
I have found this page where they explain very well how to create the .prototxt file for multiple classes.
https://towardsdatascience.com/how-to-adjust-detectnet-a9ad0452d27f

You can use it to train your own network with DIGITS.
Hope this can help you :)

denizcelik, 2020-06-26:
Hi!
No, I couldn't do it yet. 
Thank you! I think it can help. :)