dusty-nv/jetson-inference #598: visualization bug with OpenGL broken

MoonBlvd, 2020-06-02:
Hi, 
Thank you for making this great tutorial!
I have installed the project but when I try to run the detectnet-camera demo, I got this error 

    MESA-LOADER: failed to open swrast (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
    libGL error: failed to load driver: swrast
    X Error of failed request:  GLXBadContext
    Major opcode of failed request:  154 (GLX)
    Minor opcode of failed request:  6 (X_GLXIsDirect)
    Serial number of failed request:  40
    Current serial number in output stream:  39

It seems that my OpenGL lin is broken? Have you ever encountered this issue?

Thank you!

flurpo, 2020-06-18:
Hi MoonBlvd,

I have a similar error when trying to open a window through SSH/Xserver from both Nano and TX2. 

I found this link and I'm following;
https://forums.developer.nvidia.com/t/jetpack-4-3-mesa-loader-failed-to-open-swrast-while-in-xrdp-session/111199

Best

touchmii, 2021-01-08:
export DISPLAY=:0

dusty-nv, 2021-01-08:
> It seems that my OpenGL lin is broken? Have you ever encountered this issue?

You may be able to fix it by installing this:  https://github.com/NVIDIA/libglvnd
Then test with glxinfo / glxgears.

Otherwise, you may need to re-flash your SD card with JetPack to fix OpenGL.