dusty-nv/jetson-inference #599: threshold hard coded in jetson.inference.detectNet()?

bl6594, 2020-06-03:
The default value for the threshold is 0.5, but when I changed the default value to 0.2 in the code, or modified it through --threshold 0.2.  The detectNet was still loading the model using 0.5.
I even tried to change the line of code below 
net = jetson.inference.detectNet("ssd-inception-v2", sys.argv, opt.threshold)
to  
net = jetson.inference.detectNet("ssd-inception-v2", sys.argv, 0.2) 

the following part was printed to the terminal when running the code.
detectNet -- loading detection network model from:
          -- model        networks/SSD-Inception-v2/ssd_inception_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Inception-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

dusty-nv, 2020-06-03:
Hi @bl6594, can you try the below instead:

`net = jetson.inference.detectNet("ssd-inception-v2", threshold=0.2)`

Or it should work with the original code by using the command-line argument `--threshold=0.2`

And you can confirm the threshold later with:  `print(net.GetThreshold())`

bl6594, 2020-06-03:
@dusty-nv, Thank you for your quick reply!  Yes, it works after changing 
net = jetson.inference.detectNet("ssd-inception-v2", sys.argv, opt.threshold)
to 
net = jetson.inference.detectNet("ssd-inception-v2", opt.threshold)

closed 2020-06-03 by bl6594