dusty-nv/jetson-inference #604: Mobilenet V1 SSD - Small Object Detection issue

rsamvelyan, 2020-06-05:
I successfully did transfer learning on custom object detection with Mobilenet V1 SSD. I used Pytorch SSD train.py provided with this repo and my custom dataset (Open Images format). 
I'm getting very high confidence levels but as soon as the object becomes smaller than a certain size even if it is showing clearly the SSD won't detect it at all.   

Is there any way to re-configure the SSD model to make it look for smaller objects as well?

Thanks

leandrovrabelo, 2020-06-05:
i've being handling the same problem with my custom object detection and I'll try to label and train this small object as another class, I don't know if it works for you, but I think this is an idea.

rsamvelyan, 2020-06-05:
I noticed that as I add more pictures to the dataset the confidence level for smaller objects rises. 

But I like your idea. I guess in that case the new class for the smaller version of the object should be trained on small objects, right?

I wonder if there is a way of changing the sizes of the SSD pre-fabricated (not sure of the correct terminology) detection boxes to make the small ones smaller...

leandrovrabelo, 2020-06-05:
> But I like your idea. I guess in that case the new class for the smaller version of the object should be trained on small objects, right?

Yes, that's the idea.

> I wonder if there is a way of changing the sizes of the SSD pre-fabricated (not sure of the correct terminology) detection boxes to make the small ones smaller...

maybe [this](https://stackoverflow.com/questions/60537788/how-to-modify-ssd-mobilenet-config-to-detect-small-objects-using-tensorflow-obje) should help you.