dusty-nv/jetson-inference #605: Make the inference window size same as the video size?

rsamvelyan, 2020-06-05:
When I run inference with a live webcam the window that contains the webcam feed with the bounding boxes takes the full screen while my webcam video is 640x480.
So I end up with a huge black window and a tiny video in the top-left corner of the screen.

Is there a way to make that window be same size as the video?

dusty-nv, 2020-06-05:
Hi @ramvelyan, I believe you are on the `depth` dev branch, and there I have added some new functions to `glDisplay`, including [`glDisplay::SetSize()`](https://github.com/dusty-nv/jetson-utils/blob/7d8de5c31fe1199ad01ba078a7303378269fd928/display/glDisplay.h#L274)

So you should be able to do something like this:

``` c++
display->SetSize(camera->GetWidth() + 10, camera->GetHeight() + 30);
```
The +10 and +30 are for a bit of a border so the feed doesn't get chopped off by the window.  I don't think I have the Python bindings for `display.SetSize()` yet, although you can pass width and height to the display constructor now:

``` python
display = jetson.utils.glDisplay(width=camera.GetWidth()+10, height=camera.GetHeight()+30)
```

In the future, I plan to add auto-resizing of `glDisplay` to match the video feed(s).

rsamvelyan, 2020-06-06:
Thanks Dusty, I will try it out. 
I managed to display the glCamera in an OpenCV window with showimg() but I notice a very slight drop in video frame rate (Network FPS are unaffected though).

Quick question on the side, is Jetson Inference an official Nvidia software that will be supported on the long run or is it an initiative on the side? 
I hope it continues to develop because it's absolutely great.

dusty-nv, 2020-06-06:
It's pretty official by GitHub standards, it's also on the NVIDIA website and maintained in the long run.  The project is already several years old and continues to add features.  Glad that it is helpful to you.

rsamvelyan, 2020-06-08:
This totally worked. It's awesome: 
`display = jetson.utils.glDisplay(width=camera.GetWidth()+10, height=camera.GetHeight()+30)`
I hope to see all these features in the documentation at some point.
Thanks a lot!!

closed 2020-06-08 by rsamvelyan