dusty-nv/jetson-inference #607: Detecting objects from RSTP Stream (IP Camera)

tanueihorng, 2020-06-08:
Hi Guys!
I’ve implemented the object detection model by following the tutorial in YouTube (Real-Time Object Detection in 10 Lines of Python Code on Jetson Nano) and it’s using the Raspberry Pi V2 Camera attached on my Jetson Nano.

**What can I do to detect objects from my IP Camera (RSTP) instead of the raspberry pi camera?**
I had tested out the solution to edit the gstCamera.cpp code but it can’t work
(text files attached below are gstcamera.cpp, error message when launching program and mydetection python script

Could someone please provide me some well-detailed solutions?
Thank you so much!

[gstCamera.txt](https://github.com/dusty-nv/jetson-inference/files/4746577/gstCamera.txt)
[Error.txt](https://github.com/dusty-nv/jetson-inference/files/4746557/Error.txt)
[my-detection.txt](https://github.com/dusty-nv/jetson-inference/files/4746570/my-detection.txt)

dusty-nv, 2020-06-08:
Hi @tanueihorng , please check this post from HP:   https://forums.developer.nvidia.com/t/ip-camera-rstp-using-with-jetson-nano/111736/2

I have also been working on adding RTP/RTSP support to gstCamera, but it doesn't have the Python bindings ready yet.

tanueihorng, 2020-06-10:
Thanks @dusty-nv ! I managed to get it working on the detector. However, the color of the display seems very weird and only recognize stop sign. How can I fix that ? (already applied the patch by Honey) 
[image]

closed 2020-07-17 by tanueihorng

nawabali777, 2022-09-03:
@dusty-nv 
I am working on adding RTP/RTSP support to gstCamera but not got any result still. i want to integrate this camera with jetson nano.  can any Python bindings available.now.

thank you for your help

nawabali777, 2022-09-03:
import jetson.inference
import jetson.utils

net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
camera = jetson.utils.gstCamera(640, 480, "rtspsrc location=rtsp://192.168.1.204:554/test latency=0 ! rtph264depay !  h264parse ! nvv4l2decoder ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=480 ")  
display = jetson.utils.glDisplay()

while display.IsOpen():
	img, width, height = camera.CaptureRGBA()
	detections = net.Detect(img, width, height)
	display.RenderOnce(img, width, height)
	display.SetTitle("Object Detection | Network {:.0f} FPS".format(net.GetNetworkFPS()))

nawabali777, 2022-09-03:
dlinano@jetson-nano:~$ python my-detection.py
jetson.inference.__init__.py
jetson.inference -- initializing Python 2.7 bindings...
jetson.inference -- registering module types...
jetson.inference -- done registering module types
jetson.inference -- done Python 2.7 binding initialization
jetson.utils.__init__.py
jetson.utils -- initializing Python 2.7 bindings...
jetson.utils -- registering module functions...
jetson.utils -- done registering module functions
jetson.utils -- registering module types...
jetson.utils -- done registering module types
jetson.utils -- done Python 2.7 binding initialization
jetson.inference -- PyTensorNet_New()
jetson.inference -- PyDetectNet_Init()
jetson.inference -- detectNet loading build-in network 'ssd-mobilenet-v2'
 
detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1
 
[TRT]   TensorRT version 6.0.1
[TRT]   loading NVIDIA plugins...
[TRT]   Plugin Creator registration succeeded - GridAnchor_TRT
[TRT]   Plugin Creator registration succeeded - GridAnchorRect_TRT
[TRT]   Plugin Creator registration succeeded - NMS_TRT
[TRT]   Plugin Creator registration succeeded - Reorg_TRT
[TRT]   Plugin Creator registration succeeded - Region_TRT
[TRT]   Plugin Creator registration succeeded - Clip_TRT
[TRT]   Plugin Creator registration succeeded - LReLU_TRT
[TRT]   Plugin Creator registration succeeded - PriorBox_TRT
[TRT]   Plugin Creator registration succeeded - Normalize_TRT
[TRT]   Plugin Creator registration succeeded - RPROI_TRT
[TRT]   Plugin Creator registration succeeded - BatchedNMS_TRT
[TRT]   Could not register plugin creator:  FlattenConcat_TRT in namespace: 
[TRT]   completed loading NVIDIA plugins.
[TRT]   detected model format - UFF  (extension '.uff')
[TRT]   desired precision specified for GPU: FASTEST
[TRT]   requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]   native precisions detected for GPU:  FP32, FP16
[TRT]   selecting fastest native precision for GPU:  FP16
[TRT]   attempting to open engine cache file /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.GPU.FP16.engine
[TRT]   loading network profile from engine cache... /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.GPU.FP16.engine
[TRT]   device GPU, /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff loaded
[TRT]   Deserialize required 7385762 microseconds.
[TRT]   device GPU, CUDA engine context initialized with 3 bindings
[TRT]   binding -- index   0
               -- name    'Input'
               -- type    FP32
               -- in/out  INPUT
               -- # dims  3
               -- dim #0  3 (SPATIAL)
               -- dim #1  300 (SPATIAL)
               -- dim #2  300 (SPATIAL)
[TRT]   binding -- index   1
               -- name    'NMS'
               -- type    FP32
               -- in/out  OUTPUT
               -- # dims  3
               -- dim #0  1 (SPATIAL)
               -- dim #1  100 (SPATIAL)
               -- dim #2  7 (SPATIAL)
[TRT]   binding -- index   2
               -- name    'NMS_1'
               -- type    FP32
               -- in/out  OUTPUT
               -- # dims  3
               -- dim #0  1 (SPATIAL)
               -- dim #1  1 (SPATIAL)
               -- dim #2  1 (SPATIAL)
[TRT]   binding to input 0 Input  binding index:  0
[TRT]   binding to input 0 Input  dims (b=1 c=3 h=300 w=300) size=1080000
[TRT]   binding to output 0 NMS  binding index:  1
[TRT]   binding to output 0 NMS  dims (b=1 c=1 h=100 w=7) size=2800
[TRT]   binding to output 1 NMS_1  binding index:  2
[TRT]   binding to output 1 NMS_1  dims (b=1 c=1 h=1 w=1) size=4
device GPU, /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff initialized.
W = 7  H = 100  C = 1
detectNet -- maximum bounding boxes:  100
detectNet -- loaded 91 class info entries
detectNet -- number of object classes:  91
jetson.utils -- PyCamera_New()
jetson.utils -- PyCamera_Init()
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera::Create('rtspsrc location=rtsp://192.168.0.90:8554/test latency=0 ! rtph264depay !  h264parse ! nvv4l2decoder ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=480 ') -- invalid camera device requested
jetson.utils -- PyCamera_Dealloc()
Traceback (most recent call last):
  File "my-detection.py", line 5, in <module>
    camera = jetson.utils.gstCamera(640, 480, "rtspsrc location=rtsp://192.168.1.204:554/test latency=0 ! rtph264depay !  h264parse ! nvv4l2decoder ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=480 ")  
Exception: jetson.utils -- failed to create gstCamera device
PyTensorNet_Dealloc()

dusty-nv, 2022-09-03:
> camera = jetson.utils.gstCamera(640, 480, "rtspsrc location=rtsp://192.168.1.204:554/test latency=0 ! rtph264depay ! h264parse ! nvv4l2decoder ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=480 ")

That's not how gstCamera works - try running `detectnet.py --input-rtsp-latency=0 rtsp://192.168.1.204:554/test` first

nawab-sw, 2022-09-05:
root@nawab-desktop:/jetson-inference# detectnet.py --input-rtsp-latency=0 rtsp://192.168.1.204:554     
[OpenGL] glDisplay -- X screen 0 resolution:  1440x900
[OpenGL] glDisplay -- X window resolution:    1440x900
[OpenGL] glDisplay -- display device initialized (1440x900)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1440
  -- height:     900
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 0
------------------------------------------------

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]    TensorRT version 8.2.1
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::ScatterND version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMSDynamic_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_ONNX_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TFTRT_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - UFF  (extension '.uff')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    [MemUsageChange] Init CUDA: CPU +230, GPU +7, now: CPU 258, GPU 1911 (MiB)
[TRT]    [MemUsageSnapshot] Begin constructing builder kernel library: CPU 258 MiB, GPU 1905 MiB
[TRT]    [MemUsageSnapshot] End constructing builder kernel library: CPU 287 MiB, GPU 1914 MiB
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[... 50 lines cut ...]
                -- dim #0  1
                -- dim #1  1
                -- dim #2  1
[TRT]    
[TRT]    binding to input 0 Input  binding index:  0
[TRT]    binding to input 0 Input  dims (b=1 c=3 h=300 w=300) size=1080000
[TRT]    binding to output 0 NMS  binding index:  1
[TRT]    binding to output 0 NMS  dims (b=1 c=1 h=100 w=7) size=2800
[TRT]    binding to output 1 NMS_1  binding index:  2
[TRT]    binding to output 1 NMS_1  dims (b=1 c=1 h=1 w=1) size=4
[TRT]    
[TRT]    device GPU, /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff initialized.
[TRT]    W = 7  H = 100  C = 1
[TRT]    detectNet -- maximum bounding boxes:  100
[TRT]    detectNet -- loaded 91 class info entries
[TRT]    detectNet -- number of object classes:  91
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for 192.168.1.204
[gstreamer] gstDecoder -- Unauthorized
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://192.168.1.204:554
Traceback (most recent call last):
  File "/usr/local/bin/detectnet.py", line 57, in <module>
    input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device

nawab-sw, 2022-09-05:
thank you for you reply
error to run these: detectnet.py --input-rtsp-latency=0 rtsp://192.168.1.204:554

dusty-nv, 2022-09-05:
> [gstreamer] gstDecoder -- Unauthorized

It seems like you need a username/password to access this RTSP stream.   I would try disabling it on the camera first to test that you can view the stream.  Normally the username/password is written as part of the URL.

nawab-sw, 2022-09-06:
no option to disable username and password of camera.
After provide user name & password to the camera ,Same error:  

root@nawab-desktop:/jetson-inference# detectnet.py --input-rtsp-latency=0 rtsp://admin:pass@123@192.168.1.204:554/test
[OpenGL] glDisplay -- X screen 0 resolution:  1440x900
[OpenGL] glDisplay -- X window resolution:    1440x900
[OpenGL] glDisplay -- display device initialized (1440x900)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1440
  -- height:     900
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 0
------------------------------------------------

detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
          -- input_blob   'Input'
          -- output_blob  'NMS'
          -- output_count 'NMS_1'
          -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1

[TRT]    TensorRT version 8.2.1
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::ScatterND version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMSDynamic_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_ONNX_TRT version 1
[TRT]    Registered plugin creator - ::EfficientNMS_TFTRT_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - UFF  (extension '.uff')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    [MemUsageChange] Init CUDA: CPU +230, GPU +0, now: CPU 258, GPU 1950 (MiB)
[TRT]    [MemUsageSnapshot] Begin constructing builder kernel library: CPU 258 MiB, GPU 1944 MiB
[... 58 lines cut ...]
[TRT]    binding to input 0 Input  dims (b=1 c=3 h=300 w=300) size=1080000
[TRT]    binding to output 0 NMS  binding index:  1
[TRT]    binding to output 0 NMS  dims (b=1 c=1 h=100 w=7) size=2800
[TRT]    binding to output 1 NMS_1  binding index:  2
[TRT]    binding to output 1 NMS_1  dims (b=1 c=1 h=1 w=1) size=4
[TRT]    
[TRT]    device GPU, /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff initialized.
[TRT]    W = 7  H = 100  C = 1
[TRT]    detectNet -- maximum bounding boxes:  100
[TRT]    detectNet -- loaded 91 class info entries
[TRT]    detectNet -- number of object classes:  91
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for admin:pass@123@192.168.1.204
[gstreamer] gstDecoder -- Unauthorized
[gstreamer] gstDecoder -- try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder -- failed to create decoder for rtsp://admin:pass@123@192.168.1.204:554/test
Traceback (most recent call last):
  File "/usr/local/bin/detectnet.py", line 57, in <module>
    input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
Exception: jetson.utils -- failed to create videoSource device
root@nawab-desktop:/jetson-inference# vlc
bash: vlc: command not found
root@nawab-desktop:/jetson-inference# vlc
bash: vlc: command not found
root@nawab-desktop:/jetson-inference# detectnet.py --input-rtsp-latency=0 rtsp://admin:pass@123@192.168.1.204:554/test

dusty-nv, 2022-09-06:
https://forums.developer.nvidia.com/t/gst-launch-1-0-rtspsrc/227062

nawab-sw, 2022-09-07:
i have done some change from you link :https://forums.developer.nvidia.com/t/gst-launch-1-0-rtspsrc/227062
camera port change to 8000

now the message is below:

root@nawab-desktop:/jetson-inference# detectnet.py --input-rtsp-latency=0 rtsp://192.168.1.204:8000/test
[OpenGL] glDisplay – X screen 0 resolution: 1440x900
[OpenGL] glDisplay – X window resolution: 1440x900
[OpenGL] glDisplay – display device initialized (1440x900)
[video] created glDisplay from display://0
glDisplay video options:
– URI: display://0
- protocol: display
- location: 0
– deviceType: display
– ioType: output
– codec: raw
– width: 1440
– height: 900
– frameRate: 0.000000
– bitRate: 0
– numBuffers: 4
– zeroCopy: true
– flipMethod: none
– loop: 0
– rtspLatency 0
detectNet – loading detection network model from:
– model networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
– input_blob ‘Input’
– output_blob ‘NMS’
– output_count ‘NMS_1’
– class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
– threshold 0.500000
– batch_size 1

[TRT] TensorRT version 8.2.1
[TRT] loading NVIDIA plugins…
[TRT] Registered plugin creator - ::GridAnchor_TRT version 1
[TRT] Registered plugin creator - ::GridAnchorRect_TRT version 1
[TRT] Registered plugin creator - ::NMS_TRT version 1
[TRT] Registered plugin creator - ::Reorg_TRT version 1
[TRT] Registered plugin creator - ::Region_TRT version 1
[TRT] Registered plugin creator - ::Clip_TRT version 1
[TRT] Registered plugin creator - ::LReLU_TRT version 1
[TRT] Registered plugin creator - ::PriorBox_TRT version 1
[TRT] Registered plugin creator - ::Normalize_TRT version 1
[TRT] Registered plugin creator - ::ScatterND version 1
[TRT] Registered plugin creator - ::RPROI_TRT version 1
[TRT] Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT] Registered plugin creator - ::BatchedNMSDynamic_TRT version 1
[TRT] Could not register plugin creator - ::FlattenConcat_TRT version 1
[TRT] Registered plugin creator - ::CropAndResize version 1
[TRT] Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT] Registered plugin creator - ::EfficientNMS_TRT version 1
[TRT] Registered plugin creator - ::EfficientNMS_ONNX_TRT version 1
[TRT] Registered plugin creator - ::EfficientNMS_TFTRT_TRT version 1
[TRT] Registered plugin creator - ::Proposal version 1
[TRT] Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT] Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT] Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT] Registered plugin creator - ::Split version 1
[TRT] Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT] Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT] detected model format - UFF (extension ‘.uff’)
[TRT] desired precision specified for GPU: FASTEST
[TRT] requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT] [MemUsageChange] Init CUDA: CPU +230, GPU +0, now: CPU 258, GPU 1922 (MiB)
[TRT] [MemUsageSnapshot] Begin constructing builder kernel library: CPU 258 MiB, GPU 1917 MiB
[TRT] [MemUsageSnapshot] End constructing builder kernel library: CPU 287 MiB, GPU 1920 MiB
[TRT] native precisions detected for GPU: FP32, FP16
[... 54 lines cut ...]
[TRT]
[TRT] binding to input 0 Input binding index: 0
[TRT] binding to input 0 Input dims (b=1 c=3 h=300 w=300) size=1080000
[TRT] binding to output 0 NMS binding index: 1
[TRT] binding to output 0 NMS dims (b=1 c=1 h=100 w=7) size=2800
[TRT] binding to output 1 NMS_1 binding index: 2
[TRT] binding to output 1 NMS_1 dims (b=1 c=1 h=1 w=1) size=4
[TRT]
[TRT] device GPU, /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff initialized.
[TRT] W = 7 H = 100 C = 1
[TRT] detectNet – maximum bounding boxes: 100
[TRT] detectNet – loaded 91 class info entries
[TRT] detectNet – number of object classes: 91
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder – creating decoder for 192.168.1.204
[gstreamer] gstDecoder – Could not read from resource.
[gstreamer] gstDecoder – try manually setting the codec with the --input-codec option
[gstreamer] gstDecoder – failed to create decoder for rtsp://192.168.1.204:8000/test
Traceback (most recent call last):
File “/usr/local/bin/detectnet.py”, line 57, in
input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
Exception: jetson.utils – failed to create videoSource device
root@nawab-desktop:/jetson-inference#

kindly help me

nawab-sw, 2022-09-07:
i done and got the stream image continue.thank you very much sir.