dusty-nv/jetson-inference #608: ProfilerTimes may be wrong

msts35, 2020-06-09:
Hi , 
When i test the model with net.PrintProfilerTimes() it gives me 40FPS, however when I put time.time at the start and end of the while loop and calculate the difference it gives me 10FPS , which one is true ? 
does not the speed of the loop represent the speed?

dusty-nv, 2020-06-09:
Hi @msts35, `net.PrintProfileTimes()` prints out the times it takes for DNN inferencing, including pre/post-processing of the tensors.   The main loop may include other things, like camera aquisition, image loading, rendering, ect.  `glDisplay.GetFPS()` shows the speed of rendering.

leandrovrabelo, 2020-06-09:
Using the module: from datetime import datetime

I use:
start = datetime.now().time()
inference()
end = datetime.now().time()

The difference show exactly what `net.PrintProfileTimes()` shows.

msts35, 2020-06-09:
Thank you dustin and leandro, it lets me be disappointed.

So the program just gives me 10 frame with prediction at a second, am I right?

dusty-nv, 2020-06-09:
On your setup, yes - perhaps your camera is only running at 10 FPS.

msts35, 2020-06-09:
When I measure the image loading function, it takes 0.06ms but the network just use 0.03ms, so the bottleneck is loading image, isn't it funny a DNN is working faster than the process of loading a frame :) Reducing image quality makes smoother but just 5 fps improvement at 360p, is there any available way to load images faster?

closed 2020-06-15 by msts35

msts35, 2020-06-15:
thank you for mapping, I think to check if detection[n].left or top are negative value then make them 0 also if top+height or left+width bigger than resolution then decrease the height or width to fit resolution  of frame, is it logical?

I dont want to pass detected object, if it is possible