dusty-nv/jetson-inference #610: Build error when using make in official nvidia docker

TomEelbode, 2020-06-11:
I'm trying to build this project from source in a docker container (I'm using a clean nvcr.io/nvidia/l4t-tensorflow:r32.4.2-tf1.15-py3 image). 

However, the make command gives the following error:

> Scanning dependencies of target jetson-inference
[ 60%] Building CXX object CMakeFiles/jetson-inference.dir/c/detectNet.cpp.o
[ 61%] Building CXX object CMakeFiles/jetson-inference.dir/c/homographyNet.cpp.o
[ 62%] Building CXX object CMakeFiles/jetson-inference.dir/c/imageNet.cpp.o
[ 63%] Building CXX object CMakeFiles/jetson-inference.dir/c/segNet.cpp.o
[ 64%] Building CXX object CMakeFiles/jetson-inference.dir/calibration/randInt8Calibrator.cpp.o
[ 65%] Building CXX object CMakeFiles/jetson-inference.dir/c/superResNet.cpp.o
[ 67%] Building CXX object CMakeFiles/jetson-inference.dir/c/tensorNet.cpp.o
[ 68%] Building CXX object CMakeFiles/jetson-inference.dir/plugins/FlattenConcat.cpp.o
[ 69%] Linking CXX shared library aarch64/lib/libjetson-inference.so
/usr/bin/ld: cannot find -lnvcaffe_parser
collect2: error: ld returned 1 exit status
CMakeFiles/jetson-inference.dir/build.make:335: recipe for target 'aarch64/lib/libjetson-inference.so' failed
make[2]: *** [aarch64/lib/libjetson-inference.so] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

dusty-nv, 2020-06-11:
I believe nvcaffe_parser functionality is now in nvparser library, so you should be able to remove it from CMakeLists.txt

TomEelbode, 2020-06-12:
@dusty-nv Thanks for your quick reply! I have tried this, but a new error arises later on (see below).

Just to make sure that I did exactly what you meant: I replaced this line in jetson-inference/CMakeLists.txt
`target_link_libraries(jetson-inference jetson-utils nvinfer nvinfer_plugin nvcaffe_parser)`
by
`target_link_libraries(jetson-inference jetson-utils nvinfer nvinfer_plugin)`

The new error on executing make is

> [ 53%] Linking CXX executable ../../aarch64/bin/imagenet-console
../../aarch64/lib/libjetson-inference.so: undefined reference to `nvcaffeparser1::createCaffeParser()'
../../aarch64/lib/libjetson-inference.so: undefined reference to `nvuffparser::createUffParser()'
collect2: error: ld returned 1 exit status
examples/imagenet-console/CMakeFiles/imagenet-console.dir/build.make:109: recipe for target 'aarch64/bin/imagenet-console' failed
make[2]: *** [aarch64/bin/imagenet-console] Error 1
CMakeFiles/Makefile2:183: recipe for target 'examples/imagenet-console/CMakeFiles/imagenet-console.dir/all' failed
make[1]: *** [examples/imagenet-console/CMakeFiles/imagenet-console.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

TomEelbode, 2020-06-15:
UPDATE: it works now :)

I stupidly forgot to link the mentioned nvparser libary instead.

The line in CMakeLists.txt should be (mind that it's nvparsers and not nvparser):
`target_link_libraries(jetson-inference jetson-utils nvinfer nvinfer_plugin nvparsers)`

This line should be changed in the following files:
```
jetson-inference/CMakeLists.txt
jetson-inference/tools/trt-bench/CMakeLists.txt
jetson-inference/tools/trt-console/CMakeLists.txt
```

closed 2020-06-15 by TomEelbode

TomEelbode, 2020-06-15:
#611

TomEelbode, 2020-06-15:
#611

jpizarrom, 2020-07-19:
Use nvparsers instead of nvcaffe_parser works with nvcr.io/nvidia/l4t-ml:r32.4.3-py3