dusty-nv/jetson-inference #613: opencv integration

msts35, 2020-06-15:
hi dustin,

I converted videocapture process from gstreamer to opencv via using numpyTocuda functions, it worked well however when an objects bounding box extends from screen so a value of its any corner has negative value the program fails, can you please give me suggestions?

dusty-nv, 2020-06-15:
Ah, it looks like I need to limit the bounds of the boxes in this function:   https://github.com/dusty-nv/jetson-inference/blob/b734d9f5d091e81685877b19a9d1b4483c3b6321/c/detectNet.cu#L91

Will put it on my todo list, thank you for letting me know.  In the meantime, to get back up and running asap, in that function you could add a simple check the left/top coordinate of each detection to see if they are negative, and if so set them to zero.

msts35, 2020-06-15:
thank you for mapping, i think to change detections[n].left or top to 0 if they are negative, also if left+width or height are greater than image resolution accordingly decrease them, is it a possible and logical solution? 
I dont want to pass detected object for this situation,i asked because could not try it now, i will try and report to you tomorrow, it will be big honor if I could help

dusty-nv, 2020-06-16:
OK, I just patched this in commit [`0a75e3`](https://github.com/dusty-nv/jetson-inference/commit/0a75e3059a93945dcd9570a5c9f3efdd9c33ce38).  I ended up clamping the bounds in detectNet.cpp to avoid any issues with the classname label rendering too.

I haven't seen these negative bounding boxes myself, so can you try this when you get a chance and let me know if it now works?

msts35, 2020-06-16:
Hello dustin, i can try it now and the problem seems to be resolved, thank
you very much for your interest

Dustin Franklin <notifications@github.com>, 16 Haz 2020 Sal, 04:27
tarihinde şunu yazdı:

msts35, 2020-06-16:
Sorry i get the same error like this 

`jetson.utils -- cudaFromNumpy()  ndarray dim 0 = 480
jetson.utils -- cudaFromNumpy()  ndarray dim 1 = 640
jetson.utils -- cudaFromNumpy()  ndarray dim 2 = 4
jetson.utils -- freeing CUDA mapped memory
[cuda]   cudaDeviceSynchronize()
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/c/detectNet.cpp:853
jetson.inference -- PyDetection_Dealloc()
<detectNet.Detection object>
   -- ClassID: 1
   -- Confidence: 0.72304
   -- Left:    53.125
   -- Top:     -30
   -- Right:   686.25
   -- Bottom:  481.875
   -- Width:   633.125
   -- Height:  511.875
   -- Area:    324081
   -- Center:  (369.688, 225.938)
<detectNet.Detection object>
   -- ClassID: 1
   -- Confidence: 0.95116
   -- Left:    -10.625
   -- Top:     2.92969
   -- Right:   569.375
   -- Bottom:  477.188
   -- Width:   580
   -- Height:  474.258
   -- Area:    275070
   -- Center:  (279.375, 240.059)
[cuda]   cudaGetLastError()
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/utils/cuda/cudaNormalize.cu:65
[cuda]   cudaNormalizeRGBA((float4*)img, make_float2(0.0f, 255.0f), (float4*)mNormalizedCUDA, make_float2(0.0f, 1.0f), width, height)
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/utils/display/glDisplay.cpp:513
[cuda]   cudaGraphicsMapResources(1, &interop)
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/utils/display/glTexture.cpp:471
[cuda]   cudaEventElapsedTime(&cuda_time, mEventsGPU[evt], mEventsGPU[evt+1])
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:687

[TRT]    ------------------------------------------------
[TRT]    Timing Report /home/mesut/jetson-inference/data/networks/SSD-Mobilenet-v2/mb1-ssd.onnx
[TRT]    ------------------------------------------------
[cuda]   cudaEventElapsedTime(&cuda_time, mEventsGPU[evt], mEventsGPU[evt+1])
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:687
[TRT]    Pre-Process   CPU   0.10541ms  CUDA   0.00000ms
[TRT]    Network       CPU  27.47079ms  CUDA   0.00000ms
[cuda]   cudaEventElapsedTime(&cuda_time, mEventsGPU[evt], mEventsGPU[evt+1])
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:687
[TRT]    Post-Process  CPU   1.00816ms  CUDA   0.00000ms
[cuda]   cudaEventElapsedTime(&cuda_time, mEventsGPU[evt], mEventsGPU[evt+1])
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/build/aarch64/include/jetson-inference/tensorNet.h:687
[TRT]    Visualize     CPU   0.18940ms  CUDA   0.00000ms
[TRT]    Total         CPU  28.77375ms  CUDA   0.00000ms
[TRT]    ------------------------------------------------

0.1586780548095703
jetson.utils -- cudaFromNumpy()  ndarray dim 0 = 480
jetson.utils -- cudaFromNumpy()  ndarray dim 1 = 640
jetson.utils -- cudaFromNumpy()  ndarray dim 2 = 4
[cuda]   cudaHostAlloc(cpuPtr, size, 0x02)
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/build/aarch64/include/jetson-utils/cudaMappedMemory.h:50
Traceback (most recent call last):
  File "detectnet-cameraCV2.py", line 57, in <module>
    img = jetson.utils.cudaFromNumpy(frame_rgba)
Exception: jetson.utils -- cudaAllocMapped() failed
PyTensorNet_Dealloc()
[cuda]   cudaFreeHost(mDetectionSets[0])
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/c/detectNet.cpp:67
[cuda]   cudaFreeHost(mClassColors[0])
[cuda]      unspecified launch failure (error 719) (hex 0x2CF)
[cuda]      /home/mesut/jetson-inference/c/detectNet.cpp:75
[TRT]    ../rtExt/cuda/cudaFusedConvActRunner.cpp (90) - Cuda Error in destroyFilterTexture: 719 (unspecified launch failure)
[TRT]    INTERNAL_ERROR: std::exception
[TRT]    ../rtSafe/safeRuntime.cpp (32) - Cuda Error in free: 719 (unspecified launch failure)
terminate called after throwing an instance of 'nvinfer1::CudaError'
  what():  std::exception
Aborted
`

msts35, 2020-06-16:
I ignore the "box" from

model.detect(img,width,height,"labels,conf")

now I can not see the boxes however program works

Mesut Şimşek <mesutttsimsek@gmail.com>, 16 Haz 2020 Sal, 18:14 tarihinde
şunu yazdı:

dusty-nv, 2020-06-16:
After you pulled the latest, did you do a `make` followed by `sudo make install`?

since negative coordinates are still in your detection log output, I think the patched code may not have been running.

msts35, 2020-06-16:
No i did not do sudo make install, i will thank you

Dustin Franklin <notifications@github.com>, 16 Haz 2020 Sal, 19:54
tarihinde şunu yazdı:

closed 2020-06-18 by msts35