dusty-nv/jetson-inference #616: How to make bounding boxes hollow?

wrz57661, 2020-06-17:
hi dustin,

I use python for detection via mobilenetv2. It worked very well. However the bounding boxes are solid. I want to make it hollow. 
The action of adding a bounding box is operated by the cpp library.I have found the cudaOverlay.cu file which is located in jetson inference/utils/cuda/. Is this file correct? How can I change the function to make it hollow? Do I need to reinstall jetson inference after the modification is completed?
Thank you very much!

wrz57661, 2020-06-17:
Here is the content of cudaOverlay.cu.

template<typename T>
__global__ void gpuRectFill( T* input, T* output, int width, int height,
					    float4* rects, int numRects, float4 color ) 
{
	const int x = blockIdx.x * blockDim.x + threadIdx.x;
	const int y = blockIdx.y * blockDim.y + threadIdx.y;

	if( x >= width || y >= height )
		return;

	const T px_in  = input[ y * width + x ];
	      T px_out = px_in;
	
	const float fx = x;
	const float fy = y;
	
	//const float thick = 10.0f;
	const float alpha = color.w / 255.0f;
	const float ialph = 1.0f - alpha;
	
	for( int nr=0; nr < numRects; nr++ )
	{
		const float4 r = rects[nr];
		
		printf("%i %i %i  %f %f %f %f\n", numRects, x, y, r.x, r.y, r.z, r.w);
		
		if( fy >= r.y && fy <= r.w /*&& (eq_less(fx, r.x, ep) || eq_less(fx, r.z, ep))*/ )
		{
			if( fx >= r.x && fx <= r.z /*&& (eq_less(fy, r.y, ep) || eq_less(fy, r.w, ep))*/ )
			{
				printf("cuda rect %i %i\n", x, y);

				px_out.x = alpha * color.x + ialph * px_out.x;
				px_out.y = alpha * color.y + ialph * px_out.y;
				px_out.z = alpha * color.z + ialph * px_out.z;
			}
		}
	}
	
	output[y * width + x] = px_out;	 
}

dusty-nv, 2020-06-17:
Hi @wrz57661, the detectNet overlay code is in `detectNet.cu`:   https://github.com/dusty-nv/jetson-inference/blob/master/c/detectNet.cu

It is similar to cudaOverlay, except it is specialized to draw `detectNet::Detection` lists.

Also, if you note the commented out parts from cudaOverlay like `eq_less(fx, r.x, ep) || eq_less(fx, r.z, ep)` - it is from when I previously tried line rendering (instead of filled box).  However unfortunately it didn't work for some reason...

wrz57661, 2020-06-18:
Thank you very much,dustin!  I read the detectNet.cu and found that I'm not familiar with C++. I couldn't change it because I don't want to damage it.
So I took another measure to achieve my goal. 

detections = model.Detect(img, width, height,"labels,conf")

I blocked the  overlay box function and got the detections' left,top,width,height.Then use opencv to add rectangle.It worked. Thank you again for your reply!

samiit, 2020-07-28:
Hi @wrz57661 

Can you share what you set as the `overlay` parameter? E.g., what does `'conf'` stand for? How did you link it with openCV to get the bounding boxes?

Thanks,
Sam

BenSirenden, 2020-11-12:
Hi Dustin

I want to draw boxe in specific coordinate in the video, i'm using python, how i do that with jetson.utils? I also want to detect just human object, how to do that with jetson.inference. the goal is to detect human inside specific box in the video.

thank you

wrz57661, 2020-11-12:
@samiit Sorry for watching you too late!
In the example detectnet.py,63th row is "detections = net.Detect(img, overlay=opt.overlay)".
You can see opt.overlay == "box,labels,conf" in 38th row.
I replace it with None or null.
After net.Detect is done,the model will output dictionary include left,Top,Width,Height keys for every boundingbox, you can use these data to draw.

samiit, 2020-11-12:
Thanks @wrz57661, I will try it out

Shayan-Bravo, 2021-05-20:
> Thank you very much,dustin!@dusty-nv  I read the detectNet.cu and found that I'm not familiar with C++. I couldn't change it because I don't want to damage it.
> So I took another measure to achieve my goal. 
> 
> detections = model.Detect(img, width, height,"labels,conf")
> 
> I blocked the  overlay box function and got the detections' left,top,width,height.Then use opencv to add rectangle.It worked. Thank you again for your reply!
> 

When you used opencv, did it decrease your FPS? Also how were you able to integrate opencvs rectangle option into the gstreamer? I would love to see your code for it if that's possible.

OBLateral, 2021-06-14:
> @samiit Sorry for watching you too late!
> In the example detectnet.py,63th row is "detections = net.Detect(img, overlay=opt.overlay)".
> You can see opt.overlay == "box,labels,conf" in 38th row.
> I [...]

it kinda sux that you are being so vague with helping others! the image is a cuda image and is not that simple to just draw on. It would not of been hard for you to show people what had to be done.  it is clearly not as simple as drawing a rectangle on it. You asked for help in the first place and people tried to help you the least you can do is behave that way in kind!!

OBLateral, 2021-06-14:
This is the output when attempting to draw on the image before calling output.Render(img):

```
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from NULL to READY ==> h264parse1
[gstreamer] gstreamer changed state from NULL to READY ==> queue0
[gstreamer] gstreamer changed state from NULL to READY ==> qtdemux1
[gstreamer] gstreamer changed state from NULL to READY ==> filesrc0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter1
[gstreamer] gstreamer changed state from READY to PAUSED ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from READY to PAUSED ==> h264parse1
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer changed state from READY to PAUSED ==> queue0
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer stream status CREATE ==> sink
[gstreamer] gstreamer changed state from READY to PAUSED ==> qtdemux1
[gstreamer] gstreamer changed state from READY to PAUSED ==> filesrc0
[gstreamer] gstreamer stream status ENTER ==> sink

(python3:20443): GStreamer-CRITICAL **: 14:42:57.162: gst_caps_is_empty: assertion 'GST_IS_CAPS (caps)' failed

(python3:20443): GStreamer-CRITICAL **: 14:42:57.162: gst_caps_truncate: assertion 'GST_IS_CAPS (caps)' failed

(python3:20443): GStreamer-CRITICAL **: 14:42:57.162: gst_caps_fixate: assertion 'GST_IS_CAPS (caps)' failed

(python3:20443): GStreamer-CRITICAL **: 14:42:57.162: gst_caps_get_structure: assertion 'GST_IS_CAPS (caps)' failed

(python3:20443): GStreamer-CRITICAL **: 14:42:57.162: gst_structure_get_string: assertion 'structure != NULL' failed

(python3:20443): GStreamer-CRITICAL **: 14:42:57.162: gst_mini_object_unref: assertion 'mini_object != NULL' failed
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
Allocating new output: 1088x1088 (x 14), ThumbnailMode = 0
OPENMAX: HandleNewStreamFormat: 3605: Send OMX_EventPortSettingsChanged: nFrameWidth = 1080, nFrameHeight = 1080 
[gstreamer] gstDecoder -- onPreroll()
[gstreamer] gstreamer message stream-start ==> pipeline0
[gstreamer] gstreamer stream status CREATE ==> src
[gstreamer] gstreamer stream status ENTER ==> src
[gstreamer] gstreamer message duration-changed ==> h264parse1
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ /\ AVC", language-code=(string)en, bitrate=(uint)10090419;
[gstreamer] gstreamer mysink taglist, datetime=(datetime)2021-05-18T19:49:29Z, private-qt-tag=(sample){ 00000017a954494d000b000030303a30323a30303a3233:None:R3N0U2VnbWVudCwgZmxhZ3M9KEdzdFNlZ21lbnRGbGFncylHU1RfU0VHTUVOVF9GTEFHX05PTkUsIHJhdGU9KGRvdWJsZSkxLCBhcHBsaWVkLXJhdGU9KGRvdWJsZSkxLCBmb3JtYXQ9KEdzdEZvcm1hdClHU1RfRk9STUFUX1RJTUUsIGJhc2U9KGd1aW50NjQpMCwgb2Zmc2V0PShndWludDY0KTAsIHN0YXJ0PShndWludDY0KTAsIHN0b3A9KGd1aW50NjQpMTg0NDY3NDQwNzM3MDk1NTE2MTUsIHRpbWU9KGd1aW50NjQpMCwgcG9zaXRpb249KGd1aW50NjQpMCwgZHVyYXRpb249KGd1aW50NjQpMTg0NDY3NDQwNzM3MDk1NTE2MTU7AA__:YXBwbGljYXRpb24veC1nc3QtcXQtX3RpbS10YWcsIHN0eWxlPShzdHJpbmcpaXNvOwA_, 0000000ea9545343000200003235:None:R3N0U2VnbWVudCwgZmxhZ3M9KEdzdFNlZ21lbnRGbGFncylHU1RfU0VHTUVOVF9GTEFHX05PTkUsIHJhdGU9KGRvdWJsZSkxLCBhcHBsaWVkLXJhdGU9KGRvdWJsZSkxLCBmb3JtYXQ9KEdzdEZvcm1hdClHU1RfRk9STUFUX1RJTUUsIGJhc2U9KGd1aW50NjQpMCwgb2Zmc2V0PShndWludDY0KTAsIHN0YXJ0PShndWludDY0KTAsIHN0b3A9KGd1aW50NjQpMTg0NDY3NDQwNzM3MDk1NTE2MTUsIHRpbWU9KGd1aW50NjQpMCwgcG9zaXRpb249KGd1aW50NjQpMCwgZHVyYXRpb249KGd1aW50NjQpMTg0NDY3NDQwNzM3MDk1NTE2MTU7AA__:YXBwbGljYXRpb24veC1nc3QtcXQtX3RzYy10YWcsIHN0eWxlPShzdHJpbmcpaXNvOwA_, 0000000da954535a0001000031:None:R3N0U2VnbWVudCwgZmxhZ3M9KEdzdFNlZ21lbnRGbGFncylHU1RfU0VHTUVOVF9GTEFHX05PTkUsIHJhdGU9KGRvdWJsZSkxLCBhcHBsaWVkLXJhdGU9KGRvdWJsZSkxLCBmb3JtYXQ9KEdzdEZvcm1hdClHU1RfRk9STUFUX1RJTUUsIGJhc2U9KGd1aW50NjQpMCwgb2Zmc2V0PShndWludDY0KTAsIHN0YXJ0PShndWludDY0KTAsIHN0b3A9KGd1aW50NjQpMTg0NDY3NDQwNzM3MDk1NTE2MTUsIHRpbWU9KGd1aW50NjQpMCwgcG9zaXRpb249KGd1aW50NjQpMCwgZHVyYXRpb249KGd1aW50NjQpMTg0NDY3NDQwNzM3MDk1NTE2MTU7AA__:YXBwbGljYXRpb24veC1nc3QtcXQtX3Rzei10YWcsIHN0eWxlPShzdHJpbmcpaXNvOwA_ }, container-format=(string)"ISO\ MP4/M4A", image-orientation=(string)rotate-0;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", language-code=(string)en, bitrate=(uint)10090419;
[gstreamer] gstDecoder recieve caps:  video/x-raw, format=(string)NV12, width=(int)1080, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)bt709, framerate=(fraction)25/1
[gstreamer] gstDecoder -- recieved first frame, codec=h264 format=nv12 width=1080 height=1080 size=1749600
RingBuffer -- allocated 4 buffers (1749600 bytes each, 6998400 bytes total)
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer message async-done ==> pipeline0
[gstreamer] gstreamer message new-clock ==> pipeline0
[... cut ...]
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", language-code=(string)en, bitrate=(uint)10090419, minimum-bitrate=(uint)7159800, maximum-bitrate=(uint)19658800;
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Main\ Profile\)", language-code=(string)en, bitrate=(uint)10090419, minimum-bitrate=(uint)361200, maximum-bitrate=(uint)19658800;
Segmentation fault (core dumped)
```

So how does one draw on the cuda image and feed it back into gstreamer without crashing it please??

dusty-nv, 2021-06-14:
You need to use `cudaToNumpy()` and `cudaFromNumpy()` - https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-to-numpy-arrays

OBLateral, 2021-06-14:
> You need to use `cudaToNumpy()` and `cudaFromNumpy()` - https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-to-numpy-arrays

thank you very much will give it a go now :)

OBLateral, 2021-06-14:
Ok here is the code to achieve this, note I did a quick and dirty fix for getting the labels but its enough to get anyone going at leas the box issue is solved

for those wondering how much time it adds it is best to test that on your system, just uncomment the time_perf stuff. For me each box took between 0.02 and 0.03 seconds to draw and composite.

```
import jetson.inference
import jetson.utils
import argparse
import sys
import numpy
import cv2
from time import perf_counter

# parse the command line
parser = argparse.ArgumentParser(description="inference test tool",
                                 formatter_class=argparse.RawTextHelpFormatter,
                                 epilog=jetson.inference.detectNet.Usage() +
                                        jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("input_URI", type=str, default="", nargs='?', help="URI of the input stream")
parser.add_argument("output_URI", type=str, default="", nargs='?', help="URI of the output stream")
parser.add_argument("--network", type=str, default="ssd-mobilenet-v2",
                    help="pre-trained model to load (see below for options)")
parser.add_argument("--overlay", type=str, default="box,labels,conf",
                    help="detection overlay flags (e.g. --overlay=box,labels,conf)\nvalid combinations are:  'box', "
                         "'labels', 'conf', 'none'")
parser.add_argument("--threshold", type=float, default=0.5, help="minimum detection threshold to use")
# added labels argument
parser.add_argument("--labels", type=str, default="labels.txt", help="labels txt file")

is_headless = ["--headless"] if sys.argv[0].find('console.py') != -1 else [""]

try:
    opt = parser.parse_known_args()[0]
except:
    print("")
    parser.print_help()
    sys.exit(0)

# load the object detection network
net = jetson.inference.detectNet(opt.network, sys.argv, opt.threshold)

# create video sources & outputs
input = jetson.utils.videoSource(opt.input_URI, argv=sys.argv)
output = jetson.utils.videoOutput(opt.output_URI, argv=sys.argv + is_headless)

labels_for_box = opt.labels
label_file = open(labels_for_box)
label_data = label_file.readlines()
print(f'label data: {label_data}')

# process frames until the user exits
while True:

    # capture the next image
    img = input.Capture()
    detections = net.Detect(img, overlay='None')

    for detection in detections:
        print(f'detection: {detection}')
        # t1_start = perf_counter()
        box_color = (255, 255, 255)
        # obtain x0,y0 x1,y1 of the detection box
        shape_a = (round(detection.Left), round(detection.Bottom))
        shape_b = (round(detection.Right), round(detection.Top))
        # convert the cuda image to a numpy array
        img_array = jetson.utils.cudaToNumpy(img)
        # convert to RGB
        rgb_array = cv2.cvtColor(img_array, cv2.COLOR_BGR2RGB)
        # this is a quick hack for my personal use as I only have two classes (BACKGROUND is 0 remember!)
        # NOTE: when selecting colour here it is BRG!
        if detection.ClassID == 1:
            box_color = (0, 255, 255)
        if detection.ClassID == 2:
            box_color = (0, 0, 255)
        # draw box
        cv2.rectangle(rgb_array, shape_a, shape_b, box_color, 4)
        # obtain class label based on detection class id
        class_label = label_data[detection.ClassID]
        # strip new lines otherwise they appear as ? on the label
        stripped_label = class_label.rstrip('\n')
        # place text using the top left, can be modified
        cv2.putText(rgb_array, stripped_label,
                    (round(detection.Left) + 20, round(detection.Top) + 40),
                    cv2.FONT_HERSHEY_SIMPLEX,
                    1,  # font scale
                    (255, 255, 255),
                    2)  # line type
        # convert back to BGR
        new_img = cv2.cvtColor(rgb_array, cv2.COLOR_RGB2BGR)
        # composite new image onto the original
        img = jetson.utils.cudaFromNumpy(new_img)
        # t1_stop = perf_counter()
        # time_taken = t1_stop - t1_start
        # print(f'time to draw one box: {time_taken}')

    # render the image
    output.Render(img)

    # update the title bar
    output.SetStatus("{:s} | Network {:.0f} FPS".format(opt.network, net.GetNetworkFPS()))

    # print out performance info
    net.PrintProfilerTimes()

    # exit on input/output EOS
    if not input.IsStreaming() or not output.IsStreaming():
        break

```

Awais-commit, 2021-07-05:
Thanks @dusty-nv for your great great work, also @OBLateral this is exactly what I wanted and find it very useful,
Thanks for sharing your complete code and guide, newbies like me would wonder where to add stuff while not messing up other things so Thanks and it works perfectly.

Although there are something missing in it that, if you would help with that

a) The video doesn't display While Inferencing, only logs display in the terminal, I have to wait till it complete inference and then it saved in the given folder to view the output. Also `output.Render(img)` is included but doesn't show the output video I don't know why. Can we make it to show the video during the inference .

b) The inference doesn't show the confidence score, Where to change so we can get the confidence score.

Thanks again for help

closed 2023-04-27 by dusty-nv